Koenkk/zigbee2mqtt #6982: CC2531 dongle throws error when allowing devices to join

tinuthir, 2021-04-07:
## What happened
I have a hassio instance running with a CC2531 dongle. The setup has worked perfectly fine in the past. Recently, I get an error when allowing new devices to join. An interview doesn't even take place - I suspect hardware failure, but want to make sure.

## What did you expect to happen
A device interview is started

## How to reproduce it (minimal and precise)
Trigger the automation to allow Zigbee joining:

```
- id: enable_zigbee_join
  alias: Enable Zigbee joining
  trigger:
    platform: state
    entity_id: input_boolean.zigbee_permit_join
    to: 'on'
  action:
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/bridge/config/permit_join
      payload: 'true'
  - service: timer.start
    data:
      entity_id: timer.zigbee_permit_join
```

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531 dongle, specifically one from "ITSTUFF" via Amazon, firmware was pre-flashed
Adapter firmware version: 08.06.2019

Log:
```
Zigbee2MQTT:info  2021-04-07 12:03:43: Zigbee: allowing new devices to join.
Zigbee2MQTT:error 2021-04-07 12:03:43: Failed to call 'BridgeLegacy' 'onMQTTMessage' (Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":87,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,30,2,11,254,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58))
```

There's no herdsman logging because it doesn't even get that far.

TheStigh, 2021-04-07:
Try to use the Allow Join found in the Z2M Frontend to allow JOIN, don't rely on custom made scripts when troubleshooting. Follow the log while pressing the button in Z2M Frontend.

tinuthir, 2021-04-07:
The CC is a dongle, so that shouldn't be an issue. 

However, I was able to get things going again. Turns out the add-on wasn't up to date either (not sure why HASS didn't show me that an update was available, but oh well). Not sure what the issue eventually was, but here's what I did:
1. Backup the Z2M config
2. Uninstall Z2M (might not have been necessary in the end, but oh well)
3. Remove the Z2M repo
4. Re-add the Z2M repo
5. Install the latest version of Z2M
6. Crucially, **use the Z2M frontend** to allow joining, as suggested by the savior of this mess, @TheStigh (THANK YOU)
7. Re-pair every Zigbee device (only a dozen or so, not too much trouble)

And see, it works. No hardware failure, apparently. Poor bastard who sold me the CC dongle, he already said he'd send me a new one, free of charge. I have to buy him a coffee or something.

Thanks for your help :) I'll close this issue.

closed 2021-04-07 by tinuthir

TheStigh, 2021-04-07:
I can then tell you the cause of not updating... They changed the URL for repo between 1.18.1 and 1.18.2 

Good it works again, always the best solution :)