Koenkk/zigbee2mqtt #6984: Unstable communication with OSRAM bulbs

Woet, 2021-04-07:
## What happened
Sending commands to the OSRAM E14 and E27 bulbs fail most of the time, resulting in "MAC no ack" messages:
`Data request failed with error: 'MAC no ack' (233)`

Specifically these devices:
https://www.zigbee2mqtt.io/devices/AC03642.html#osram-ac03642
https://www.zigbee2mqtt.io/devices/AB32840.html#osram-ab32840

There are no such issues with the Hue bulbs that I use in the same network.

I've seen that the source routing firmware solved the issue for some people, but can I switch to this firmware without re-pairing my 54 devices?

## What did you expect to happen
The commands to always succeed.

## How to reproduce it (minimal and precise)
Turn the OSRAM E14 or E27 bulb on/off repeatedly, eventually it will fail.

## Debug info
**Zigbee2MQTT version:** 1.18.2-1
**Adapter hardware:** zzh! CC2652R 
**Adapter firmware version:** CC2652R_coordinator_20210120.hex
**Normal and herdsman debug logs:** https://gist.github.com/Woet/cbc9f2de8f5a32918a868599c83c5381

Koenkk, 2021-04-07:
Likely this is caused by interference/connection problems.

Woet, 2021-04-07:
Sorry, I forgot to mention, I've already tried the following steps:
- Changed my Zigbee channel to 26
- Changed my 2.4 GHz WiFi channel to 1
- Put the Pi/Zzh in the centre of my house
- Put the Zzh on an USB 2 extension cable
- Added plenty of routes (Hue and OSRAM lights)

And I didn't have any of these issues using deCONZ/ConBee II nor do I have any issues with the 47 other devices I have (IKEA, Hue, Aqara, Heiman). The issues are limited to the OSRAM bulbs.

hellcry37, 2021-04-09:
I've also try this and idd at one point bulb remained on and i could not do anything to it for a few seconds (like it was forced on) after a few seconds passed everything went normal and did not manifest again.

sakara, 2021-04-17:
Same problem here, no problem with Ikea bulbs or switch/relays

Woet, 2021-04-26:
Is there anything I can do to help resolve this issue?  I have random lights in my house that are either refusing to turn on or turn off and it's getting a bit annoying. These OSRAMs are often for sale for 8-9 euros on Amazon, but I'll be happy to send you one if that would help. :)

Is it possible a firmware update would resolve it? See my comment here: https://github.com/Koenkk/zigbee2mqtt/discussions/2921#discussioncomment-659409

hellcry37, 2021-05-04:
> 
> 
> Likely this is caused by interference/connection problems: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html

this is not an interference problem 100%, please consider investigating this issue

Woet, 2021-05-30:
This issue is still happening with all my OSRAM bulbs, they will randomly refuse to turn on/off several times a day. It's pretty much a ~ 25% chance whether they work.

Besides the "MAC no ack" error, there are a few more now:
```
Timeout - 1105 - 3 - 178 - 6 - 11 after 10000ms
Data request failed with error: 'No network route' (205)
SREQ '--> ZDO - extRouteDisc - {"dstAddr":63305,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)')
```

Full logs here:
https://gist.github.com/Woet/a5b9a690b69abcc4020483291b8c7662

Can we please get some feedback or at least some pointers in how to help resolve this issue?

GSzabados, 2021-05-31:
Anyone who follows this topic. Osram bulbs are famous about their terrible firmware. And there is no real solution. They cannot route messages. That is a fact.

Please see the comments here:

https://community.hubitat.com/t/sylvania-lightify-br30-bulb-becomes-unresponsive-with-24-h/14521

https://community.hubitat.com/t/zigbee-issues/26481/99

If you just Google _Osram zigbee bulb issue_, you will see hundreds of similar posts.

Usually the  symptoms are, bulb doesn't turn on/not responding, bulb turns on by itself.

The best course of action, avoid Osram bulbs, use instead IKEA ones which has a far more better firmware and still supported, updated.

If it works better with deCONZ, then try to look at the issues on GitHub, closed ones as well, what people have experienced and what was the fix for it.

hellcry37, 2021-05-31:
laugh of the day: "use instead IKEA ones which has a far more better firmware" 

Do NOT do that!
- https://www.reddit.com/r/tradfri/comments/au903n/firmware_bugs_in_ikea_bulbs/

Ikea has an infamous bug not solved in years where routers (zigbee powered devices) fail to route any traffic and the only way you can make them work again is to power cycle them. This way you always loose connectivity with some of your end devices.

GSzabados, 2021-05-31:
> Ikea has an infamous bug not solved in years where routers (zigbee powered devices) fail to route any traffic and the only way you can make them work again is to power cycle them. This way you always loose connectivity with some of your end devices.

Can you post a reference link to this infamous bug, because the one you posted has no connection with that?

And just for the argument, Osram/Lightify is pulling out from the Zigbee market, and shuts down all the servers in a few months. There will be no updates anymore to any of the bulbs, led strips or sockets.

https://www.osram-group.com/en/media/press-releases/pr-2020/09-03-2020

IKEA is still supporting all of its device and they are far more superior compared to the Osram ones.

I have thrown away plenty Osram bulbs, because they just stop working from one day to another without any change. I haven't experienced that with any IKEA bulbs.(Neither this infamous bug.)

MattWestb, 2021-06-03:
The big different is the platform that the manufacture is using and the zigbee standard implanted.
OSRRAM was (perhaps is) using old very week CC-2530 chips and most home devices is ZLL or HA devices and they cant upgrading  to Zigbee 3 then the chips is to week and the chip manufacture have no active support for the chips.

IKEA and most large manufactures is using Silabs EFR32 first gen chips that is running Zigbee 3 without and problems (if not having bugs).
The first gen IKEA devices was released with ZLL and all have being upgraded to ZB3 with exception the old motion sensor and the CWS1 that have getting one update in September but is still ZLL.

The "IKEA BUG" that i was the first to pinpointing is not one IKEA elusive its hitting Sonoff, Ewlink, Smartthings, New Philips HUE, tuya and so on = most devices that normal HA user is having but not Xiaomi and old OSRAM but both have much worse problems in there firmware and shall being avoided if its possible (SED not changing parent if its not online, corrupted packages, not aging out children and so on).

The new third gen IKEA devices is very likely getting firmware that is fixed then  they is having one EFR32 second gen chip with enough flash and RAM for running EZSP 6.7.8.0 and newer ziggbee stacks but its very unlikely happens with all first and second gen devices but you is newer knowing with the crazy swedes.

Also to triggering the bug is normally not being done if not re power router devices and large network with over 100 devices is normally running stable until re power router devices.

In the end IKEA have getting one off the the most zigbee 3  compatible devices on the market that also is working well in most systems.

Compared with Philip HUE that is not implanting many zigbee functions like attribute reporting that is making them braking the zigbee comparability but is working very well in the native HUE systems.

If like having one stable network IKEA routers is working very well with most devices and avoiding old OSRAM and Philips HUE devices you can getting all Xiaomi sensors working very well.

Jhonattan-Souza, 2021-06-30:
@Woet There are two friends with this issue, with differente zigbee devices, 'Mack no Ack' , I think that is a problem with ZZH , There's no other reason

MattWestb, 2021-07-01:
Old OSRAM is famous for not doing the 15.4 Acking OK and some sensors is having problem and leaving the network because of that.

But if having good routers around its possible going around the bad devices but its still can getting problems with routing unicast around the mesh but normally broadcasts is working OK then they is not using 15.4 Acks.

Woet, 2021-08-01:
Still happening:
```
Zigbee2MQTT:error 2021-07-27 10:26:54: Publish 'set' 'brightness' to 'Bedroom bedside' failed: 'Error: Command 0x7cb03eaa00a96839/3 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
Zigbee2MQTT:error 2021-07-27 10:38:43: Publish 'set' 'color_temp' to 'Bedroom bedside' failed: 'Error: Command 0x7cb03eaa00a96839/3 lightingColorCtrl.moveToColorTemp({"colortemp":196,"transtime":10}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
Zigbee2MQTT:error 2021-07-27 10:39:53: Publish 'set' 'brightness' to 'Bedroom bedside' failed: 'Error: Command 0x7cb03eaa00a96839/3 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
```

MattWestb, 2021-08-01:
@Woet Do you have trying making light groups for the lights and sending group commands (its broadcast and not using IEEE Ack and can going around bad router with little luck but is limited to 8 broadcast in 9 seconds or the mesh is ignoring the commands). 
Broadcast is the preferred method sending commands to light devices and usincast to / from "HA" devices and have good and bad things but is working good mixed in the system.

closed 2021-09-08 by github-actions[bot]

GSzabados, 2022-01-25:
> Ikea has an infamous bug not solved in years where routers (zigbee powered devices) fail to route any traffic and the only way you can make them work again is to power cycle them. This way you always loose connectivity with some of your end devices.

I wanted to let you know that I think IKEA has probably fixed another ongoing crash issue as well in the latest firmware releases:

https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html

`Solves an issue where some products become unreachable after a gateway restart.`

MattWestb, 2022-01-25:
I thinking this was little interesting:
- Upgrade to stack v6.3.1

I was hoping on EZSP / Gecko SDK 6.7.10,x that is the last OK for there old 1st and 2ond gen devices.

The problem IKEA is "tuning" ther firmware for working in Light Link mode ao is not having one coordinator so they is not seeing the bugs we is finding in trustcenter setup systems.

The "old" firmware versions (february 2021) that we have getting from the OTA files can being fund [here](https://github.com/zigpy/zigpy/discussions/660#discussioncomment-358222) but we cant getting it from the new GBL format.

Still good that they have updating 95% of there devices firmware in 12 months and still OSRAM outdoor plug is not working with end devices as children or that the communications going thru them and they is corrupting packages and end device is leaving the network.