Koenkk/zigbee2mqtt #6986: Zigbee Stick C2531 crashing with zigbee2mqtt @ Home Assistant

joefernwright, 2021-04-07:
## What happened
Zigbee nodes configured in my Home Assistant stopped reporting their states. They are not communicative (HA does not receive messages from them and they do not react on sent requests).  

## What did you expect to happen
I am expecting zigbee2mqtt to be robust and do not crash.

## How to reproduce it (minimal and precise)
Difficult to say, I have a suspicious that the problem is somehow caused by Blitzwolf BW-SHP13 in my network but I am not sure. The device is normally recognized by Home Assistant and works fine for some time already. 

## Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20201127.zip.

##Reported errors:
Zigbee2MQTT:error 2021-04-07 16:27:41: Publish 'set' 'state' to 'BlitzWolf Socket K1' failed: 'Error: Command 0x588e81fffed38fb4/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
Zigbee2MQTT:info  2021-04-07 16:27:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'BlitzWolf Socket K1' failed: 'Error: Command 0x588e81fffed38fb4/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"BlitzWolf Socket K1"},"type":"zigbee_publish_error"}'
(node:328) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
(node:328) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 21)

westerncanada, 2021-04-07:
I'm seeing something similar recently, I don't have that device in my network.  My symptoms are otherwise identical. HA just stops getting updates, so I check the zigbee2mqtt log and it has in fact just "not seen" any updates for a number of hours.
I suspected it was a bad controller so I replaced it with a spare, and flashed the new one with the latest firmware, and just yesterday tried the source routing firmware, however that also crashed over night.
running dmesg didn't show the device disconnected, and restarting the service brought everything back on line this time, but generally doesn't (with the older firmware 20190608)

Last messages in the log:
info  2021-04-06 20:51:35: MQTT publish: topic 'zigbee2mqtt/0x00158d0003f4c07e', payload '{"battery":28,"humidity":39.71,"linkquality":65,"pressure":927,"temperature":1.52,"voltage":2805}'
info  2021-04-06 20:51:35: MQTT publish: topic 'zigbee2mqtt/0x00158d0003f4c07e', payload '{"battery":28,"humidity":40.45,"linkquality":68,"pressure":927,"temperature":1.52,"voltage":2805}'
info  2021-04-06 20:51:35: MQTT publish: topic 'zigbee2mqtt/0x00158d0003f4c07e', payload '{"battery":28,"humidity":40.45,"linkquality":76,"pressure":927,"temperature":1.52,"voltage":2805}'
info  2021-04-06 20:51:37: MQTT publish: topic 'zigbee2mqtt/LaundryRoom-96cf', payload '{"led":false,"led_state":false,"linkquality":68,"state":true}'
info  2021-04-06 20:51:40: MQTT publish: topic 'zigbee2mqtt/0x00124b001f8ae54f', payload '{"linkquality":65,"state":"ON"}'

Startup messages
info  2021-04-07 08:01:04: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2021-04-07 08:01:04: Disconnecting from MQTT server
info  2021-04-07 08:01:05: zigbee-herdsman stopped
info  2021-04-07 08:01:17: Logging to console and directory: 'data/log' filename: log.txt
info  2021-04-07 08:01:17: Starting Zigbee2MQTT version 1.18.1 (commit #b6085fc)
info  2021-04-07 08:01:17: Starting zigbee-herdsman (0.13.71)
info  2021-04-07 08:01:19: zigbee-herdsman started
info  2021-04-07 08:01:19: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201128,"transportrev":2},"type":"zStack12"}'
info  2021-04-07 08:01:19: Currently 26 devices are joined:

Koenkk, 2021-04-07:
You are probably reaching the limits of what the CC2531 can handle (it does not have very powerful hardware). The source routing firmware might provide some better performance: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing , if not I recommend upgrading to a more powerful adapter: https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter

westerncanada, 2021-04-07:
Something that might just be my reading comprehension of the "what needs a re-pairing of devices" I'm not sure if upgrading from the CC2531 I'm running now,  to a board I have here "CC2530 with a CC2592" or a "Sonoff Zigbee bridge flashed with tasmota" will need a re-pairing?
Thanks for taking the time to respond, I love this software and greatly appreciate your work.

Koenkk, 2021-04-07:
Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#what-does-and-does-not-require-repairing-of-all-devices ?

westerncanada, 2021-04-07:
Yes, this is the part that confused me:
Doesn’t require repairing
Switching from CC2530/CC2531 device (physically).
which to me reads like if you have one of these devices and switch to anything else, you don't have to.
if I'm reading that right, then one could potentially start with the 30/31 device and then switch to any other device which doesn't make sense to me? Which is why I questioned it.

Koenkk, 2021-04-07:
I see, it means switching from 1 CC2531/CC2530 to another CC2531/CC2530 doesn't require reparing devices. But note that the CC2531 and CC2530 are identical performance wise so they won't solve your stability problems.

westerncanada, 2021-04-07:
Got it, I'll have to find a better unit in stock and do the re-pairing that I've been dreading.  Thanks again.

joefernwright, 2021-04-07:
> You are probably reaching the limits of what the CC2531 can handle (it does not have very powerful hardware). The source routing firmware might provide some better performance: [...]

I have the following Zigbee devices paired with C2531 in my HA:
2 x Sonoff SNZB-02 (Temp&Humid)
1 x Sonoff SNZB-03 (PIR)
1x Sonoff SNZB-01 (push button)
1 x Sonoff BASICZBR3 (switch,  only paired, recently completely disconnected)
1 x BlitzWolf TS0121_plug

So six devices in total, not very talkative devices. Am I reaching the hardware limits of C2531 with so small number of hardware units not used extensively? Hmmm...it does not sound right...so is C2531 supported or not? 

Do not get me wrong, I highly value your work but I firmly believe that proposing to upgrade to more powerful hardware is only a half-backed solution to avoid a bug or software design flaw which may any time manifest itself in the "challenging" conditions.

I had more crashes with the previous firmware for C2531 (this one from 2019) and I was hoping that flashing this November 2020 version will solve all the issues but now I see that this whole zigbee2mqtt is not the most reliable solution. We use it at home to toggle a little lamp in a baby room with use of a portable push button. When my wife swears under her nose when it's not working as expected I just feel bad and ask myself if this whole solution is just a toy or something at least half reliable reliable.

I believe that hardware limits which lead to errors should be always somehow handled, There must be some sort of solution like watchdog or forced full restart of firmware in teh coordinator when something goes wrong. Discovering from time to time that sensors cannot send their data for a dozen of hours is just very annoying :)

GSzabados, 2021-04-08:
Does any of you before the missing state updates has restarted Z2M from the frontend, not from the Supervisor menu?

There is something similar in this issue, a Blitzwolf plug stops reporting power, current and voltage when Z2M is restarted from the frontend.

https://github.com/Koenkk/zigbee2mqtt/issues/6713

nopap, 2021-04-08:
Hello,
Similar issue here:
I have been running for nearly a year with no issues on Zigbee2MQTT 1.13, with CC2531 and firmware 20190608.
Yesterday i upgraded the CC2531 firmware to 20201127 and Zigbee2MQTT to 1.18.2, and after 24h, crash with similar errors (see below).
A Pi reboot fixed the issue for now.
I have zzh CC2652R on the way, but until then, as i did not add new devices i wonder if just putting back firmware 20190608 would be enough ?

```
Apr  8 17:51:31 zigbee2mqtt npm[1286]: (node:1301) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
Apr  8 17:51:31 zigbee2mqtt npm[1286]:     at Timeout.waiter.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/util
s/waitress.js:67:35)
Apr  8 17:51:31 zigbee2mqtt npm[1286]:     at ontimeout (timers.js:436:11)
Apr  8 17:51:31 zigbee2mqtt npm[1286]:     at tryOnTimeout (timers.js:300:5)
Apr  8 17:51:31 zigbee2mqtt npm[1286]:     at listOnTimeout (timers.js:263:5)
Apr  8 17:51:31 zigbee2mqtt npm[1286]:     at Timer.processTimers (timers.js:223:10)
Apr  8 17:51:31 zigbee2mqtt npm[1286]: (node:1301) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by thro
wing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
```

Koenkk, 2021-04-08:
In case you are running with only a few devices on the network and you get crashes (even with the source routing firmware, https://github.com/Koenkk/zigbee2mqtt/issues/6986#issuecomment-815071843). Please provide the herdsman debug logging from starting z2m till the crash.

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

nopap, 2021-04-11:
Well i have 14 devices, not sure if it's _few_ , but including 2 cc2531 routers.
I noticed also router issues after the upgrade (router lost and disappearing), so I downgraded back the firmware on all 3 cc2531 keys to the previous versions (20190608) but kept latest zigbee2mqtt 1.18.2 and all is back to normal.
I guess the firmware updates maybe eat a bit more memory and i reach the limit with  my 14 devices.

I'll retry the firmware upgrade and herdsman debug log later, but as my zzh! are on their way maybe might not be worth the effort.

joefernwright, 2021-05-01:
I replaced my Blizwolf plug with Sonoff Zigbee switch. No improvement. Still, from time to time it stops working:
 `    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
(node:525) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 32)
(node:525) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
(node:525) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 33)`

mulyonost, 2021-05-18:
Hi, I have same issue with you, did your problem has been solved ?
I only ran 10 devices on my network.

joefernwright, 2021-05-21:
I can confirm that the source routing firmware performs much better. No issues for a few weeks already:
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

Friendly0Fire, 2021-05-26:
I also seem to be having a similar issue. I've always had occasional Z2M freezes that required a restart (Dockerized, so fairly quick), but the past few days have gotten unbearably spotty (less than 24h uptime). I've added quite a few new zigbee devices, but I'm still below 10 and I'm using a zzh so I'd expect it to be fine?

Unfortunately, since there's no source routing firmware for the CC2652R, I can't test that. I'll enable herdsman debugging and see if I can get additional insights.

closed 2021-07-04 by github-actions[bot]