Koenkk/zigbee2mqtt #6987: Template rendering warnings in latest Home Assistant beta for dictionary missing attribute

mdegat01, 2021-04-07:
## What happened

Starting in Home Assistant `2021.4.0` templates referencing an attribute of a dictionary which does not exist generate an warnings like this:

```
2021-04-07 15:05:31 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'action' when rendering '{{ value_json.action }}'
```

I traced these errors back to the source and its a result of the discovery payload sent by Z2M to Home Assistant. The discovery payload sends this as the template to use to parse the value for a sensor out of the JSON payload:

```
value_template: '{{ value_json.action }}'
```

For devices which don't report all their attributes all the time (ex. remotes that report an action on button press) this creates a problem. Anytime a payload is received without the attribute a warning is created. This warning will become an error and a failure starting in `2021.10` (see https://github.com/home-assistant/core/pull/48140 ) 

Fortunately there appears to be a relatively easy fix. If the discovery template is changed to this then all the errors should go away:

```
{{ value_json.action | default('', true) }}
```

It's a bit strange to always default to `''` even for numeric attributes but this is what was occurring before. Returning `None` as the default also works and does not generate a warning (I tested). That might be more correct but not sure if that has any other consequences downstream. The `true` is so it only appllies the default if things truly don't exist and not just if they evaluate to `false` (see [default](https://jinja.palletsprojects.com/en/master/templates/#default)).

## What did you expect to happen

Not to see a warning for attribute did not exist

## How to reproduce it (minimal and precise)

The error above is showing what can occur if you have a remote type device such as [Philips 929002398602](https://www.zigbee2mqtt.io/devices/929002398602.html) that checks in all the time with basic attributes like battery and linkquality and includes an action on button press. You can also create an error like this with any device by doing the following:

1. Proceed with Home Assistant discovery for the device
2. Use `filtered_attributes` to tell Z2M to stop reporting one or more attributes
3. See you get warnings for the sensors trying to read an attribute that is no longer being reported in

This creates errors like this:

```
2021-04-06 21:24:44 | 2021-04-06 21:24:44 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: dict object has no attribute voltage
```

I realize in this particular use case the right fix is just to disable the sensor in Home Assistant (which is what I did). But I don't know all the devices that report some attributes occaisionally, figured it would be good to have a way to simulate that with any device.

## Debug info
Zigbee2MQTT version: 1.18.2
Home Assistant version: 2021.4.0b5
Adapter hardware: Conbee II
Adapter firmware version: 0x264a0700

mdegat01, 2021-04-07:
> After this does: https://www.zigbee2mqtt.io/integration/home_assistant.html#via-home-assistant-entity still work? If yes, please create a PR.

Made one already actually, its linked above (#6988 ). I don't really see why it wouldn't, the PR is just changing how the state of the sensor gets set somewhat to prevent errors. That type of trigger is just looking for state change events so its downstream of that part.

I don't have an easy way to do a mock test of a dev build of Z2M unfortunately since I don't have the spare devices I would need. Later I can try faking it directly using MQTT if need be though. Send discovery messages identical to what Z2M sends and then fake state change events. Will have to do some digging to come up with the mocks though.

tteck, 2021-04-07:
Same here. `Template variable warning: 'dict object' has no attribute`

tteck, 2021-04-08:
Still coming in:
`Logger: homeassistant.helpers.template
Source: helpers/template.py:1335
First occurred: 8:06:00 PM (63 occurrences)
Last logged: 8:06:05 PM

Template variable warning: 'dict object' has no attribute 'action' when rendering '{{ value_json.action }}'
Template variable warning: 'dict object' has no attribute 'sensitivity' when rendering '{{ value_json.sensitivity }}'
Template variable warning: 'dict object' has no attribute 'linkquality' when rendering '{{ value_json.linkquality }}'
Template variable warning: 'dict object' has no attribute 'temperature' when rendering '{{ value_json.temperature }}'
`

Caroga, 2021-04-08:
I'm curious to any specific zigbee2mqtt configuration. Do you have any (device specific) configuration about using the legacy api or anything?

tteck, 2021-04-08:
> I'm curious to any specific zigbee2mqtt configuration. Do you have any (device specific) configuration about using the legacy api or anything?

Myself, just a basic/vanilla configuration.

1igeza, 2021-04-08:
I'm using on device like motion sensors, door sensors
```
    filtered_attributes:
      - temperature
```
and have the same error. If I disable in HA entity sensor "temperature" in those devices, then there is no error. So I think the problem is sending to HA information for blocked sensors for example: 
`{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Rafal Czujnik Ruchu/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000302fd05"],"manufacturer":"Xiaomi","model":"Aqara human body movement and illuminance sensor (RTCGQ11LM)","name":"Rafal Czujnik Ruchu","sw_version":"Zigbee2MQTT 1.18.2-dev"},"device_class":"temperature","json_attributes_topic":"zigbee2mqtt/Rafal Czujnik Ruchu","name":"Rafal Czujnik Ruchu temperature","state_topic":"zigbee2mqtt/Rafal Czujnik Ruchu","unique_id":"0x00158d000302fd05_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}`

McGiverGim, 2021-04-08:
It seems is being produced by some attributes that for some reason "don't have value". For example, a Xiaomi relay:

>Template variable warning: 'dict object' has no attribute 'interlock' when rendering '{{ value_json.interlock }}'

[image]

If I press `false` in the UI, then it has a value and the error disappears from logs.

Mr-Groch, 2021-04-09:
There is also have problem with [Neo NAS-AB02B0 Siren Alarm](https://www.zigbee2mqtt.io/devices/NAS-AB02B0.html#neo-nas-ab02b0):

```
Template variable warning: 'dict object' has no attribute 'duration' when rendering '{{ value_json.duration }}'
Template variable warning: 'dict object' has no attribute 'temperature_min' when rendering '{{ value_json.temperature_min }}'
Template variable warning: 'dict object' has no attribute 'temperature_max' when rendering '{{ value_json.temperature_max }}'
Template variable warning: 'dict object' has no attribute 'humidity_min' when rendering '{{ value_json.humidity_min }}'
Template variable warning: 'dict object' has no attribute 'humidity_max' when rendering '{{ value_json.humidity_max }}'
```

Koenkk, 2021-04-12:
Fixed now in the dev branch

closed 2021-04-12 by Koenkk

ikke-zelf, 2021-04-15:
> 
> 
> Fixed now in the dev branch

@Koenkk ; Can't seem to find how often you make official release. in other words; When will this issue also be available in the stable branch? So I can benefit from it in my HA installation.

mdegat01, 2021-04-15:
If you look in the releases list for this repo there's a new minor release on the first of the month every month for the past 6 months like clockwork. So I would guess May 1st unless Koenkk decides this issue or some other recent issue is serious enough to warrant a patch.

Koenkk, 2021-04-17:
Next release will be 1 may indeed.

jacekpaszkowski, 2021-04-22:
Hello,

i'm running dev version and still get this kind of warnings in HA, i think for my Tuya/Moes thermostats:

```
2021-04-22 21:42:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: dict object has no element Undefined when rendering '{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}'
2021-04-22 21:42:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: dict object has no element Undefined when rendering '{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}'
2021-04-22 21:42:12 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: dict object has no element Undefined when rendering '{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}'
2021-04-22 21:42:22 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: dict object has no element Undefined when rendering '{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}'

```

Any chance to fix that?

Koenkk, 2021-04-23:
@jacekpaszkowski can you try changing https://github.com/Koenkk/zigbee2mqtt/blob/abd8a0926403d2b6d6c9a1a175f917262926c72f/lib/extension/homeassistant.js#L182 with "` %}{{ values[value_json.${state.property}] | default('idle', true) }}`;"

jacekpaszkowski, 2021-04-25:
@Koenkk i'm running zigbee2mqtt as HA add-on, i don't know how to change this file :(

Koenkk, 2021-04-26:
SSH into your machine, go into docker container by executing `docker exec -it CONTAINER_ID /bin/bash`, `nano lib/extension/homeassistant.js`, change line, exit container and restart z2m with `docker restart CONTAINER_ID`.

kaykoch, 2021-04-26:
If I understand well, this will be fixed on may.
I added:
```
logger:
  default: ERROR
```
in configuration.yaml. 
Kay

playersct, 2021-04-29:
hello @Koenkk 

> SSH into your machine, go into docker container by executing `docker exec -it CONTAINER_ID /bin/bash`, `nano lib/extension/homeassistant.js`, change line, exit container and restart z2m with `docker restart CONTAINER_ID`.

in dev version nano not work, only vi  and i have homeassistant.js in ./app/lib/extension/
It's ok?

Koenkk, 2021-04-30:
@playersct due to code changes line numbers may vary, but there is only one occurrence of "` %}{{ values[value_json.${state.property}] }}`"

dylan09, 2021-05-10:
Is there anything I have do change in configuration after update to 1.18.3?

After I have updated my Z2M Docker image to 1.18.3 i have still a lot of "template warning" messages in my home assistant log file.
I think most of them are related to my Danfoss TVR.

Here a few lines from HA log:

```
2021-05-10 18:08:15 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'setpoint_change_source' when rendering '{{ value_json.setpoint_change_source }}'
2021-05-10 18:08:15 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'keypad_lockout' when rendering '{{ value_json.keypad_lockout }}'
2021-05-10 18:08:15 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'window_open_external' when rendering '{{ value_json.window_open_external }}'
2021-05-10 18:08:15 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'heat_available' when rendering '{{ value_json.heat_available }}'
2021-05-10 18:08:15 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'day_of_week' when rendering '{{ value_json.day_of_week }}'
2021-05-10 18:08:15 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'trigger_time' when rendering '{{ value_json.trigger_time }}'
2021-05-10 18:08:15 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'algorithm_scale_factor' when rendering '{{ value_json.algorithm_scale_factor }}'
```

kaykoch, 2021-05-11:
I got the same Warnings, as I descriped here: 
- https://github.com/Koenkk/zigbee2mqtt/issues/6995#issuecomment-831234263
- https://github.com/Koenkk/zigbee2mqtt/issues/4571#issuecomment-829948789

I guess following: (Z2M: zigbee2mqtt Add-On)

1. HA throws Warnings for a missing variable since April (?) Update.
2. Z2M does not send all possible values (no default values)
3. Z2M does not send some variables any more (Eurotronic SPZB0001, "system_mode")
4. If you set these values in states.json, you have default values, so no Warning. But no Update (->3)

Maybe, someone can confirm these ideas.

Nephiel, 2021-05-17:
> Is there anything I have do change in configuration after update to 1.18.3?
>
> After I have updated my Z2M Docker image to 1.18.3 i have still a lot of "template warning" messages in my home assistant [...]

Same here on 1.18.3 (commit #669489fe). HA complains of:
```
Template variable warning: 'dict object' has no attribute 'update_available' when rendering '{{ value_json.update_available}}'
Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'
Template variable warning: 'dict object' has no attribute 'action' when rendering '{{ value_json.action }}'
```
Relevant options in my `configuration.yaml` are:
```
homeassistant:true

advanced:
  homeassistant_legacy_triggers: false
  legacy_api: false

device_options:
  legacy: false
```

Already cleared persistence on Mosquitto and restarted it, but even so I still get those messages.

Any pointers?

Nephiel, 2021-05-17:
Nevermind, it turns out I had _another_ Z2M instance connected to the same Mosquitto server, and _that_ one was not updated to 1.18.3, hence the `click` and `action` errors.

The only one I still have is `update_available`:
```
Template variable warning: 'dict object' has no attribute 'update_available' when rendering '{{ value_json.update_available}}'
```
I think it is related to OSRAM plugs and bulbs.

Xitro01, 2021-06-11:
Hi, it's june now and I have everything up-to-date and still see those errors coming by:

Logger: homeassistant.helpers.template
Source: helpers/template.py:1366
First occurred: 1:00:29 (89 occurrences)
Last logged: 8:42:28

Template variable warning: 'dict object' has no attribute 'requested_brightness_level' when rendering '{{ value_json.requested_brightness_level }}'
Template variable warning: 'dict object' has no attribute 'requested_brightness_percent' when rendering '{{ value_json.requested_brightness_percent }}'
Template variable warning: 'dict object' has no attribute 'sensitivity' when rendering '{{ value_json.sensitivity }}'

Electronlibre2012, 2021-06-15:
Hello,

same here
```
Template variable warning: 'dict object' has no attribute 'tamper' when rendering '{{ value_json.tamper }}'
Template variable warning: 'dict object' has no attribute 'sensitivity' when rendering '{{ value_json.sensitivity }}'
Template variable warning: 'dict object' has no attribute 'power_outage_memory' when rendering '{{ value_json.power_outage_memory }}'
Template variable warning: 'dict object' has no attribute 'operation_mode' when rendering '{{ value_json.operation_mode }}'
Template variable warning: 'dict object' has no attribute 'voltage' when rendering '{{ value_json.voltage }}'
```
no changes with the last release of june

Gollam, 2021-06-20:
Hi, same for me.

mdegat01, 2021-06-20:
> Hello,
>
> same here
>
> ```
> Template variable warning: 'dict object' has no attribute 'tamper' when rendering '{{ value_json.tamper }}'
> Template variable warning: 'dict object' has no attribute [...]

If you are seeing this error that means you have a sensor in HA created by Z2M for a device. However that device is not providing any data for that particular sensor. 

If you're OK with the sensor having status unknown because you don't care about it then there's a very simple solution - disable the sensor in HA. It will then stop trying to parse the json looking for data which does not exist and reporting `unknown` for a sensor you don't care about.

If you do care about this sensor because perhaps it is not unknown all the time and reports an actual value occasionally then please provide more info. For example, what device is it and what causes it to report in an actual value for that particular attribute?

This warning (not error) is designed to alert you to issues already occurring that you didn't know about. Like having a sensor which always has value `unknown`. But there are tons of supported devices in Z2M so if one type is behaving badly under these changes please provide that info in order to get it addressed.

SaturnusDJ, 2021-06-25:
Got this issue. Error: `Template variable warning: 'dict object' has no attribute 'action' when rendering '{{ value_json.action }}'`
Relevant devices: WXKG11LM and MFKZQ01LM.
Z2M: 1.19.1 (restarted)
HA: 2021.5.5 (restarted)
Mosquitto: 1.6.14 (restarted)

Possible relevant Z2M configuration options:
```
advanced:
  homeassistant_legacy_triggers: false
#  legacy_api: false
device_options:
  legacy: false
```
Need the legacy api for https://github.com/rgruebel/ha_zigbee2mqtt_networkmap
But it looked like disabling it did not make a difference anyways.

HA MQTT entity:
```
  - platform: "mqtt"
    name: "Button hallway"
    state_topic: "zigbee2mqtt/button_hallway"
    availability_topic: "zigbee2mqtt/bridge/state"
    value_template: "{{ value_json.action }}"
    json_attributes_topic: "zigbee2mqtt/button_hallway"
```
This is not used in automation because it cannot trigger when the previous action occurs again (e.g. pressing a button a single time, then another single click, the last single click does not trigger). I use it to log the button usage to InfluxDB.

HA automation:
```
- alias: Button hallway single click > Hallway lamp on 5 min
  mode: restart
  trigger:
  - platform: mqtt
    topic: zigbee2mqtt/button_hallway
  condition:
  - "{{ trigger.payload_json.action == 'single' }}"
```
This triggers on any incoming message, even if it is the same as the previous one.

Looking at the error I think it relates to the 'HA MQTT entity' part. In MQTT Explorer I see the action attribute is only supplied when a user uses the button. I thought the Z2M code changes should result in the action attributes to be present always, so that the workaround on HA side (topic start) is not needed anymore, or am I wrong?

Bojan023, 2021-07-07:
> @jacekpaszkowski can you try changing
> 
> https://github.com/Koenkk/zigbee2mqtt/blob/abd8a0926403d2b6d6c9a1a175f917262926c72f/lib/extension/homeassistant.js#L182
> with `` ` %}{{ values[value_json.${state.property}] | default('idle', true) }}\`; ``

Still raised and error in my case. For completeness, raised error:

`Template variable warning: dict object has no element Undefined when rendering '{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}'`

homeassistant.js:

`            const state = firstExpose.features.find((f) => f.name === 'running_state');
            if (state) {
                discoveryEntry.discovery_payload.action_topic = true;
                discoveryEntry.discovery_payload.action_template = `{% set values = ` +
                        `{'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'}` +
                        ` %}{{ values[value_json.${state.property}] | default('idle', true) }}`;
            }
`

``

Bojan023, 2021-07-07:
> @jacekpaszkowski can you try changing
> 
> https://github.com/Koenkk/zigbee2mqtt/blob/abd8a0926403d2b6d6c9a1a175f917262926c72f/lib/extension/homeassistant.js#L182
> with `` ` %}{{ values[value_json.${state.property}] | default('idle', true) }}\`; ``

Have tested this. This removed the error from HA. PR coming

dylan09, 2021-07-20:
Hello,

I discovered this error/message with the latest Zigbee2MQTT version 1.20.0 (commit #41b67fd).
Device is a Danfoss Ally eTRV0100 Modell 014G2461

Z2M Logs show following message:

```
2021-07-20 20:47:21 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'system_mode' when rendering '{{ value_json.system_mode }}'
2021-07-20 20:47:21 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: 
```
And every time I set the operation mode to "Heat" in Home Assistant, it will be reset to nothing after a short while.

What could I do to get rid off this log messages?

dylan09, 2021-07-26:
Maybe anyone could help me with my above problem?
I have deleted the device from zigbee2mqtt and from Home Assistant. After pairing device again error still persists. 
Tried ```legacy=false``` and ```legacy=true```. Nothing changed. Get the above log message every few seconds.

Is there a fix I could add to ```lib/extension/homeassistant.js```?

And here published topic as it is logged by zigbee2mqtt:
```
MQTT publish: topic 'zigbee2mqtt/bu_cl_heizung', payload '{"algorithm_scale_factor":null,"battery":53,"day_of_week":null,"heat_available":null,"heat_required":false,"keypad_lockout":null,"linkquality":191,"load_estimate":null,"local_temperature":19.88,"mounted_mode_active":false,"mounted_mode_control":null,"occupied_heating_setpoint":18,"pi_heating_demand":0,"setpoint_change_source":null,"thermostat_vertical_orientation":null,"trigger_time":null,"update":{"state":"idle"},"update_available":false,"viewing_direction":null,"window_open_external":null,"window_open_internal":"closing"}'
```

3gyptian, 2021-07-30:
I too was also to resolve  these same warnings by following the above suggestion.

Warning message:

```
[homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'action' when rendering '{{ value_json.action }}'
```

Resolved by modifying zigbee2mqtt/lib/extension/homeassistant.js and replacing the line:
```
` %}{{ values[value_json.${state.property}] }}`; 
```
with
```
` %}{{ values[value_json.${state.property}] | default('idle', true) }}\`;
```

Thanks for everyone's posts in this issue

Altycoder, 2021-08-04:
Still an issue:

```
2021-08-04 08:12:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'mappingproxy object' has no attribute 'min_value' when rendering '{{states.sensor.battery_levels.attributes.min_value}}',
2021-08-04 08:12:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'mappingproxy object' has no attribute 'min_value' when rendering '{{states.sensor.battery_levels.attributes.min_value}}',
2021-08-04 08:12:41 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Low Battery' trigger: In 'numeric_state' condition: entity sensor.min_battery_level state '' cannot be processed as a number,
2021-08-04 08:12:43 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ value_json.value }}',
2021-08-04 08:12:43 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ value_json.value }}',
2021-08-04 08:12:43 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ value_json.value }}',
2021-08-04 08:12:43 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ value_json.value }}',
2021-08-04 08:12:43 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ {0: "idle",2: "Overheat detected",6: "Underheat detected"}[value_json.value] | default(value_json.value) }}',
2021-08-04 08:12:43 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ value_json.value }}',
2021-08-04 08:12:43 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ value_json.value }}',
2021-08-04 08:12:43 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ value_json.value }}',
2021-08-04 08:12:43 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ value_json.value }}',
2021-08-04 08:12:43 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ value_json.value }}',
2021-08-04 08:12:43 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ value_json.value }}',
2021-08-04 08:12:43 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ {0: "idle",2: "Overheat detected",6: "Underheat detected"}[value_json.value] | default(value_json.value) }}'
```

Andoramb, 2021-08-06:
I also got a lot of

`
WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'power_status' when rendering '{{value_json.power_status}}'`

maybe that falls into the same category?

Yurhik, 2021-08-21:
Z2M  1.21.0-4. 
The warning remained:
Logger: homeassistant.helpers.template
Source: helpers/template.py:1366
First occurred: 20 августа 2021 г., 20:39:35 (3842 occurrences)
Last logged: 10:04:16

Template variable warning: dict object has no element Undefined when rendering '{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}'
Template variable warning: 'dict object' has no attribute 'update_available' when rendering '{{ value_json.update_available}}'

wishhdd, 2021-09-03:
@Yurhik the same problem.
How to treat? Help please.

Altycoder, 2021-09-03:
Still an issue for me in the latest September build on docker

gomble, 2021-09-30:
i have also the problem:

Logger: homeassistant.helpers.template
Source: helpers/template.py:1485
First occurred: 27. September 2021, 0:56:08 (1381 occurrences)
Last logged: 27. September 2021, 6:54:47

Template variable warning: 'dict object' has no attribute 'valve_detection' when rendering '{{ value_json.valve_detection }}'
Template variable warning: dict object has no element Undefined when rendering '{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}'
Template variable warning: 'dict object' has no attribute 'operation_mode' when rendering '{{ value_json.operation_mode }}'
Template variable warning: 'dict object' has no attribute 'action' when rendering '{{ value_json.action }}'
Template variable warning: 'dict object' has no attribute 'linkquality' when rendering '{{ value_json.linkquality }}'

Altycoder, 2021-10-19:
I'm still having problems too:

2021-10-19 08:46:35 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'mappingproxy object' has no attribute 'min_value' when rendering '{{states.sensor.battery_levels.attributes.min_value}}',

2021-10-19 08:46:38 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ value_json.value }}'

2021-10-19 08:46:38 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'value' when rendering '{{ {0: "idle",2: "Overheat detected",6: "Underheat detected"}[value_json.value] | default(value_json.value) }}',

SaturnusDJ, 2021-11-06:
Still got it in 1.22.0 as in https://github.com/Koenkk/zigbee2mqtt/issues/6987#issuecomment-868620428

Did I miss something?

adekloet, 2021-12-25:
core 12.5
Z2MQTT Current version: 1.22.1-1

Still has the problem with errors in the log for some devices

2021-12-25 10:31:43 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.battery }}'
2021-12-25 10:31:43 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.occupancy }}'
2021-12-25 10:31:43 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.occupancy }})
2021-12-25 10:31:43 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.illuminance_above_threshold }}'
2021-12-25 10:31:43 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.illuminance_above_threshold }})
2021-12-25 10:31:43 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'value_json' is undefined (value: , template: {{ value_json['update']['state'] == "available" }})
2021-12-25 10:32:11 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: access to attribute 'update' of 'dict' object is unsafe. (value: {"battery":100,"illuminance_above_threshold":false

Flop2006, 2022-01-15:
probably the same error here with a new added Sinterwell GS361A-H04 in Homeassistant:

After connection Home Assistant shows some errors:
`Invalid ['off', 'heating', 'cooling', 'drying', 'idle', 'fan'] action: None`
also
```
Template variable warning: 'dict object' has no attribute 'local_temperature' when rendering '{{ value_json.local_temperature }}'
Template variable warning: 'dict object' has no attribute 'current_heating_setpoint' when rendering '{{ value_json.current_heating_setpoint }}'
```
and mqtt shows the error
`Could not parse temperature from `
later
```
Template variable error: 'value_json' is undefined when rendering '{{ value_json.window_detection }}'
Template variable error: 'value_json' is undefined when rendering '{{ value_json.valve_detection }}'
Template variable error: 'value_json' is undefined when rendering '{{ value_json.local_temperature }}'
Template variable error: 'value_json' is undefined when rendering '{{ value_json.current_heating_setpoint }}'
Template variable error: 'value_json' is undefined when rendering '{{ value_json.system_mode }}'
```
and
```
Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.window_detection }})
Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.valve_detection }})
Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.local_temperature }})
Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.current_heating_setpoint }})
Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.system_mode }})
```
what did i do wrong?

delicon, 2022-04-11:
I'm still getting these warnings on HA startup.
Home Assistant 2022.4.2
Zigbee2mqtt 1.25.0-1 with the latest coordinator firmware for zzh! (CC2652R) 20220219.

`2022-04-11 13:17:45 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'illuminance_above_threshold' when rendering '{{ value_json.illuminance_above_threshold }}'
2022-04-11 13:17:45 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'illuminance_above_threshold' when rendering '{{ value_json.illuminance_above_threshold }}'
2022-04-11 13:17:45 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'illuminance_above_threshold' when rendering '{{ value_json.illuminance_above_threshold }}'
2022-04-11 13:17:45 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'illuminance_above_threshold' when rendering '{{ value_json.illuminance_above_threshold }}'`

Am I missing something?

Altycoder, 2022-04-11:
@delicon I moved away from mqtt to websockets for my zwave as this issue never received a solution from the devs

Not sure if that's possible for you but worth considering

ovizii, 2022-06-13:
I can confirm, I jsut got:

```
2022-06-13 19:14:44 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: 
2022-06-13 19:14:44 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.battery }}'
```

this is a brand-new raspberry pi install with only one device connected: https://www.zigbee2mqtt.io/devices/HS2WD-E.html#heiman-hs2wd-e

PR #7032 (2021-04-12): Home Assistant: set unknown values to null #6987
Fixes #6987
Alternative for https://github.com/Koenkk/zigbee2mqtt/pull/6988

Home Assistant 2021.4 => throws warnings when values are missing from the state. This PR sets all unknown values to null.

E.g. given that we have a temperature/humidity sensor where the humidity and temperature is unknown yet. Next the sensor reports the temperature so `{"temperature": 20}` is published. Now HA will generate a warning that the `humidity` property is missing. With this pr `{"temperature": 20, "humidity": null}` would be published.

@mdegat01 @frenck can [... cut]