Koenkk/zigbee2mqtt #6991: Blitzwolf BW-SHP13 Availability Timeout

massiant70, 2021-04-07:
## What happened
When I set availability timeout to 60" for one device, the others stop to update values & states. All BW-SHP13.

## What did you expect to happen
I expect that the others device, not in the passlist, continue to work as before.

## How to reproduce it (minimal and precise)
Zigbee2MQTT, Settings, Advanced, Availability Timeout to 60. Add one device to Availability passlist. Submit. Restart Zigbee2MQTT.

## Debug info
Zigbee2MQTT version:  1.18.2-1
Adapter hardware: CC2538 + CC2592
Adapter firmware version: 20200211

Core Version: 2021.3.4
Supervisor Version: 2021.03.9
Operating System: Home Assistant OS 5.12

GSzabados, 2021-04-08:
@massiant70, isn't this what you see?

https://github.com/Koenkk/zigbee2mqtt/issues/6713

massiant70, 2021-04-08:
@GSzabados no it isn't. As I write before, the others stop to update values & states.
It seems that after changing something in Availability Timeout, the SHP13 does not update anymore.

massiant70, 2021-04-12:
I have tried also setting specific poll interval to 30 but nothing.

massiant70, 2021-04-15:
Is this a problem of Availability Timeout feature, or is it a problem specific with BW-SHP13 ???

massiant70, 2021-04-25:
I think that there is a problem with Zigbee2Mqtt - Settings - Advanced menu because I change the log level into this menu, and then all my BW-SHP13 stop to update numbers. As soon as I restart Zigbee2Mqtt, and the log level return to default setting, the BW work well.

GSzabados, 2021-04-25:
> I think that there is a problem with Zigbee2Mqtt - Settings - Advanced menu because I change the log level into this menu, and then all my BW-SHP13 stop to update numbers. As soon as I restart Zigbee2Mqtt, and the log level return to default setting, the BW work well. Then it seems that all changes into Settings - Advanced takes to stop the BW.

That behaviour is what I referenced you before, it is a bug and it has been fixed already for the dev branch (edge).

https://github.com/Koenkk/zigbee2mqtt/issues/6713

GSzabados, 2021-05-02:
@massiant70, give another go. The fix for #6713 has been published with 1.18.3. But you need HA 2021.4.5.

Release notes here:

https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.18.3

closed 2021-05-08 by massiant70