Koenkk/zigbee2mqtt #6995: Eurotronic SPZB0001 does not report status any more after updating from 1.18.1 to 1.18.2

wendehals, 2021-04-07:
## What happened
After updating from 1.18.1 to 1.18.2, all my Eurotronic SPZB0001 (5 pieces) stopped reporting their status e.g. when setting current_heating_setpoint.

## What did you expect to happen
Status should be reported after each /set command sent to the device.

## How to reproduce it (minimal and precise)
Upgrade from 1.18.1 to 1.18.2.

## Debug info
Zigbee2MQTT version: 1.18.2
Adapter hardware: CC26X2R1
Adapter firmware version: 20200925

Morphy99, 2021-04-09:
I think I've got the same problem. I noticed setting hvac_mode off or heat doesn't do anything. Also, is this eurotronic_trv_mode new? Tried changing it to no effect. Will roll back to 18.1 see if it fixes it for now.

[image]

wendehals, 2021-04-12:
I also rolled back to an older version (1.17.1) but also had to delete state.json. After rolling back, status reports have been sent again, but most of the properties like current_heating_setpoint, local_temperature and pi_heating_demand have not been updated any more. Deleting state.json fixed the problem.

bbayszczak, 2021-04-14:
Hello,

Same issue on my side, I just bought an Eurotronic Spirit SPZB0001 TRV and connected it to a Zigbee2MQTT instance running using the Home Assistant official addon.

Here is what I found in the logs when trying to reach the `running_state`

# Debug info
```
Debug Received MQTT message on 'zigbee2mqtt/spzb_a/get' with data '{"running_state":""}'
Debug Publishing get 'get' 'running_state' to 'spzb_a'
Debug Received Zigbee message from 'spzb_a', type 'readResponse', cluster 'hvacThermostat', data '{}' from endpoint 1 with groupID null
Error Publish 'get' 'running_state' to 'spzb_a' failed: 'Error: Read 0x00158d00030ca742/1 hvacThermostat(["runningState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug Error: Read 0x00158d00030ca742/1 hvacThermostat(["runningState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:196:23) at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:251:26) at Generator.next (<anonymous>) at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58)
```

# My setup

Zigbee2MQTT version: 1.18.2 commit: 1.18.2
Coordinator type: ConBee2/RaspBee2
Coordinator revision: 0x26660700
ZigBee2MQTT running in Home Assistant using the official add-on

I only tried with `1.18.2` on my side, @wendehals you confirm it was working in `1.18.1` or was it only in `1.17.1` ?
I'd like to check changes between these versions if I see anything explaining this.

BeistleA, 2021-04-15:
Hi,
also had the problems with 1.18.2, got back to 1.18.1 and seems to be working again.
@wendehals: thanks for the hint with deleting state.json (cause as you said not all updates are send), is it just stop zigbee2mqtt, delete it and start again or anything else?

Greetings
Andy

bbayszczak, 2021-04-18:
Here are herdsman's logs when changing the mode from `heat` to `auto`

```
2021-04-18T11:28:02.180Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2021-04-18T11:28:02.181Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-04-18T11:28:02.185Z zigbee-herdsman:deconz:driver query aps data indication
2021-04-18T11:28:02.190Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 252
2021-04-18T11:28:02.193Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 252 srcAddr: 0xdf15 destAddr: 0x0 profile id: 0x104 cluster id: 0x201
2021-04-18T11:28:02.193Z zigbee-herdsman:deconz:frameParser response payload: 28,55,16,21,10,3,64,41,58,7,8,64,34,1,0,0
2021-04-18T11:28:02.199Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":21,"manufacturerCode":4151,"commandIdentifier":10},"Payload":[{"attrId":16387,"dataType":41,"attrData":1850},{"attrId":16392,"dataType":34,"attrData":1}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":57109,"endpoint":1,"linkquality":46,"groupID":null,"wasBroadcast":false}'
2021-04-18T11:28:02.202Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
```

wendehals, 2021-04-20:
@BeistleA, should be fine with stopping zigbee2mqtt, delete state.json, and restart zigbee2mqtt. It seems to build a new state.json.
> 
> Hi,
> also had the problems with 1.18.2, got back to 1.18.1 and seems to be working again.
> @wendehals: thanks for the hint with deleting state.json (cause as you said not all updates are send), is it just stop zigbee2mqtt, delete it and start again or anything else?
> 
> Greetings
> Andy

wendehals, 2021-04-20:
@bbayszczak, I was running 1.18.1 before upgrading to 1.18.2 and it was working fine. When downgrading to 1.18.1 I was having trouble with single parameters not getting updated in status reports. So I went further back to 1.17.1 but had the same problem. I found out that deleting state.json fixed the problem, but I kept the version 1.17.1. The version 1.18.1 should be fine, as said before, it worked before upgrading.
> 
> Hello,
> 
> Same issue on my side, I just bought an Eurotronic Spirit SPZB0001 TRV and connected it to a Zigbee2MQTT instance running using the Home Assistant official addon.
> 
> Here is what I found in the logs when trying to reach the `running_state`
> ## Debug info
> 
> ```
> Debug Received MQTT message on 'zigbee2mqtt/spzb_a/get' with data '{"running_state":""}'
> Debug Publishing get 'get' 'running_state' to 'spzb_a'
> Debug Received Zigbee message from 'spzb_a', type 'readResponse', cluster 'hvacThermostat', data '{}' from endpoint 1 with groupID null
> Error Publish 'get' 'running_state' to 'spzb_a' failed: 'Error: Read 0x00158d00030ca742/1 hvacThermostat(["runningState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
> Debug Error: Read 0x00158d00030ca742/1 hvacThermostat(["runningState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:196:23) at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:251:26) at Generator.next (<anonymous>) at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58)
> ```
> 
> ## My setup
> 
> Zigbee2MQTT version: 1.18.2 commit: 1.18.2
> Coordinator type: ConBee2/RaspBee2
> Coordinator revision: 0x26660700
> ZigBee2MQTT running in Home Assistant using the official add-on
> 
> I only tried with `1.18.2` on my side, @wendehals you confirm it was working in `1.18.1` or was it only in `1.17.1` ?
> I'd like to check changes between these versions if I see anything explaining this.
> I'm new to Zigbee2MQTT I'll probably need some help to identify what's wrong 😄

kaykoch, 2021-04-28:
Zigbee2mqtt Current version: 1.18.2-1 (Home Assistant Add-on)
Not sure, when it started, but not long ago.

I only get less information after deleting states.json. Not even battery.
system_mode is not available.

```
{
    "linkquality": 105,
    "local_temperature": 18.4,
    "pi_heating_demand": 100,
    "update": {
        "state": "idle"
    },
    "update_available": false,
    "occupied_heating_setpoint": 20,
    "unoccupied_heating_setpoint": 16
}
```

kaykoch, 2021-05-03:
After upgrading to: 
**Zigbee2mqtt
Current version: 1.18.3-1
Zigbee2mqtt add-on**

### The mqtt payload changed to:
```
{
    "battery": null,
    "eurotronic_trv_mode": null,
    "eurotronic_valve_position": null,
    "linkquality": 153,
    "local_temperature": 19.5,
    "occupied_heating_setpoint": 18,
    "pi_heating_demand": 1,
    "unoccupied_heating_setpoint": 16,
    "update": {
        "state": "idle"
    },
    "update_available": false
}
```
### Missing:
 - system_mode
 - eurotronic_host_flags
 
### Therefore I have warnings and errors in Log.
**missing  value_json.running_state results in**

```
2021-05-03 14:26:53 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: dict object has no element Undefined when rendering '{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}'
```
**missing  value_json.system_mode results in**

```
2021-05-03 14:26:53 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'system_mode' when rendering '{{ value_json.system_mode }}'
2021-05-03 14:26:53 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode:
```

Papnaas, 2021-05-30:
Hi,
I can confirm that moving from v1.18.1 to v1.18.2 or even 1.18.3 renders problems with this TRV.

I made a simple setup in Node-RED sending `{"eurotronic_host_flags": {"mirror_display": true}}` and `{"eurotronic_host_flags": {"mirror_display": false}}` as well as `{"eurotronic_host_flags": {"window_open": true}}` and `{"eurotronic_host_flags": {"window_open": false}}` to control the display setup and window open mode.

With v1.18.1 everything works as expected, every time I toggle, I receive a message from the TRV reporting the new state of the flags:
_Toggle mirror_display true_
`Zigbee2MQTT:info 2021-05-30 21:49:42: MQTT publish: topic 'zigbee2mqtt/EUROTRONIC Heizungsventil Keller', payload '{"battery":100,"current_heating_setpoint":23,"eurotronic_error_status":0,"eurotronic_host_flags":{"boost":false,"child_protection":false,"mirror_display":true,"window_open":false},"linkquality":69,"local_temperature":20.96,"pi_heating_demand":57}'`

_Toggle mirror_display false_
`Zigbee2MQTT:info 2021-05-30 21:49:43: MQTT publish: topic 'zigbee2mqtt/EUROTRONIC Heizungsventil Keller', payload '{"battery":100,"current_heating_setpoint":23,"eurotronic_error_status":0,"eurotronic_host_flags":{"boost":false,"child_protection":false,"mirror_display":false,"window_open":false},"linkquality":66,"local_temperature":20.96,"pi_heating_demand":57}'`

_Toggle window_open true_
`Zigbee2MQTT:info  2021-05-30 22:05:45: MQTT publish: topic 'zigbee2mqtt/EUROTRONIC Heizungsventil Keller', payload '{"battery":100,"current_heating_setpoint":23,"eurotronic_error_status":0,"eurotronic_host_flags":{"boost":false,"child_protection":false,"mirror_display":false,"window_open":false},"linkquality":60,"local_temperature":21.4,"pi_heating_demand":57}'`

`Zigbee2MQTT:info  2021-05-30 22:05:46: MQTT publish: topic 'zigbee2mqtt/EUROTRONIC Heizungsventil Keller', payload '{"battery":100,"current_heating_setpoint":5,"eurotronic_error_status":0,"eurotronic_host_flags":{"boost":false,"child_protection":false,"mirror_display":false,"window_open":true},"linkquality":63,"local_temperature":21.4,"pi_heating_demand":57}'`

_Toggle window_open false_
`Zigbee2MQTT:info  2021-05-30 22:05:58: MQTT publish: topic 'zigbee2mqtt/EUROTRONIC Heizungsventil Keller', payload '{"battery":100,"current_heating_setpoint":5,"eurotronic_error_status":0,"eurotronic_host_flags":{"boost":false,"child_protection":false,"mirror_display":false,"window_open":true},"linkquality":66,"local_temperature":21.4,"pi_heating_demand":0}'`

`Zigbee2MQTT:info  2021-05-30 22:05:59: MQTT publish: topic 'zigbee2mqtt/EUROTRONIC Heizungsventil Keller', payload '{"battery":100,"current_heating_setpoint":23,"eurotronic_error_status":0,"eurotronic_host_flags":{"boost":false,"child_protection":false,"mirror_display":false,"window_open":false},"linkquality":63,"local_temperature":21.4,"pi_heating_demand":0}'`

(Interesting by the way, why it always sends the current status after sending the window_open flags...)

With v1.18.2 or v1.18.3, I receive hardly any message at all and if I do, the flags are always 'false'. However, the TRV reacts to the toggle and mirros the display.
In case I send the `window_open true` flag, the TRV changes correctly to the "OFF" state (without a message on the output)  and stays there forever, as further issues of `window_open false` won't change anything...

closed 2021-07-07 by github-actions[bot]