Koenkk/zigbee2mqtt #7002: Multiple entries at same second

jbrepogmailcom, 2021-04-08:
## What happened
In the log, there are multiple entries when device sends data. Is it because device sends it several times, or other reason?
```
info  2021-04-08 19:59:55: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffec98a1b', payload '{"battery":50,"humidity":29.96,"linkquality":136,"temperature":23.8,"voltage":2700}'
info  2021-04-08 19:59:55: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffec98a1b', payload '{"battery":50,"humidity":29.96,"linkquality":136,"temperature":23.8,"voltage":2700}'
info  2021-04-08 19:59:55: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffec98a1b', payload '{"battery":50,"humidity":29.96,"linkquality":136,"temperature":23.8,"voltage":2700}'
info  2021-04-08 19:59:55: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffec98a1b', payload '{"battery":50,"humidity":29.96,"linkquality":136,"temperature":23.8,"voltage":2700}'
info  2021-04-08 19:59:55: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffec98a1b', payload '{"battery":50,"humidity":29.96,"linkquality":136,"temperature":23.8,"voltage":2700}'
info  2021-04-08 19:59:55: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffec98a1b', payload '{"battery":50,"humidity":29.96,"linkquality":136,"temperature":23.8,"voltage":2700}'
info  2021-04-08 19:59:55: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffec98a1b', payload '{"battery":50,"humidity":29.96,"linkquality":136,"temperature":23.8,"voltage":2700}'

```

## What did you expect to happen
Only one entry per data packet

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.8.12
Adapter hardware: CC2531
Adapter firmware version:

eoris, 2021-04-10:
@jbrepogmailcom you can try to use `debounce` option (https://www.zigbee2mqtt.io/information/configuration.html).

jbrepogmailcom, 2021-04-10:
Thanks, ```debounce: 1``` solved it.

closed 2021-04-10 by jbrepogmailcom