Koenkk/zigbee2mqtt #7004: Too much wake per seconde

roumano, 2021-04-08:
## What happened
After the service start and in a idle state , zigbee2mqtt wake up too much time per second

It's visible via 2 kind of command :  
- powertop
arround 250 to 300 events/sec 
for comparaison, mosquitto : 5 events/sec and home assistant less than 10 events/sec 
- strace
```
strace -p $(pidof node index.js) > /tmp/1 2>&1
# stop it after 10 second
```
on the log file, it's content like **3000 clock_gettime per second**( like `clock_gettime(CLOCK_MONOTONIC, {tv_sec=125521, tv_nsec=997610561}) = 0`
exact number can be found via : `grep -c clock_gettime /tmp/1` then divise by 10 (for 10 second)

## What did you expect to happen
it's should not wake-up so often or need to change how it's working ? 
## How to reproduce it (minimal and precise)
```
systemctl start zigbee2mqtt.service
powertop
```
## Debug info
Zigbee2MQTT version: 1.18.2
Adapter hardware: Conbee II
Adapter firmware version: lasted available : deCONZ_ConBeeII_0x26680700.bin.GCF : release the 30-Nov-2020

nurikk, 2021-04-09:
Interesting finding, this might come from nodeserial package.

Can you describe your setup? OS / Serial device (uart/usb)?

roumano, 2021-04-09:
Hi,
Debian 10 on raspberry 4 8Gio
All installation was done on the OS (no VM, no docker).
Combee2 mounted on a usb port. Pi connected via wire cable.

A harddisk is mounted on a other usb port

Koenkk, 2021-04-09:
Same as https://github.com/Koenkk/zigbee2mqtt/discussions/6747 (some bug in the Conbee 2 code). @chrishae can you look at this?

ChrisHae, 2021-04-12:
Can you please provide zigbee-herdsman debug logs from the startup of z2mqtt. https://www.zigbee2mqtt.io/information/debug.html

roumano, 2021-04-13:
Hi @ChrisHae , sorry for the delay ...
please found the log output
```
~$ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.18.2 start /data/zigbee2mqtt
> node index.js

Zigbee2MQTT:debug 2021-04-13 10:20:10: Loaded state from file /data/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2021-04-13 10:20:10: Logging to console and directory: '/tmp/' filename: log.txt
Zigbee2MQTT:info  2021-04-13 10:20:10: Starting Zigbee2MQTT version 1.18.2 (commit #abd8a09)
Zigbee2MQTT:info  2021-04-13 10:20:10: Starting zigbee-herdsman (0.13.88)
Zigbee2MQTT:debug 2021-04-13 10:20:10: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/data/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/data/zigbee2mqtt/data/database.db.backup","databasePath":"/data/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'DeconzAdapter' +0ms
  zigbee-herdsman:deconz:driver Set delay to 0 +0ms
CREATED DECONZ ADAPTER
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/data/zigbee2mqtt/data/database.db","databaseBackupPath":"/data/zigbee2mqtt/data/database.db.backup","backupPath":"/data/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0 +8ms
  zigbee-herdsman:deconz:driver Serialport opened +9ms
  zigbee-herdsman:controller:log Started with result 'resumed' +15ms
  zigbee-herdsman:controller:log Injected database: true, adapter: true +0ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1 +3ms
  zigbee-herdsman:deconz:frameParser MAC: 0x00212effff06d42f +0ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 2 paramId: 7 +7ms
  zigbee-herdsman:deconz:frameParser NWKADDR: 0 +4ms
Zigbee2MQTT:info  2021-04-13 10:20:11: zigbee-herdsman started
  zigbee-herdsman:deconz:driver send read firmware version request from queue. seqNr: 3 +10ms
  zigbee-herdsman:deconz:frameParser read firmware version response - version: 0,7,104,38 +9ms
Zigbee2MQTT:info  2021-04-13 10:20:11: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":104,"product":0,"revision":"0x26680700","transportrev":0},"type":"ConBee2/RaspBee2"}'
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 4 paramId: 5 +5ms
  zigbee-herdsman:deconz:frameParser PANID: 1a62 +6ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 5 paramId: 11 +5ms
  zigbee-herdsman:deconz:frameParser APS_EXT_PANID: 0xdddddddddddddddd +5ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 6 paramId: 28 +5ms
  zigbee-herdsman:deconz:frameParser CHANNEL: 11 +5ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 7 paramId: 24 +6ms
  zigbee-herdsman:deconz:frameParser NETWORK_KEY: hidden +5ms
Zigbee2MQTT:debug 2021-04-13 10:20:11: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:info  2021-04-13 10:20:11: Currently 43 devices are joined:
Zigbee2MQTT:info  2021-04-13 10:20:11: Prise Test (0x00158d0003569c29): SP 120 - Innr Smart plug (Router)
Zigbee2MQTT:info  2021-04-13 10:20:11: Lumiere Piscine (0x01124b001baea393): KS-SM001 - Ksentry Electronics [Zigbee OnOff Controller](http://ksentry.manufacturer.globalsources.com/si/6008837134660/pdtl/ZigBee-module/1162731630/zigbee-on-off-controller-modules.htm) (Router)
Zigbee2MQTT:info  2021-04-13 10:20:11: Interrupteur Garage (0xbc33acfffe5010e7): ZM-L03E-Z - Zemismart Smart light switch - 3 gang with neutral wire (Router)
Zigbee2MQTT:info  2021-04-13 10:20:11: Detecteur Garage (0x00158d00036018a1): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:info  2021-04-13 10:20:11: Inter Connecter Garage (0xec1bbdfffea8a710): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
Zigbee2MQTT:info  2021-04-13 10:20:11: Store (0x086bd7fffefb55de): ZM-CSW032-D - Zemismart Curtain/roller blind switch (Router)
Zigbee2MQTT:info  2021-04-13 10:20:11: Detecteur Porte d'entrée (0x001788010646eee2): 9290019758 - Philips Hue motion outdoor sensor (EndDevice)
Zigbee2MQTT:info  2021-04-13 10:20:11: Porte d'entrée Lumiere (0xec1bbdfffe97250b): LED1836G9 - IKEA TRADFRI LED bulb E26/E27 806 lumen, dimmable, warm white (Router)
Zigbee2MQTT:info  2021-04-13 10:20:11: Buanderie (0x00158d00041ce03d): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
Zigbee2MQTT:info  2021-04-13 10:20:11: Detecteur Buanderie (0x00158d000360182f): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:info  2021-04-13 10:20:11: Grenier Milieu (0x00158d0006401355): HGZB-41 - Nue / 3A Smart one gang wall switch (Router)
Zigbee2MQTT:info  2021-04-13 10:20:11: Grenier Droite (0x00158d0005669e40): HGZB-41 - Nue / 3A Smart one gang wall switch (Router)
Zigbee2MQTT:info  2021-04-13 10:20:11: Grenier gauche (0x00158d00044cadd6): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
[... cut ...]
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 56 srcAddr: 0x7af destAddr: 0x0 profile id: 0x0 cluster id: 0x801f +11ms
  zigbee-herdsman:deconz:frameParser response payload: 0,132,1,47 +0ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +4ms
  zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 57 +7s
  zigbee-herdsman:deconz:frameParser device state: 10100010 +7s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
^C

```

ChrisHae, 2021-04-14:
I think the cause is too fast time intervals for checking internal queues. This came with the new "adapter_delay" option in the configuration.yaml in z2mqtt v1.18.0.

setting a delay value between 10 and 60 will reduce the cpu load.

configuration.yaml:

> advanced:
>   adapter_delay: 50

But I just noticed the value is not passed correctly. So I uploaded a fix to zigbee-herdsman. i think it will be soon in the latest z2mqtt dev-branch version. https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

roumano, 2021-04-14:
Great, i will happy to test the Dev branch, please notify me when this change will be in the branch. so i will test.

Regards

Koenkk, 2021-04-14:
Will be in the dev branch in 1 hours from now.

roumano, 2021-04-14:
Build and use the dev branch is working as expected.
I have put `adapter_delay: 50` and now,i got around **65 events/s** (before it was 300) for the zigbee2qttt service (`node index.js`) and the cpu usage decrease to 1.7%.
65, it's a normal value (compared to other zigbee stick ? )

Gr4ffy, 2021-04-15:
Same [here](https://github.com/Koenkk/zigbee2mqtt/discussions/6747#discussion-3279634)
Docker "dev" is working fine now.

closed 2021-05-24 by github-actions[bot]