Koenkk/zigbee2mqtt #7007: Zemismart ZM25TQ - the position of the Curtain cannot be set via HomeKit

EddieDSuza, 2021-04-09:
## What happened
The position can't be set properly via Zigbee2MQTT

## What did you expect to happen
Using the Homebridge Z2M plugin - On interaction - the device is exposed to HomeKit but the device does not close or open. 

Using Manual MQTT commands - the device can only be set using the open and close commands. It does not follow the position set command. The device does open and close in the frontend 

## How to reproduce it (minimal and precise)
MQTT Device information - [](https://pastebin.com/2PEFSAu8)

## Debug info
Zigbee2MQTT version: 1.18.1-dev commit: f8f79d3
Adapter hardware: Conbee II
Adapter firmware version: 0x26660700

EddieDSuza, 2021-04-12:
So I have been doing some research and t looks like in the device.js there is no 

**supports: 'open, close, stop, position',**

Any idea on how to add this for HomeBridge

EddieDSuza, 2021-04-13:
@Koenkk can you please take a look at this issue .... Have been informed to raise an issue [here](https://github.com/itavero/homebridge-z2m/issues/129#issuecomment-813817084)

Koenkk, 2021-04-14:
What happens when you send the following message via MQTT: topic `zigbee2mqtt/DEVICE_FRIENDLY_NAME/set` message: `{"position": 30}`

EddieDSuza, 2021-04-14:
Well I did try that and no reaction from the Curtain. 

Plus I went ahead and installed Home Assistant to see it would work once it was exposed to HomeKit. Thru Home Assistant - works flawlessly. Thru HomekIt, the device appears but no reaction.

Thru MQTT - it only works with  `zigbee2mqtt/DEVICE_FRIENDLY_NAME/set message: OPEN or STOP or CLOSE`

Koenkk, 2021-04-15:
@DomiStyle it seems position is not supported for all TS0601 curtains, do you know what is wrong here or do some simply don't support setting a position?

DomiStyle, 2021-04-15:
@Koenkk I'm not aware of any curtains that do not support position so far, except when they are not calibrated.

@EddieDSuza Did you calibrate the curtain already?

EddieDSuza, 2021-04-16:
Hello @DomiStyle - I havent installed it yet and havent configured the curtains position ?

do I need to that physically prior to adding it to Z2M ?

Please let me know

DomiStyle, 2021-04-16:
@EddieDSuza It depends on the device, maybe there's a manual that came with the curtain?

You do need to install and calibrate it before position is reported and can be set.

EddieDSuza, 2021-04-16:
@DomiStyle thanks for the Tip - will go ahead and install it 

will report the status once its done. Cheers

closed 2021-05-25 by github-actions[bot]