Koenkk/zigbee2mqtt #7011: LEDVANCE Bulb Model AC25697 Power source show unknown

shirou93, 2021-04-09:
[image]

z2m 1.18.2-1

Propably fast fix :) Thanks

Koenkk, 2021-04-09:
Can you try repairing the device? (remove it first). Power source has to come from the device itself.

shirou93, 2021-04-09:
Yes, repairing device don't help. I try remove etc.  I installed ota firmware 02036550.

Alweys Power source is "?"

Koenkk, 2021-04-10:
Can you provide the data/database.db entry of this device?

shirou93, 2021-04-10:
`{"id":26,"type":"Router","ieeeAddr":"0xf0d1b8000018fff7","nwkAddr":17029,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Unknown","modelId":"A60 RGBW Value II","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"A60 RGBW Value II","manufacturerName":"LEDVANCE","powerSource":0,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":2,"swBuildId":"02036550"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":526,"colorTemperature":370,"currentX":27525,"currentY":24379,"currentHue":0,"currentSaturation":254}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021a8eae9","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0021a8eae9","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b0021a8eae9","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":300,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":3,"maxRepIntval":300,"repChange":1},{"cluster":768,"attrId":7,"minRepIntval":3,"maxRepIntval":300,"repChange":1},{"cluster":768,"attrId":3,"minRepIntval":3,"maxRepIntval":300,"repChange":1},{"cluster":768,"attrId":4,"minRepIntval":3,"maxRepIntval":300,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":2,"swBuildId":"02036550","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2,"reporting":1},"lastSeen":1618068354161}`

Koenkk, 2021-04-10:
Should be fixed now, to apply click the reconfigure button (yellow) once in the frontend

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-04-11 by shirou93