Koenkk/zigbee2mqtt #7016: Light brightness always 100% after Zigbee2MQTT restart

titus-kruse, 2021-04-09:
## What happened

After restarting Zigbee2MQTT and turning on a light, it shines with 100% brightness. Even if it was set less bright before.

## What did you expect to happen

If only ON is sent, the brightness should not be affected.

## How to reproduce it (minimal and precise)

zigbee2mqtt/FRIENDLY_NAME/set: {"state":"ON"}
zigbee2mqtt/FRIENDLY_NAME/set: {"brightness":128}
zigbee2mqtt/FRIENDLY_NAME/set: {"state":"OFF"}

Restart zigbee2MQTT docker container.

zigbee2mqtt/FRIENDLY_NAME/set: {"state":"ON"}

MQTT response:

{"brightness":254,"linkquality":78,"state":"ON"}

Tried to disable Zigbee2MQTT cache, but no improvement. Tried to use device power off behavior, but no improvement.

## Debug info
Zigbee2MQTT version: version 1.17.0 (commit #07cdc9d)
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608
Thomas Nordquist MQTT Explorer 0.3.5
Mosquitto MQTT v3.1/v3.1.1 Broker
Philips Hue White A60 Single bulb E27/B22 (8718696449691).

## Log output

[zigbee2mqtt.log.txt](https://github.com/Koenkk/zigbee2mqtt/files/6284951/zigbee2mqtt.log.txt)

titus-kruse, 2021-04-18:
Ich nochmal. Ich bin ein wenig enttäuscht, dass niemand etwas zu meiner Frage geschrieben hat. Ich wäre über jeden Hinweis dankbar. Ich habe schon einige Stunden investiert, um selbst dem Fehlverhalten auf die Schliche zu kommen. Falls es überhaupt ein Fehler ist?

Koenkk, 2021-04-18:
Seems similar to https://github.com/Koenkk/zigbee2mqtt/issues/7106 , let's continue there.

closed 2021-04-18 by Koenkk