Koenkk/zigbee2mqtt #7019: FrankEver Smart Water Valve - Al files there - Need help to get them into release

ArJay60, 2021-04-09:
Created support of a new device with the option of external converter in configuration.yaml. Need assistance to fold this into regular supported device. In zip are md-file, image-file and the external converter js that used and is working.

[FrankEver.zip](https://github.com/Koenkk/zigbee2mqtt/files/6287073/FrankEver.zip)

device link: http://www.frankever.com/product_nei.aspx?id=1485

ArJay60, 2021-04-17:
@Koenkk . Can you help me to fold this external converter and supporting files in a new release? I am not familiar with git. Regards.

Koenkk, 2021-04-18:
Created https://github.com/Koenkk/zigbee-herdsman-converters/pull/2491 , let's continue discussion there.

closed 2021-04-18 by Koenkk