Koenkk/zigbee2mqtt #7024: Document binding for each device

vk496, 2021-04-09:
## Is your feature request related to a problem? Please describe
It's very annoying to buy a device and discover that it doesn't cover your needs. In my case, it happened with a Tuya switch. I tried to bind it to a zbmini but it seems that the switch is not able to bind.

## Describe the solution you'd like
I would like to have documented which devices are known to have working the binding feature.

By default, all supported devices could have a "not reported" status, and edit device by device which of them have it.

The same could apply for Ota upgrades, since some devices are capable (the genOta cluster, right?) but not available publically.

BR,
Valentín :)

sjorge, 2021-04-11:
Some notes from personal experience, most TuYa devices use a TuYA SDK that runs on top of their wifi and zigbee stracks.
They seem to not support a lot of zigbee features like binding, reporting, .... personally I now avoid all TuYa devices unless I buy one specifically for testing. I also tend to avoid Xiaomi for similar reasons, although they do  tend to be a bit better.

Having extra info like OTA, Binding, Reporting to indicate if they support it would be very welcome indeed.

MattWestb, 2021-04-11:
The large things is if devices  is "real ZB(3)LL devices  = made for sending commands to Zigbee Light (groups) or Zigbee Home Automation (seeding "clicks").
User like have one autonomy function of lights is important getting the right type and the same is like using node read you is getting more functions to play with HA devices but is loosing the function that your  light is working then your host system is down.

Old Xiaomi (no ZB3) was PITA and most (but not all) tuya devices is using Zigbee as one MQTT tunnel and is not ZLL or HA and most users is not knowing wot is working with other things and is getting problem like one user did not getting one new IKEA shortcut button not sending one off command (that is not made for doing).

The best wold being on standard writing for device "HA click", Light command (ZLL or Z3L) or other (like tuya and explaining wot is different).

closed 2021-05-29 by github-actions[bot]