Koenkk/zigbee2mqtt #7025: Docker and External Converters help

Gwenlad, 2021-04-10:
Hello.
I'm always trying to put my to lexman bulb in my HA since 2 weeks now (I'm very bad but persistent)

I follow "the how to support new devices" doc for docker but I'm stuck at this point
**Now add the volumes to the Docker container by adding the following to your docker run command.:**
`-v /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters`

I'm not sure to understand where I have to put the code
I try on the terminal after the first part of the tutorial but it doesn't work

Thank You ! :)

McGiverGim, 2021-04-10:
This is a command that you need to pass when creating the docker container. It simply mounts a folder with the cubetera to be accessible from both sides

Gwenlad, 2021-04-10:
but I don't know where I have to put this command. The container was created automaticly when I install the zigbee2mqtt addon with HA weeks ago.

closed 2021-05-18 by github-actions[bot]