Koenkk/zigbee2mqtt #7026: TS0601_thermostat: action and valve_detection properties are missing in state

antroids, 2021-04-10:
## What happened

Action and valve_detection properties are described in HA config topics, but missing in state.

Climate config with **action**:
`{"action_template":"{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}","action_topic":"zigbee2mqtt/th1","availability":[{"topic":"zigbee2mqtt/bridge/state"}],"away_mode_command_topic":"zigbee2mqtt/th1/set/away_mode","away_mode_state_template":"{{ value_json.away_mode }}","away_mode_state_topic":"zigbee2mqtt/th1","current_temperature_template":"{{ value_json.local_temperature }}","current_temperature_topic":"zigbee2mqtt/th1","device":{"identifiers":["zigbee2mqtt_0x847127fffe11dd6a"],"manufacturer":"TuYa","model":"Radiator valve with thermostat (TS0601_thermostat)","name":"th1","sw_version":"Zigbee2MQTT 1.18.2"},"hold_command_topic":"zigbee2mqtt/th1/set/preset","hold_modes":["schedule","manual","boost","complex","comfort","eco"],"hold_state_template":"{{ value_json.preset }}","hold_state_topic":"zigbee2mqtt/th1","json_attributes_topic":"zigbee2mqtt/th1","max_temp":"35","min_temp":"5","mode_command_topic":"zigbee2mqtt/th1/set/system_mode","mode_state_template":"{{ value_json.system_mode }}","mode_state_topic":"zigbee2mqtt/th1","modes":["heat","auto","off"],"name":"th1","temp_step":0.5,"temperature_command_topic":"zigbee2mqtt/th1/set/current_heating_setpoint","temperature_state_template":"{{ value_json.current_heating_setpoint }}","temperature_state_topic":"zigbee2mqtt/th1","temperature_unit":"C","unique_id":"0x847127fffe11dd6a_climate_zigbee2mqtt"}`

**valve_detection** config:
`{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/th1/set/valve_detection","device":{"identifiers":["zigbee2mqtt_0x847127fffe11dd6a"],"manufacturer":"TuYa","model":"Radiator valve with thermostat (TS0601_thermostat)","name":"th1","sw_version":"Zigbee2MQTT 1.18.2"},"json_attributes_topic":"zigbee2mqtt/th1","name":"th1 valve detection","payload_off":"OFF","payload_on":"ON","state_off":"OFF","state_on":"ON","state_topic":"zigbee2mqtt/th1","unique_id":"0x847127fffe11dd6a_valve_detection_zigbee2mqtt","value_template":"{{ value_json.valve_detection }}"}`

Actual state:
`topic 'zigbee2mqtt/th1', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"UNLOCK","comfort_temperature":20,"current_heating_setpoint":26,"eco_temperature":15,"force":"normal","holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"linkquality":78,"local_temperature":24.5,"local_temperature_calibration":-1,"max_temperature":35,"min_temperature":5,"position":100,"preset":"manual","system_mode":"auto","update":{"state":"idle"},"update_available":false,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":10,"temperature":5},"workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}]}'
`
## What did you expect to happen
All properties described in HA config should be presented in state payload.

## Debug info
Zigbee2MQTT version
1.18.2 commit: abd8a09
Coordinator type
zStack12
Coordinator revision
20201127
Frontend version
0.3.114
Adapter hardware: CC2531
Adapter firmware version: CC2531 DEFAULT 20201127

handfreezer, 2021-04-13:
Hello, I lost the manual of my TS0601, may I ask you a scan or a phot of it please?
Best regards

antroids, 2021-04-13:
> Hello, I lost the manual of my TS0601, may I ask you a scan or a phot of it please?
> Best regards

You can take it here: https://www.moeshouse.com/products/tuya-zigbee3-0-new-mini-wifi-radiator-actuator-smart-programmable-thermostat-temperature-trv-controller

closed 2021-05-22 by github-actions[bot]