Koenkk/zigbee2mqtt #7029: Need assistance to add Tuya smoke alarm TS0601 - manufacturerName: _TZE200_vzekyi4c

mrTuomoK, 2021-04-10:
## Information about the device + link
Smoke and possibly temperature sensor ([link](https://www.aliexpress.com/item/1005002283181842.html))

Below is some info of it & debug data. What should I try next?

## data/database.db entry of the device

```
{
  "id": 45,
  "type": "EndDevice",
  "ieeeAddr": "0x847127fffe0c676f",
  "nwkAddr": 12375,
  "manufId": 4098,
  "manufName": "_TZE200_vzekyi4c",
  "powerSource": "Battery",
  "modelId": "TS0601",
  "epList": [
    1
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 81,
      "inClusterList": [
        0,
        4,
        5,
        61184
      ],
      "outClusterList": [
        25,
        10
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "TS0601",
            "manufacturerName": "_TZE200_vzekyi4c",
            "powerSource": 3,
            "zclVersion": 3,
            "appVersion": 68,
            "stackVersion": 0,
            "hwVersion": 1,
            "dateCode": ""
          }
        }
      },
      "binds": [
        {
          "cluster": 0,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0021cc4134",
          "endpointID": 1
        }
      ],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 68,
  "stackVersion": 0,
  "hwVersion": 1,
  "dateCode": "",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {
    "configured": 1
  },
[... 72 lines cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 30 - 2 - 4 - 129 - [0,0,0,239,37,80,1,1,0,153,0,221,58,226,0,0,10,9,3,1,0,143,1,4,0,1,1,98,6,27] - 41 +9ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":20517,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":153,"securityuse":0,"timestamp":14826205,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[9,3,1,0,143,1,4,0,1,1]}} +3s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":3,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"status":0,"transid":143,"dp":1,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[1]}},"Command":{"ID":1,"parameters":[{"name":"status","type":32},{"name":"transid","type":32},{"name":"dp","type":32},{"name":"datatype","type":32},{"name":"fn","type":32},{"name":"data","type":65}],"name":"getData"}},"address":20517,"endpoint":1,"linkquality":153,"groupID":0,"wasBroadcast":false}' +3s
Zigbee2MQTT:debug 2021-04-10 12:32:30: Received Zigbee message from '0x847127fffe0c676f', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"status":0,"transid":143}' from endpoint 1 with groupID 0
  zigbee-herdsman:controller:endpoint DefaultResponse 0x847127fffe0c676f/1 61184(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +3s
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x847127fffe0c676f:20517/1 (0,0,1) +3s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":20517,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":26,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,3,11,1,0]}} +3s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,37,80,1,1,0,239,26,0,30,5,24,3,11,1,0,160] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +15ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,26,220] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,26,220] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,26] - 220 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":26} +29ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,165,178,1,43,32,153] +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,165,178,1,43,32,153] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [165,178,1,43,32] - 153 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":45733,"relaycount":1,"relaylist":[8235]} +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
```

Koenkk, 2021-04-10:
Instead of `type: ['raw'],` in fromZigbee try `type: ['commandGetData'],`

mrTuomoK, 2021-04-11:
Hey thanks @Koenkk 👍  Now I'm getting somewhere with this and after quick testing I get the linkquality and smoke status correctly when pushing the test button.

Koenkk, 2021-04-12:
Good, can you make a pull request?

mrTuomoK, 2021-04-13:
I could, a bit busy with work currently but I'll try some day soon'ish.

closed 2021-05-22 by github-actions[bot]

vadim-mezhlumov, 2021-09-25:
I have the same sensor. Here is my external converter:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const fze = {
    tuya_smoke:
    {
        cluster: 'manuSpecificTuya',
        type: ['commandGetData', 'commandSetDataResponse'],
        convert: (model, msg, publish, options, meta) =>
        {
            const dp = msg.data.dp;
            const value = tuya.getDataValue(msg.data.datatype, msg.data.data);

            switch (dp)
            {
                // Smoke
                case tuya.dataPoints.state:
                    return {smoke: value === 0};

                // Battery level (%)
                case 15:
                    return {battery: value};
        
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:TuYaSmoke: NOT RECOGNIZED DP #${dp} with data ${JSON.stringify(msg.data)}`);
            }
        },
    },
};

const definition =
{
    fingerprint:
    [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_yh7aoahi'
        },
    ],
    model: 'TS0601_smoke',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [ fze.tuya_smoke ],
    toZigbee: [],
    onEvent: tuya.setTime,
    configure: async (device, coordinatorEndpoint, logger) => 
    {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [ e.smoke(), e.battery() ],
};

module.exports = definition;
```
It seems, data point 15 is battery level. There are 2 additional data points: 14 and 16.
> {"status":0,"transid":28,"dp":14,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[2]}}
> {"status":0,"transid":15,"dp":16,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[1]}}

I haven't seen that these values are changed. I have no idea what they mean. Does anybody know?