Koenkk/zigbee2mqtt #7030: Problem to pairing : Xiaomi WSDCGQ11LM

kwetnico, 2021-04-10:
Hello,

I have 29 devices on my configuration, I wanted to re-pair all my devices, for that I reset my database.db and state.json file to blank.

I was able to restore my xiaom smoke, human presence, door opening devices without any problem.
On the other hand my xiaomi temperature sensors (WSDCGQ11LM) on 12 I was able to re-pair only 2 impossible I remove the batteries restart zigbee2mqtt, my raspberry (I even reflashed my conbee2) nothing is happening I looked in the logs there there is nothing ...

What to do ? thank you

## Debug info
Rapsberry pi4
Zigbee2MQTT version: 1.18.2
Adapter hardware:  Conbee II
Adapter firmware version: the last

Krazy998, 2021-04-10:
i had serious issues trying to pair some of my aqara/xiaomi temperature and door sensors with the new CC2652R zzh stick. No problems on my old cc2531.

The approach i took was this (its extremely painful).

1. stop zigbee2mqtt addon in supervisor.
2. unplug cc2652R from usb for 5 seconds.
3. plug back in.
4. Start zigbee2mqtt. Allow joining.
5. Hold Reset button on temp sensor for 5 seconds and let go.
6. Then _**every**_ 2 seconds single press (for 1 second - should single flash and keep the device awake) the button on the temp sensor until it pairs (up to 1 minute!!!)
7. You should see it pair in the logs and once the interview process finishes then you can stop pressing the button on the temp sensor.

I had to do step 1 - 7 for each temp/door sensor...

Once paired there's no issues, but there is clearly something pretty wrong with pairing these devices with later versions of hardware for some reason.

kwetnico, 2021-04-11:
> i had serious issues trying to pair some of my aqara/xiaomi temperature and door sensors with the new CC2652R zzh stick. No problems on my old cc2531.
>
> The approach i took was this (its extremely [...]

Hello, I have already carried out the following operations and nothing has done there .. Since version 1.18.x I have big problems with my conbee2 to add my temperature sensors, I have no problem with the others ...  to remove the batteries for 1 night and I will retest ... But it's weird I never had a problem adding the temperature sensors ...

kwetnico, 2021-04-11:
i have tested with the new xiaomi sensor human and no problem ... juste temperature ...i don't understand ... :(

Krazy998, 2021-04-11:
Had same problem. No issue with human sensors just door and temp sensors. They require very specific pairing approach which I outlined above. You need to follow it exactly and also maybe do it a couple of times before it pairs. 

It shouldn't be like that, but it's the only way I got them to pair...

kwetnico, 2021-04-11:
> Had same problem. No issue with human sensors just door and temp sensors. They require very specific pairing approach which I outlined above. You need to follow it exactly and also maybe do it a couple of times before it pairs.
> 
> It shouldn't be like that, but it's the only way I got them to pair...

sensors door for me no problem.... (remove or add...)

kwetnico, 2021-04-11:
> Had same problem. No issue with human sensors just door and temp sensors. They require very specific pairing approach which I outlined above. You need to follow it exactly and also maybe do it a couple of times before it pairs.
> 
> It shouldn't be like that, but it's the only way I got them to pair...

I just tested with your method:
Conbee2 + z2m 1.18.2 don't work
cc2531 + zm 1.16.2 or 1.18.2: don't work

kwetnico, 2021-04-11:
i have a temperature working but if connect to ZNCZ02LM (xiaomi) i don't have the information (battery ...)

Always juste 3 sensors temperature ok and 6 not detected.... (firmware ??)

kwetnico, 2021-04-11:
I found the problem ...

The batteries of my sensors were at 35% and apparently this blocks the pairing mode on the sensors ... I saw it with a friend who had the xiaomi gateway, she also refused, by putting a new battery the sensors I could add them !!!!! I put the batteries back to 35% to finish them and it works very well !!
Thank you

closed 2021-04-11 by kwetnico