Koenkk/zigbee2mqtt #7031: Remote connection to CC2652RB fails (but works with CC2531)

bertinets, 2021-04-10:
I'm opening an issue here, not sure if it's related to this one:
#6517

The adapter I have and the error I get are the same as in that issue. However, as it seems that that issue was solved from 1.18.0 and I am running 1.18.2, which is supposed to be working. On the other hand, I'm trying to connect remotely to the adapter, so that might be the source of the problem.
I have tried with Z2M 1.16.2, 1.17.0, 1.18.0, 1.18.2, 1.18.2-dev (commit #e91a23b) but I always get the same error.

## What happened
I have just bought a CC2652RB from Slaesh which I connected to a RPi (running Ubuntu 18.04.5 LTS) to upgrade a (working) CC2531 (also connected to the RPi).
I run a docker image of Zigbee2MQTT (v. 1.18.2) configured with minimal settings. 
The serial port of the CC2652RB is exposed to the network using ser2net.

ser2net config:
`3336:raw:0:/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_4F_3F_0E-if00-port0:115200 8DATABITS NONE 1STOPBIT`
or
`3336:raw:0:/dev/ttyUSB0:115200 8DATABITS NONE 1STOPBIT`

with both of the ser2net configs the adapter fails to connect (Error: SRSP - SYS - ping after 6000ms)

Log:
```
zigbee2mqtt@1.18.2-dev start /app
 node index.js

winston:create-logger: Define prototype method for "error"
winston:create-logger: Define prototype method for "warn"
winston:create-logger: Define prototype method for "info"
winston:create-logger: Define prototype method for "http"
winston:create-logger: Define prototype method for "verbose"
winston:create-logger: Define prototype method for "debug"
winston:create-logger: Define prototype method for "silly"
winston:create-logger: Define prototype method for "emerg"
winston:create-logger: Define prototype method for "alert"
winston:create-logger: Define prototype method for "crit"
winston:create-logger: Define prototype method for "error"
winston:create-logger: Define prototype method for "warning"
winston:create-logger: Define prototype method for "notice"
winston:create-logger: Define prototype method for "info"
winston:create-logger: Define prototype method for "debug"
Zigbee2MQTT:debug 2021-04-10 11:27:02: Can't load state from file /app/data/state.json (doesn't exist)
winston:file: written true false
Zigbee2MQTT:info  2021-04-10 11:27:02: Logging to console and directory: '/app/data/log/2021-04-10.11-27-01' filename: log.txt
2winston:file: written true false
winston:file: logged 91 debug 2021-04-10 11:27:02: Can't load state from file /app/data/state.json (doesn't exist)
winston:file: logged 206 info  2021-04-10 11:27:02: Logging to console and directory: '/app/data/log/2021-04-10.11-27-01' filename: log.txt

winston:file: ENOENT ok /app/data/log/2021-04-10.11-27-01/log.txt
winston:file: stat done: log.txt { size: 0 }
winston:file: create stream start /app/data/log/2021-04-10.11-27-01/log.txt { flags: 'a' }
winston:file: create stream ok /app/data/log/2021-04-10.11-27-01/log.txt
Zigbee2MQTT:info  2021-04-10 11:27:02: Starting Zigbee2MQTT version 1.18.2-dev (commit #e91a23b)
winston:file: written true false
Zigbee2MQTT:info  2021-04-10 11:27:02: Starting zigbee-herdsman (0.13.89)
winston:file: written true false
Zigbee2MQTT:debug 2021-04-10 11:27:02: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"tcp://192.168.2.191:3336"}}'
winston:file: written true false
 serialport/bindings loading LinuxBinding +0ms
 serialport/stream .list +0ms
winston:file: logged 85 info  2021-04-10 11:27:02: Starting Zigbee2MQTT version 1.18.2-dev (commit #e91a23b)
winston:file: logged 147 info  2021-04-10 11:27:02: Starting zigbee-herdsman (0.13.89)

winston:file: logged 563 debug 2021-04-10 11:27:02: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"tcp://192.168.2.191:3336"}}'

winston:file: file open ok /app/data/log/2021-04-10.11-27-01/log.txt
  serialport/stream .list +31ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"tcp://192.168.2.191:3336"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening TCP socket with 192.168.2.191:3336 +0ms
  zigbee-herdsman:adapter:zStack:znp:log Socket connected +3ms
  zigbee-herdsman:adapter:zStack:znp:log Socket ready +0ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
Zigbee2MQTT:error 2021-04-10 11:27:21: Error while starting zigbee-herdsman
winston:file: written true false
Zigbee2MQTT:error 2021-04-10 11:27:21: Failed to start zigbee
winston:file: written true false
Zigbee2MQTT:error 2021-04-10 11:27:21: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
winston:file: written true false
Zigbee2MQTT:error 2021-04-10 11:27:21: Exiting...
winston:file: written true false
Zigbee2MQTT:error 2021-04-10 11:27:21: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31 
  at Generator.throw (<anonymous>)
   at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
winston:file: written true false
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.2-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.18.2-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-04-10T09_27_21_339Z-debug.log
```

Running
`python3 ./test.py `
on the RPi returns:

```
Opening port /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_4F_3F_0E-if00-port0, baud 115200

--> Bootloader is working!
--> ZigBee coordinator is working!
`---> transportrev 2, product 1, majorrel 2, minorrel 7, maintrel 1, revision: 20210120

Check done!

```

## What did you expect to happen
I have another docker image running Z2M 1.18.2 that connects remotely to a CC2531 connected to the same RPi. With that adapter everything works absolutely fine, so I would expect Zigbee2MQTT to remotely connect to the Slaesh's CC2652RB adapter.

## How to reproduce it (minimal and precise)

My configuration.yaml:

```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://10.42.197.8'
serial:
  port: 'tcp://xxx.xxx.xxx.xxx:3336'
```

ser2net config (not working with the cc2652rb):
`3336:raw:0:/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_4F_3F_0E-if00-port0:115200 8DATABITS NONE 1STOPBIT`

ser2net config (working with the cc2531):
`3334:raw:0:/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00014D90A2-if00:115200 8DATABITS NONE 1STOPBIT`

ser2net version 2.10.1

docker-compose:

```
version: '2'
 services:
  zigbee2mqtt:
    image: forepe/zigbee2mqtt-socat
    environment:
      DEBUG: '*'
      DEBUG_VERBOSE: '1'
      TZ: Europe/Berlin
```

Another docker image with the same configuration.yaml (but for the port which is set to 3334) works with the cc2531

## Debug info
Zigbee2MQTT version: 1.16.2, 1.17.0, 1.18.2, 1.18.2-dev
Adapter hardware: CC2652RB
Adapter firmware version: both the 20210120 or the debug version of Slaesh

bertinets, 2021-04-11:
I confirm this is an issue with the remote binding: I have installed Z2M on the RPi and Z2M starts and connects to the CC2652RB adapter. 
It would be nice, though, if it could connect remotely as well.

derekoharrow, 2021-04-19:
I am running this configuration just fine. I believe you need a specific version of ser2net to get it working properly (it wouldn't work for me at first either).

I'm running ser2net v3.5

This is my set2net.conf
`20108:raw:0:/dev/ttyUSB0:115200 8DATABITS NONE 1STOPBIT
`

And this is my Z2M configuration.yaml
```
serial:
  port: 'tcp://hostname:20108'
```

bertinets, 2021-04-22:
Thanks for the hint, I really hoped the ser2net verision could have been the issue. However, although I upgraded now my ubuntu to 20.04 so I could install ser2net v. 3.5 (in ubuntu 18.04 only ser2net up to 2.10.1 is supported), I still get the same error, and again, only with the CC2652RB, the CC2531 works just fine...
I can't figure out where the problem is. I tried
`3336:raw:0:/dev/ttyUSB0:115200 8DATABITS NONE 1STOPBIT`
as well as
`3336:raw:0:/dev/serial/by-it/...:115200 8DATABITS NONE 1STOPBIT`
but none works...
And again, when installing Z2M directly on ubuntu everything works just fine...

derekoharrow, 2021-04-22:
I think I also had to add these to the z2m config file:

```
advanced:
  rtscts: false
  pan_id: 6755
```

bertinets, 2021-04-22:
Nothing, I added a pan_id and the rtscts: false lines but it's still giving the same error

closed 2021-05-30 by github-actions[bot]

maichai, 2021-07-19:
Same problem here, how did you solve it?

bertinets, 2021-07-19:
Unfortunately I didn't. Still couldn't remote connect the CC2652RB. The CC2531 works just fine under a comparable configuration... So I ended up installing Z2M directly on the RPi that physically hosts the CC2652RB. So the on the RPi, Z2M communicates to my mqtt docker container on my server and that sort of works.
I didn't check with the latest Z2M version though. I'll check in the next days again, if that works I'll post the results.

sjorge, 2021-08-01:
The bootloader needs to be skipped, there is a whole discussion on this here https://github.com/Koenkk/zigbee2mqtt/issues/6437

Basically you need a very new ser2net or to use a little node wrapper.