Koenkk/zigbee2mqtt #7037: CC2652RB Not Working

digimon-Denmark, 2021-04-10:
I just get a new CC2652RB with fireware zigbee2mqtt I've been trying for two days now to make it work. both on a RIP 3B + and an ubuntu machine laptop. i have followed all the guides around it and still nothing works.
my configuration.yaml file 

# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'localhost'
  # MQTT server authentication, uncomment if required:
  # user:         
  # password:                 

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_4F_1F_CD-if00-port0
  skipBootloader: true
\n\nadvanced:\n    network_key: GENERATE

advanced:
  log_level: 

> debug

my python3 ./test.py 
`File "./test.py", line 7
    <!DOCTYPE html>
    ^`

my sudo npm start 

`igbee2MQTT:debug 2021-04-10 22:37:32: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2021-04-10 22:37:32: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-04-10.22-37-32' filename: log.txt
Zigbee2MQTT:debug 2021-04-10 22:37:32: Removing old log directory '/opt/zigbee2mqtt/data/log/2021-04-10.21-56-47'
Zigbee2MQTT:info  2021-04-10 22:37:33: Starting Zigbee2MQTT version 1.18.2 (commit #abd8a09)
Zigbee2MQTT:info  2021-04-10 22:37:33: Starting zigbee-herdsman (0.13.88)
Zigbee2MQTT:debug 2021-04-10 22:37:33: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_4F_1F_CD-if00-port0"}}'
Zigbee2MQTT:error 2021-04-10 22:37:53: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-04-10 22:37:53: Failed to start zigbee
Zigbee2MQTT:error 2021-04-10 22:37:53: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-04-10 22:37:53: Exiting...
Zigbee2MQTT:error 2021-04-10 22:37:53: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.2 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.18.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/kasper/.npm/_logs/2021-04-10T20_37_53_848Z-debug.log`

Someone help me out

shirou93, 2021-04-11:
I have this stick on port:
port: /dev/ttyUSB0

Try without network_key: GENERATE.  You can try change everyone number on default key.

FaBRiK74, 2021-04-15:
> I have this stick on port:
> port: /dev/ttyUSB0

It is recommended not to use "/dev/ttyUSB0" but check the "Hardware" and use something like "/dev/serial/by-id/usb-..."

From slae.sh site:
"Another thing we need to take care of, is to use the "long" serial-port names!
Do NOT use the syntax like /dev/ttyUSBx.
Instead use the one like /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_A8_E3_95-if00-port0!

Why should we do this?
Cause the serial-port using the enumerated-syntax can (and may will) change!
Using the /dev/serial/by-id/.... syntax will never change.
And docker, or the running linux, will find the correct associated serial-port for our device! ;)

How can we check the correct name?
Go to the Hass.io/Supervisor-tab, go to System and click on the Hardware-button.

And please do this for all other USB-devices you are using, even in other add-on's. :)"

closed 2021-05-24 by github-actions[bot]