Koenkk/zigbee2mqtt #7038: brightness_step:0 does not update the state

olafvs, 2021-04-10:
## What happened
for start of dimming i use {"brightness_step": 254,"transition":5} and when I want to stop I use {"brightness_step": 0}. After {"brightness_step": 0} zigbee2mqtt does not update brightness to the values it is stopped (I stop it within the transition). It sets the brightness to 254

## What did you expect to happen
zigbee2mqtt updating the brightness value at the position it is being stopped at.

## How to reproduce it (minimal and precise)

1. start with a bulb being on and brightness of 2
2. set zigbee2mqtt/bulb/set {"brightness_step": 254,"transition":5}
3. stop before transistion times out using this zigbee2mqtt/bulb/set {"brightness_step": 0}
4. look at the brightness in zigbee2mqtt frontend for bulb
5. do zigbee2mqtt/bulb/get
6. look at the brightness in zigbee2mqtt frontend for bulb and compare. Second time it's correct.

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Koenkk, 2021-04-11:
> for start of dimming i use {"brightness_step": 254,"transition":5}

I don't understand this, `brightness_step` doesn't start dimming, it just increases/decreases the brightness with the given value. Do you mean `brightness_move` instead?

olafvs, 2021-04-11:
The combination of brightness_step and transition gives a dimming brighten effect on a bulb. Unable to test brightness_move right now.

olafvs, 2021-04-11:
Okay, just did a small test with brightness_move and I can use this as well. Thank you.

I thought brightness_step works in the same way. Setting brightness_step with a transistion starts the dimming and then firing  {"brightness_step": 0} actually stops this process. It just doesn't update the state like  {"brightness_move": 0} does. The way it's being handled right now isn't good I guess. I think the state should be updated. Either way thanks.

olafvs, 2021-04-16:
So I moved over to brightness_move for dimming/brightening. But this still causes issues. After the second brightening set my bulbs drift towards full brightness.

This is what I I send to a group that is already on:
1. {"state":"ON"} < result is ok
2. {"brightness_move":45} < result is ok
3. {"brightness_move":0} < result is ok
4. {"brightness_move":45} < result is ok
5. {"brightness_move":0} < result lights don't stop brightening

Here is the output of my logs:

`Debug Received MQTT message on 'zigbee2mqtt/Z_GR_Eetkamer/set' with data '{"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Eetkamer', payload '{"brightness":1,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'
Debug Received MQTT message on 'zigbee2mqtt/Z_GR_Eetkamer/set' with data '{"brightness_move":45}'
Debug Received MQTT message on 'zigbee2mqtt/Z_GR_Eetkamer/set' with data '{"brightness_move":0}'
Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Eetkamer', payload '{"brightness":1,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Eetkamer', payload '{"brightness":32,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'
Debug Received MQTT message on 'zigbee2mqtt/Z_GR_Eetkamer/set' with data '{"brightness_move":45}'
Debug Received MQTT message on 'zigbee2mqtt/Z_GR_Eetkamer/set' with data '{"brightness_move":0}'
Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Eetkamer', payload '{"brightness":32,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Eetkamer', payload '{"brightness":59,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'
`

Koenkk, 2021-04-17:
I've tried to reproduce the brightness_move issue but failed to do it with a Innr bulb. If this doesn't work properly I believe this is an issue of the bulb firmware, from Zigbee2MQTT we just send the native Zigbee brightness move commands.

I've fixed the brightness state not being correct with the brightness_step 0 now. Hopefully you can use that instead now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

olafvs, 2021-04-18:
First of all: thanks Koen,

Although Z2M throws an error now.

`Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Eetkamer', payload '{"brightness":65,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'
Debug Received Zigbee message from 'zitkamer_spot4', type 'readResponse', cluster 'genLevelCtrl', data '{"currentLevel":66}' from endpoint 11 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Woonkamer', payload '{"brightness":66,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/zitkamer_spot4', payload '{"brightness":66,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"linkquality":21,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Eetkamer', payload '{"brightness":66,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'
Error Publish 'set' 'brightness_step' to 'Z_GR_Eetkamer' failed: 'TypeError: Cannot read property 'currentLevel' of undefined'
Debug TypeError: Cannot read property 'currentLevel' of undefined at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:507:88) at async EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.js:224:36) at async Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:388:21)
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness_step' to 'Z_GR_Eetkamer' failed: 'TypeError: Cannot read property 'currentLevel' of undefined'","meta":{"friendly_name":"Z_GR_Eetkamer"},"type":"zigbee_publish_error"}'`

I have some lights that are in two groups, Z_GR_Eetkamer and Z_GR_Woonkamer. I only try to control Z_GR_Eetkamer in this case. Could this cause any trouble?

Koenkk, 2021-04-18:
Issue should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

olafvs, 2021-04-18:
Errors are gone. But I still noticed states not being updated. I looked a the code a bit and noticed that the brightness of the first device in the group is taken. So I looked if this devices was updated and it was. So only the first device in the group is being updated after both brightness_move and brightness_step.

Koenkk, 2021-04-19:
Can you provide the debug log again and you configuration.yaml? Indeed the brightness of the first device is taken but this value should be applied to all other members of the group.

olafvs, 2021-04-19:
`homeassistant: false
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://x.x.x.x'
serial:
  port: >-
    /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_4F_25_0E-if00-port0
  adapter: zstack
advanced:
  adapter_delay: null
  network_key:
    - 109
    - 222
    - 231
    - 10
    - 154
    - 245
    - 235
    - 195
    - 223
    - 129
    - 25
    - 32
    - 2
    - 172
    - 99
    - 111
  log_level: debug
  ikea_ota_use_test_url: false
  homeassistant_legacy_triggers: false
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
frontend:
  port: xxxx
  host: x.x.x.x
devices:
  '0x00178801103a2d6c':
    friendly_name: gang_hang1
  '0x0017880104f52014':
    friendly_name: Dimmer_Kantoor_1
    qos: 2
  '0x001788010495cbd8':
    friendly_name: zitkamer_spot1
'0x0017880104976f9b':
    friendly_name: zitkamer_spot2
  '0x0017880104977086':
    friendly_name: zitkamer_spot3
  '0x001788010495f0b4':
    friendly_name: zitkamer_spot4
  '0x0017880104972111':
    friendly_name: zitkamer_spot5
  '0x0017880104972110':
    friendly_name: zitkamer_spot6
  '0x0017880104972134':
    friendly_name: zitkamer_spot7
  '0x00178801103a4452':
    friendly_name: zitkamer_staand1
  '0x00178801105c602c':
    friendly_name: zitkamer_hang1
  '0x90fd9ffffe084635':
    friendly_name: eetkamer_hang1
[... 243 lines cut ...]
Debug Received MQTT message on 'zigbee2mqtt/Z_GR_Eetkamer/set' with data '{"brightness_step":254,"transition":5}'
Debug Publishing 'set' 'brightness_step' to 'Z_GR_Eetkamer'
Info MQTT publish: topic 'zigbee2mqtt/zitkamer_spot4', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"linkquality":57,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Eetkamer', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'
Debug Publishing 'set' 'transition' to 'Z_GR_Eetkamer'
Info MQTT publish: topic 'zigbee2mqtt/zitkamer_spot5', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/zitkamer_spot6', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/zitkamer_spot7', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/eetkamer_hang1', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"linkquality":66,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Woonkamer', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'
Debug Received Zigbee message from 'dimmer_eetkamer_zitkamer', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"commandID":20,"frameCounter":9416,"options":5280,"payloadSize":255,"srcID":24210295}' from endpoint 242 with groupID 2948
Info MQTT publish: topic 'zigbee2mqtt/dimmer_eetkamer_zitkamer', payload '{"action":"recall_scene_4","linkquality":57}'
Debug Received MQTT message on 'zigbee2mqtt/Z_GR_Eetkamer/set' with data '{"brightness_step":0}'
Debug Publishing 'set' 'brightness_step' to 'Z_GR_Eetkamer'
Debug Received Zigbee message from 'zitkamer_spot4', type 'readResponse', cluster 'genLevelCtrl', data '{"currentLevel":141}' from endpoint 11 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Woonkamer', payload '{"brightness":141,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/zitkamer_spot4', payload '{"brightness":141,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"linkquality":81,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Eetkamer', payload '{"brightness":141,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/zitkamer_spot4', payload '{"brightness":141,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"linkquality":81,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Eetkamer', payload '{"brightness":141,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/zitkamer_spot5', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/zitkamer_spot6', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/zitkamer_spot7', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/eetkamer_hang1', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"linkquality":66,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/Z_GR_Woonkamer', payload '{"brightness":141,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp":383,"state":"ON"}'`

olafvs, 2021-04-19:
I changed the only occurences and restarted. No luck.
 '0x0000000001716b77':
    friendly_name: dimmer_eetkamer_zitkamer
    filtered_attributes: []
    optimistic: true

I'm fairly new to zigbee, but to my understanding I am just catching the triggers from this switch it's not directly controlling anything. Everything goed to z2m > mqtt> node red >mqtt > z2m

Furthermore. I don't rememeber setting optimistic to false. Maybe I did but I guess I didn't. All other checkboxes for optimistic in the frontend aren't ticked either so I guess I pushed submit for another setting and it edited the default state of true. I'll check and if so create another issue.

Koenkk, 2021-04-20:
Did you restart z2m afterwards? If yes, can you provide the debug logging again when brightness stepping?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

olafvs, 2021-04-20:
Yes, I restareted and did one extra just now.

`debug 2021-04-20 17:30:34: Received Zigbee message from 'dimmer_eetkamer_zitkamer', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"command$
debug 2021-04-20 17:30:35: Received MQTT message on 'zigbee2mqtt/Z_GR_Eetkamer/set' with data '{"brightness_step":254,"transition":5}'
debug 2021-04-20 17:30:35: Publishing 'set' 'brightness_step' to 'Z_GR_Eetkamer'
info  2021-04-20 17:30:35: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot4', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:35: MQTT publish: topic 'zigbee2mqtt/Z_GR_Eetkamer', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp$
debug 2021-04-20 17:30:35: Publishing 'set' 'transition' to 'Z_GR_Eetkamer'
info  2021-04-20 17:30:35: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot5', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:35: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot6', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:35: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot7', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:35: MQTT publish: topic 'zigbee2mqtt/eetkamer_hang1', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:35: MQTT publish: topic 'zigbee2mqtt/Z_GR_Woonkamer', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
debug 2021-04-20 17:30:36: Received Zigbee message from 'dimmer_eetkamer_zitkamer', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame"$
info  2021-04-20 17:30:36: MQTT publish: topic 'zigbee2mqtt/dimmer_eetkamer_zitkamer', payload '{"action":"recall_scene_4","linkquality":75}'
debug 2021-04-20 17:30:36: Received Zigbee message from 'dimmer_eetkamer_zitkamer', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"command$
debug 2021-04-20 17:30:36: Received MQTT message on 'zigbee2mqtt/Z_GR_Eetkamer/set' with data '{"brightness_step":0}'
debug 2021-04-20 17:30:36: Publishing 'set' 'brightness_step' to 'Z_GR_Eetkamer'
debug 2021-04-20 17:30:38: Received Zigbee message from 'zitkamer_spot4', type 'readResponse', cluster 'genLevelCtrl', data '{"currentLevel":107}' from endpoint 11 wit$
info  2021-04-20 17:30:38: MQTT publish: topic 'zigbee2mqtt/Z_GR_Woonkamer', payload '{"brightness":107,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:38: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot4', payload '{"brightness":107,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:38: MQTT publish: topic 'zigbee2mqtt/Z_GR_Eetkamer', payload '{"brightness":107,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp$
info  2021-04-20 17:30:38: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot4', payload '{"brightness":107,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:38: MQTT publish: topic 'zigbee2mqtt/Z_GR_Eetkamer', payload '{"brightness":107,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp$
info  2021-04-20 17:30:38: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot5', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:38: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot6', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:38: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot7', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:38: MQTT publish: topic 'zigbee2mqtt/eetkamer_hang1', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:38: MQTT publish: topic 'zigbee2mqtt/Z_GR_Woonkamer', payload '{"brightness":107,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
debug 2021-04-20 17:30:43: Received Zigbee message from 'rechtsachter_spot4', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersi$
debug 2021-04-20 17:30:45: Received Zigbee message from 'dimmer_eetkamer_zitkamer', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame"$
info  2021-04-20 17:30:45: MQTT publish: topic 'zigbee2mqtt/dimmer_eetkamer_zitkamer', payload '{"action":"recall_scene_1","linkquality":93}'
debug 2021-04-20 17:30:45: Received Zigbee message from 'dimmer_eetkamer_zitkamer', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"command$
debug 2021-04-20 17:30:45: Received Zigbee message from 'dimmer_eetkamer_zitkamer', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame"$
debug 2021-04-20 17:30:45: Received Zigbee message from 'dimmer_eetkamer_zitkamer', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"command$
info  2021-04-20 17:30:45: MQTT publish: topic 'zigbee2mqtt/dimmer_eetkamer_zitkamer', payload '{"action":"recall_scene_5","linkquality":81}'
debug 2021-04-20 17:30:45: Received MQTT message on 'zigbee2mqtt/Z_GR_Eetkamer/set' with data '{"state":"OFF"}'
[... cut ...]
info  2021-04-20 17:30:46: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot7', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:46: MQTT publish: topic 'zigbee2mqtt/eetkamer_hang1', payload '{"brightness":254,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-20 17:30:46: MQTT publish: topic 'zigbee2mqtt/Z_GR_Woonkamer', payload '{"brightness":107,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
d`

olafvs, 2021-04-20:
Did a test with brightness_move and brightness_step on different groups. Both report the right brightness on the first bulb in the group but on the second bulb and onward they act different. While brightness_step:0 ends in the brighness_step calculated in the previous command (so either 254 or 1) brightness_move:0 does not update at all.

So for brightness_step second bulb in group current brightness= 10
set {"brightness_step": 254,"transition":5}
set {"brightness_step": 0}
result when I lookup the state for the device in Z2M: brightness 254

And for brightness_move second bulb in group current brightness= 10
set {"brightness_move": 45}
set {"brightness_move": 0}
result when I lookup the state for the device in Z2M: brightness 10

Koenkk, 2021-04-21:
I understood the `optimistic` feature wrong. The behaviour you are seeing is as documented: https://www.zigbee2mqtt.io/information/groups.html#state-changes

In case you also want to receive state updates for the other bulbs I suggest to use reporting: https://github.com/Koenkk/zigbee2mqtt/issues/7038#issuecomment-822671111

olafvs, 2021-04-21:
In that case I just don't undestand the documentation. 

It says: _Also when the state of a group is changed by a command (so not via a state change of a device in it; see previous line), all devices in the group will also change its state. This behaviour can be disabled by setting optimistic: false for the group._

So if I set the brightness of the group  all devices in the group should change unless optimistic is set to false. Optimistic should be true by default, but just to be sure I just set it explicitly in configuration.yaml. 

Could you please tell me how I misinterpret the documentation?
But even so. Wouldn't it be better to update the other states as well at the same time the first device in the group is updated? To me it sounds like a small loop through the group. In that case everything is reported with the right/actual states.

Koenkk, 2021-04-22:
Woops, you are totally right. Somehow I thought you were sending the brightness_step to the device, not to the group.

Issue should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

olafvs, 2021-04-22:
Thanks. It looks like that did the trick for brightness_step. Brightness_move isn't there yet.

Koenkk, 2021-04-23:
With the brightness_move issue you mean: https://github.com/Koenkk/zigbee2mqtt/issues/7038#issuecomment-821550492 ? If  yes, I believe this is a firmware problem of the bulbs which we cannot fix from the z2m side. (https://github.com/Koenkk/zigbee2mqtt/issues/7038#issuecomment-821828128)

olafvs, 2021-04-23:
No, it's this one :
[https://github.com/Koenkk/zigbee2mqtt/issues/7038#issuecomment-825130532](url)

Koenkk, 2021-04-23:
think I found & fixed the issue, can you test again?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

olafvs, 2021-04-23:
States are reported now, so that's great. Two remarks

1. I think some things are reported/done double now. See my log below.
2. I think the drifting of the lights got worse https://github.com/Koenkk/zigbee2mqtt/issues/7038#issuecomment-821550492. Not sure though. 

Can do some more testing on everything tomorrow.

`info  2021-04-23 20:28:06: MQTT publish: topic 'zigbee2mqtt/dimmer_eetkamer_zitkamer', payload '{"action":"recall_scene_2","linkquality":69}'
debug 2021-04-23 20:28:06: Received Zigbee message from 'dimmer_eetkamer_zitkamer', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame"$
debug 2021-04-23 20:28:06: Received MQTT message on 'zigbee2mqtt/Z_GR_Zitkamer/set' with data '{"brightness_move":-45}'
debug 2021-04-23 20:28:06: Publishing 'set' 'brightness_move' to 'Z_GR_Zitkamer'
debug 2021-04-23 20:28:06: Received Zigbee message from 'dimmer_eetkamer_zitkamer', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame"$
info  2021-04-23 20:28:06: MQTT publish: topic 'zigbee2mqtt/dimmer_eetkamer_zitkamer', payload '{"action":"recall_scene_6","linkquality":69}'
debug 2021-04-23 20:28:06: Received Zigbee message from 'dimmer_eetkamer_zitkamer', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"command$
debug 2021-04-23 20:28:06: Received MQTT message on 'zigbee2mqtt/Z_GR_Zitkamer/set' with data '{"brightness_move":0}'
debug 2021-04-23 20:28:06: Publishing 'set' 'brightness_move' to 'Z_GR_Zitkamer'
debug 2021-04-23 20:28:07: Received Zigbee message from 'linksachter_spot6', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersio$
debug 2021-04-23 20:28:09: Received Zigbee message from 'zitkamer_spot1', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID 0
info  2021-04-23 20:28:09: MQTT publish: topic 'zigbee2mqtt/Z_GR_Woonkamer', payload '{"brightness":127,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-23 20:28:09: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot1', payload '{"brightness":127,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_tem$
info  2021-04-23 20:28:09: MQTT publish: topic 'zigbee2mqtt/Z_GR_Zitkamer', payload '{"brightness":127,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp$
debug 2021-04-23 20:28:09: Received Zigbee message from 'zitkamer_spot1', type 'readResponse', cluster 'genLevelCtrl', data '{"currentLevel":87}' from endpoint 11 with$
info  2021-04-23 20:28:09: MQTT publish: topic 'zigbee2mqtt/Z_GR_Woonkamer', payload '{"brightness":87,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp$
info  2021-04-23 20:28:09: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot1', payload '{"brightness":87,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp$
info  2021-04-23 20:28:09: MQTT publish: topic 'zigbee2mqtt/Z_GR_Zitkamer', payload '{"brightness":87,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp"$
info  2021-04-23 20:28:09: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot1', payload '{"brightness":87,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp$
info  2021-04-23 20:28:09: MQTT publish: topic 'zigbee2mqtt/Z_GR_Zitkamer', payload '{"brightness":87,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp"$
info  2021-04-23 20:28:09: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot3', payload '{"brightness":87,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp$
info  2021-04-23 20:28:09: MQTT publish: topic 'zigbee2mqtt/zitkamer_spot2', payload '{"brightness":87,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp$
info  2021-04-23 20:28:09: MQTT publish: topic 'zigbee2mqtt/zitkamer_staand1', payload '{"brightness":87,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_te$
info  2021-04-23 20:28:09: MQTT publish: topic 'zigbee2mqtt/zitkamer_hang1', payload '{"brightness":87,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp$
info  2021-04-23 20:28:09: MQTT publish: topic 'zigbee2mqtt/Z_GR_Woonkamer', payload '{"brightness":87,"color":{"x":0.46737931469056,"y":0.412142647582159},"color_temp$
d`

olafvs, 2021-04-24:
Just did one more test let's skip the drifting part in my previous comment. I don't think it became more.
For brightness_move as you can see above in my logs I thought it reports double. But in fact that is because in the first instance it reports brightness 127 and in the second 87. I don't know how this is possible. 

Another thing I just found:
1. Group is on and brightness of all bulbs is 2 (value doesn't matter i guess)
2. Then I change brightness of one bulb in the group to 240 in the frontend of z2m
3. set brightness_move 45
4. set brightness_move:0
5. I look at brightness of the bulb I set through the frontend and it report a very low value. Let's say 47

It looks like the brightness of the bulb is taken from either the group or the first light in the group, not from the bulb itself

Koenkk, 2021-04-25:
The reason that you first get a brightness 127 report and then 87 is the following.
- When you send a brightens_move: 0, Z2M first reads the onOff state of the first bulb
- The bulb responds (`Received Zigbee message from 'zitkamer_spot1', type 'readResponse', cluster 'genOnOff'`) -> now the new on/off state will be send to MQTT including the previous known state (brightness 127). At this point we don't know the new brightness yet since we didn't read it yet.
- Next z2m reads the new brightness, bulb responds: `Received Zigbee message from 'zitkamer_spot1', type 'readResponse', cluster 'genLevelCtrl'` -> updated brightness will be send to MQTT. 

Since the brightness and on off state are in different Zigbee cluster we cannot read them simultaneously, therefore 2 state updates are send. If this is problematic for your use case you can use the `debounce` option but this currently doesn't work for groups: https://www.zigbee2mqtt.io/information/configuration.html#all-devices

> Another thing I just found:

This is expected behaviour, Zigbee2MQTT assumes all groups members share the same state. Therefore only the state of the first device is taken. Also when we would read all device states we generate a lot of messages which can impact the network performance. You can manually request an update of the state by sending to `zigbee2mqtt/MY_DEVICE/get` payload `{"brightness": ""}`.

olafvs, 2021-04-25:
Okay I get the part were you say brightness and on off state is in different clusters and that's why this is a two step process. Final remark on this: I think brightness_step does this differently. 

For the second part. I am a little bit worried. This means, I guess, that by controlling a group with brightness_move or brightness_step, you'll never get the right brightness of the individual lights, not in z2m, not in home assistant, not in node red, not in any other app. As I see it brightness_step and brightness_move are great ways of dimming/brightening a group with a minimal impact on performance.

Some thoughts on this
1.  How is something like Hue doing this?
2.  It would be a simple get on the group, that's one message. Of course it would be depending on the group size how many messages return.
3. Would it be sane to make this optional? In my view now I guess I would take the hit on performance in favour of states being right.

Koenkk, 2021-04-26:
> This means, I guess, that by controlling a group with brightness_move or brightness_step, you'll never get the right brightness of the individual lights, not in z2m, not in home assistant, not in node red, not in any other app.

This depends, I think in many uses cases all devices in a group share the same state (e.g. a few spots above a table) which are conceptually viewed as one light. Reading all states of the bulbs would disregard the performance aspect, in that case you can also control the lights individually. 

1. I don't know if it is possible to send commands like brightness move. However you can sniff a Hue network if you are interested: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html
2/3. You cannot do a get on the group. Zigbee messages to groups never return something (it's a broadcast). A possible feature would be to translate this get to a group in z2m to a get on all individual devices (which will be the same as doing a `/get` on each device via MQTT), currently we just do a get on the first device in the group.

olafvs, 2021-04-26:
>  A possible feature would be to translate this get to a group in z2m to a get on all individual devices (which will be the same as doing a /get on each device via MQTT), currently we just do a get on the first device in the group.

I think the get on a group is already translated. When I do a get via mqtt on a group I get the right brightness in z2m. The only problem with this is timing I guess. Is everything done on z2m side.

> I don't know if it is possible to send commands like brightness move. However you can sniff a Hue network if you are interested: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

The point I was trying to make is: it should be possible without flooding the network. Is the hue controller/hardware so much more powerful than a cc2652rb with many bulbs functioning as a router? 

> This depends, I think in many uses cases all devices in a group share the same state (e.g. a few spots above a table) which are conceptually viewed as one light. Reading all states of the bulbs would disregard the performance aspect, in that case you can also control the lights individually.

But I can also think of many use cases where individual lights in a group don't share the same brightness. The most important one being, after setting a scene I want to control the group with a dimmer. I know how this works now, but I don't think other members of the house will get this.

Don't get me wrong, I don't want to whine you into making this. I just want to discuss this. If the result is, I cannot do this, because the impact will be to high it's okay. I just want to let you know my point of view.

Koenkk, 2021-04-26:
> The point I was trying to make is: it should be possible without flooding the network. Is the hue controller/hardware so much more powerful than a cc2652rb with many bulbs functioning as a router?

I don't expect the Hue controller to be more powerful, but for sure Hue can make a lot more assumptions. E.g. you can pair only 50 devices to a Hue controller and they know that all the routers (Hue devices) are also powerful.

For sure there can be many other uses cases, but covering them all with 100% correctness and also dealing with the big variety of networks (possible low end controller like the CC2531, crappy devices on the network) can be hard. Therefore I try to limit the network requests as much as possible. I hope you can workaround this problem by adding some scripting to do a `/get`. If more people run into such issues we can always discuss how we can implement this best at the Z2M side. And ofc it's never bad to have the discussion, that only makes Z2M better :)

closed 2021-04-26 by olafvs