Koenkk/zigbee2mqtt #7043: Devices becoming stale and need repairing when running on EZSP v8

grinco, 2021-04-11:
## What happened
I've moved my entire network to EZSP (changed adapter). Ever since random devices become stale (don't report status, don't react to requests) and need repairing.

## What did you expect to happen
Naively enough, I was hoping of having a stable network, even though running on a development branch

## How to reproduce it (minimal and precise)
Just create a network and wait. I'm repairing stale devices every day since I moved to the development branch to give EZSP a try.

## Debug info
Zigbee2MQTT version:     1.18.2-dev commit: 1726899
Adapter hardware:    EZSP v8
 Adapter firmware version: Tasmota 9.3.1(zbbridge) (not sure about the zigbee adapter firmware)

grinco, 2021-04-11:
Just updated to 1.18.2-dev commit: 487ab06 will re-pair the affected devices and update the issue.

Koenkk, 2021-04-11:
The EZSP adapter integration is still in a very early stage (alpha). There is an issue here where you can discuss your findings: https://github.com/Koenkk/zigbee-herdsman/issues/319.

closed 2021-04-11 by Koenkk

Hedda, 2021-05-06:
> Adapter firmware version: Tasmota 9.3.1(zbbridge) (not sure about the zigbee adapter firmware)

@grinco  Also note that WiFi based Zigbee bridges are not recommended with the EZSP protocol. See the warnings about it:

https://www.zigbee2mqtt.io/information/supported_adapters.html

WiFi-based Serial-to-IP bridges are not recommended for Silicon Labs EZSP adapters as the serial protocol does not have enough fault-tolerance to handle packet loss or latency delays that can normally occur over WiFi connections. If cannot use a locally connected USB or UART/GPIO adapter then the recommendation is to use remote adapter that connected via Ethernet (wired) to avoid issues with EZSP caused by WiFi connections.

https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

WiFi-based Serial-to-IP bridges are not recommended for Silicon Labs EZSP adapters as the serial protocol does not have enough fault-tolerance to handle packet loss or latency delays that can normally occur over WiFi connections.

grinco, 2021-05-06:
Thanks for the reply. I've been thinking that it could have been the wifi
network at the root cause of my issues. Unfortunately the bridge doesn't
have an Ethernet connection, I'll check if it's possible to work around it
somehow.

Hedda, 2021-05-07:
> Thanks for the reply. I've been thinking that it could have been the wifi network at the root cause of my issues. 

FYI, I also understand that the EFR32 board antenna design of the "SM-011 V1.0" Zigbee module used inside ITead's Sonoff ZBBridge have very high receive sensitivity to electromagnetic interference (EMI) / radio-frequency interference. This means that it will be extra important to keep it away from any other electronic devices or appliances that may generate an electromagnetic field (EM Field) or send our interfering radio signals. That includes the computer that you run your home automation application software on as well as any devices such as for example external harddrives or storage devices, power-supplies, etc., and of course any Wi-Fi access points or routers nearby.

PS: Note that the new ITead Zigbee 3.0 USB Dongle Model: 9888010100045 (Hardware Revision Version 1.3) circuit board share the same type of antenna design so its radio is as well extremely sensitive to signal interference on its receiving end. The currect recommendation for using it is to connect it via a long USB extension cable and try to move it as far away from all electronics and other possible sources of signal interferance.

grinco, 2021-05-07:
Thanks, that helps. I've moved the bridge far from all sources of
electromagnetic interfectence and it did indeed help with the stability
issues. I'll keep an eye on it and consider moving the network to a
different device.

Hedda, 2021-05-07:
Another general tip that always applies when building a Zigbee network is to begin by adding a few good mains-powered Zigbee router devices so that they will make use a stable baseline. Many start by buying three "IKEA TRADFRI Signal Repeater" (E1746).

https://zigbee.blakadder.com/Ikea_E1746.html

https://www.ikea.com/us/en/p/tradfri-signal-repeater-30400407/

https://www.ikea.com/gb/en/p/tradfri-signal-repeater-80424255/

https://www.ikea.com/se/sv/p/tradfri-signalfoerstaerkare-10400408/

Tonio16, 2021-05-20:
Hello
I have the same problem but I noticed that by me it's affecting the ikea devices. If I unplug them and plug them again, they are again on the network for some times.
I have also a EZSP key.

Antoine