Koenkk/zigbee2mqtt #7046: Interrupteur dimmer legrand celiane

Jonathan77390, 2021-04-11:
Hello, I have a problem with a Legrand Céliane switch, the ON / OFF function works well, but not the dimmer function, and I lose the dimmer even physically on the button
Bonjour, j'ai un problème sur un interrupteur Legrand Céliane, la fonction ON/OFF marche bien, mais pas la fonction variateur, et je perd le variateur même physiquement sur le bouton !--
Before submitting an issue make sure you've searched for a similar issue and read the documentation: https://www.zigbee2mqtt.io/.

Rules (don't ignore these, your issue will be closed without further notice):
- English only
- Make sure you are on the latest Zigbee2MQTT version
- Provide a clear description of the problem
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.
- Make sure you are running the latest firmware https://github.com/koenkk/z-stack-firmware.

Did you read the FAQ?
- https://www.zigbee2mqtt.io/information/FAQ.html

Zigbee2MQTT fails to start?
- https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

Having issues when using a CC2531?
- Make sure the CC2531 is connected through a USB extension cable
- Try the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing
- With larger networks (30/40+) devices, your CC2531 might not be powerful enough. This will cause weird issues, in this case it's advised to use a more powerful adapter: https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-cc26x2r1

Unsupported device?
- https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Device does not pair or interview fails?
- https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

Bug report?
- If applicable, provide steps how to reproduce the problem.
- Provide the herdsman debug logging: https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

-->

## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

kristobal62, 2021-04-13:
Hello I have the same problem. I managed to activate the dimmer but the control does not work.
What do you need to help us.
Zigbee2MQTT version
1.18.2 commit: abd8a09
Coordinator type ConBee2/RaspBee2
Coordinator revision 0x26680700
Frontend version 0.3.114

`Debug Received MQTT message on 'zigbee2mqtt/inter_legrand_cuisine/set' with data '{"brightness":69}'
Debug Publishing 'set' 'brightness' to 'inter_legrand_cuisine'
Info MQTT publish: topic 'zigbee2mqtt/inter_legrand_cuisine', payload '{"brightness":69,"device":{"applicationVersion":0,"dateCode":" \u000020201001\u0000\u0000\u0000\u0000\u0000","friendlyName":"inter_legrand_cuisine","hardwareVersion":9,"ieeeAddr":"0x00047400008d7333","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067771","networkAddress":28793,"powerSource":"Mains (single phase)","softwareBuildID":"0026","stackVersion":66,"type":"Router","zclVersion":2},"linkquality":56,"state":"ON"}'
Debug Received MQTT message on 'zigbee2mqtt/inter_legrand_cuisine/set' with data '{"brightness":137}'
Debug Publishing 'set' 'brightness' to 'inter_legrand_cuisine'
Info MQTT publish: topic 'zigbee2mqtt/inter_legrand_cuisine', payload '{"brightness":137,"device":{"applicationVersion":0,"dateCode":" \u000020201001\u0000\u0000\u0000\u0000\u0000","friendlyName":"inter_legrand_cuisine","hardwareVersion":9,"ieeeAddr":"0x00047400008d7333","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067771","networkAddress":28793,"powerSource":"Mains (single phase)","softwareBuildID":"0026","stackVersion":66,"type":"Router","zclVersion":2},"linkquality":56,"state":"ON"}'`

thx

silver0007, 2021-04-24:
Hi,
have the same issue - I paired Legrand 067771 using CC2652 (Z2M 1.18.1).
State ON/OFF works fine. After enabling dimmer it works correctly when I use physical button. Brightness level is correctly passed through MQTT from the dimmer to HA in my case. However sending brightness level to the dimmer do not work. Brightness attribute gets updated in HA, but bulb do not react.

closed 2021-06-01 by github-actions[bot]

adnev, 2022-10-16:
Hi,

First of all thank you Koenkk for your work and all support provided by the community.

I open this ticket again because I get exactly the same problem on the same device Legrand Dimmer switch w/o neutral 067771.

My Zigbee2MQTT config is a little different : 

- Zigbee adpater : Slaesh CC2652RB (lastest  Zigbee2MQTT coordinator firmware 20220219)
- Zigbee2MQTT version [1.28.0] commit: [03ba647d]

I get error every time I try the following features : 

- device_mode - dimmer_off / dimmer_on
- led_in_dark - OFF / ON
- led_if_on - OFF / ON
- set the ballast_minimum_level or ballast_maximum_level

Here are the error logs in debug mode : 

`Debug 2022-10-16 20:43:09 Received MQTT message on 'zigbee2mqtt/salon_switch/set' with data '{"device_mode":"dimmer_on"}'
Debug 2022-10-16 20:43:09 Publishing 'set' 'device_mode' to 'salon_switch'
Error 2022-10-16 20:43:09 Publish 'set' 'device_mode' to 'salon_switch' failed: 'Error: Write 0x0004740000791a6d/1 manuSpecificLegrandDevices({"0":{"value":257,"type":9}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4129,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-10-16 20:43:09 Error: Write 0x0004740000791a6d/1 manuSpecificLegrandDevices({"0":{"value":257,"type":9}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4129,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:4700:13) at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:246:36)
Debug 2022-10-16 20:43:11 Received MQTT message on 'zigbee2mqtt/salon_switch/set' with data '{"led_in_dark":"ON"}'
Debug 2022-10-16 20:43:11 Publishing 'set' 'led_in_dark' to 'salon_switch'
Error 2022-10-16 20:43:11 Publish 'set' 'led_in_dark' to 'salon_switch' failed: 'Error: Write 0x0004740000791a6d/1 manuSpecificLegrandDevices({"1":{"value":true,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4129,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-10-16 20:43:11 Error: Write 0x0004740000791a6d/1 manuSpecificLegrandDevices({"1":{"value":true,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4129,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:4659:13) at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:246:36)
Debug 2022-10-16 20:43:13 Received MQTT message on 'zigbee2mqtt/salon_switch/set' with data '{"led_if_on":"ON"}'
Debug 2022-10-16 20:43:13 Publishing 'set' 'led_if_on' to 'salon_switch'
Error 2022-10-16 20:43:13 Publish 'set' 'led_if_on' to 'salon_switch' failed: 'Error: Write 0x0004740000791a6d/1 manuSpecificLegrandDevices({"2":{"value":true,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4129,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-10-16 20:43:13 Error: Write 0x0004740000791a6d/1 manuSpecificLegrandDevices({"2":{"value":true,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4129,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:4674:13) at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:246:36)
Debug 2022-10-16 20:43:15 Received MQTT message on 'zigbee2mqtt/salon_switch/set' with data '{"ballast_minimum_level":60}'
Debug 2022-10-16 20:43:15 Publishing 'set' 'ballast_minimum_level' to 'salon_switch'
Error 2022-10-16 20:43:16 Publish 'set' 'ballast_minimum_level' to 'salon_switch' failed: 'Error: Write 0x0004740000791a6d/1 lightingBallastCfg({"minLevel":60}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefau [... cut ...] l,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:600:17) at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:246:36)`