Koenkk/zigbee2mqtt #7049: EcoDim07 does not handle transitions (but says so in docs).

jelmerkk, 2021-04-11:
## What happened
My EcoDim07 dimmers don't respond to 'transition' commands, while the Z2M documentation says it supports it. When Z2M receives a transition, it doesn't send it to the dimmer. 

Log here: https://pastebin.com/Y0KKFWTc

## What did you expect to happen
That Z2M would relay the transition to the device

## How to reproduce it (minimal and precise)
Happens to every dimmer (I have 5)

## Debug info
Zigbee2MQTT version: 1.18.2
Adapter hardware: CC26X2R1
Adapter firmware version:

Koenkk, 2021-04-12:
If it doesn't work it means your device doesn't support it (however it might be that different firmwares on this device support it).

jelmerkk, 2021-04-14:
Talked to the manufacturer, indeed it is not (yet) supported. That's a bummer. I thought it was based on the docs.

closed 2021-04-14 by jelmerkk