Koenkk/zigbee2mqtt #7051: Neo NAS-AB02B0 support new hardware identify

leroyloren, 2021-04-12:
## What happened
I have a new siren, but after successful pairing it does not transmit any information (only LQ). According to the information obtained, this is a new version of HW, which the existing integration of Z2M does not support.

## What did you expect to happen
Z2M does not show any information from the device.

https://pastebin.com/6UVQu8gm

[image]
[image]

## How to reproduce it (minimal and precise)
Pair the device with the Z2M
## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, Conbee II
Adapter firmware version: zStack12 (20201127)

zbuh, 2021-04-12:
I also received this new version.
Logs in debug:

`Zigbee2MQTT:debug 2021-04-09 21:40:51: Received Zigbee message from '0x804b50fffe048344', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,10,17,0,27,68],"type":"Buffer"}' from endpoint 1 with groupID 0 Zigbee2MQTT:debug 2021-04-09 21:40:51: No converter available for 'NAS-AB02B0' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,10,17,0,27,68],"type":"Buffer"}'`

enigmaspb, 2021-04-12:
Some problem 
check inside board
NAS-AB02B0-V3
20190910

esion-net, 2021-04-12:
I ordered from 2 different shops at aliexpress, both had some trouble delivering in march (restock problem/version switch?), both delivered NAS-AB02B0-V3 version which is not working currently.

`Debug Received Zigbee message from '0x804b50fffe0482a8', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID null
Debug Received Zigbee message from '0x804b50fffe0482a8', type 'commandSetTimeRequest', cluster 'manuSpecificTuya', data '{"payloadSize":1792}' from endpoint 1 with groupID null
Debug No converter available for 'NAS-AB02B0' with cluster 'manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":1792}'
Debug Received Zigbee message from '0x804b50fffe0482a8', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,3,17,0,8,68],"type":"Buffer"}' from endpoint 1 with groupID null
Debug No converter available for 'NAS-AB02B0' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,3,17,0,8,68],"type":"Buffer"}'`

[-> Full log on pastebin](https://pastebin.com/52miAMUB)

[image]

dfrome, 2021-04-13:
hello,
upvote:
I think I have the same problem. Pairs perfectly, but only sends linkquality. Zigbee2mqtt logs show:
`No converter available for 'NAS-AB02B0' with cluster 'manuSpecificTuya'  and type 'raw' and data '{\"data\":[9,111,17,0,229,68],\"type\":\"Buffer\"}'`

leroyloren, 2021-04-18:
Hi, has anyone found anything new or should I look for another supported siren? :-)

Lefuneste83, 2021-04-19:
_Zigbee2MQTT version 1.18.2 (commit #abd8a092)
Coordinator firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190523,"transportrev":2},"type":"zStack30x"}'_

Hello. I confirm I see similar issues with the same HW version of this device (NAS-AB02B0-V3).
Pairing is fine and quite fast compared to usual Zigbee devices.

```
Zigbee2MQTT:info  2021-04-19 02:44:11: Successfully interviewed 'Alarme', device has successfully been paired
Zigbee2MQTT:info  2021-04-19 02:44:11: Device 'Alarme' is supported, identified as: Neo Temperature & humidity sensor and alarm (NAS-AB02B0)
Zigbee2MQTT:info  2021-04-19 02:44:11: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature & humidity sensor and alarm","exposes":[{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":3,"name":"humidity_alarm","property":"humidity_alarm","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Indicates if the battery of this device is almost empty","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"access":3,"name":"temperature_alarm","property":"temperature_alarm","type":"binary","value_off":false,"value_on":true},{"access":3,"name":"alarm","property":"alarm","type":"binary","value_off":false,"value_on":true},{"access":3,"name":"melody","property":"melody","type":"enum","values":["1","2","3","4","5","6","7","8","9","10","11","12","13","14","15","16","17","18"]},{"access":3,"name":"duration","property":"duration","type":"numeric","unit":"second"},{"access":3,"name":"temperature_min","property":"temperature_min","type":"numeric","unit":"°C"},{"access":3,"name":"temperature_max","property":"temperature_max","type":"numeric","unit":"°C"},{"access":3,"name":"humidity_min","property":"humidity_min","type":"numeric","unit":"%"},{"access":3,"name":"humidity_max","property":"humidity_max","type":"numeric","unit":"%"},{"access":3,"name":"volume","property":"volume","type":"enum","values":["low","medium","high"]},{"access":1,"name":"power_type","property":"power_type","type":"enum","values":["battery_full","battery_high","battery_medium","battery_low","usb"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"NAS-AB02B0","supports_ota":false,"vendor":"Neo"},"friendly_name":"Alarme","ieee_address":"0x804b50fffe04825f","status":"successful","supported":true},"type":"device_interview"}'

```

It also registers properly in my configuration file every time.
```
  '0x804b50fffe04825f':
    friendly_name: Alarme
```
Then it gets displayed properly in Z2MQTT-HA
[image]

And I can see the device and entities in HA
[image]

The entities are static and as if the device wasn't connected.

I have tried deleting the conf entry, repairing, restart everything... Same state is obtained.

I notice that upon restart Z2MQTT while definitely recognizing the device, sends a topic to MQTT in the wrong topic tree. See below : 
```
`Zigbee2MQTT:info  2021-04-19 02:29:26: Alarme (0x804b50fffe04825f): NAS-AB02B0 - Neo Temperature & humidity sensor and alarm (EndDevice)`

Zigbee2MQTT:info  2021-04-19 02:29:32: Connected to MQTT server
```

But while other devices get published properly under zigbee2mqtt topic in my MQTT queue, this device does not appear. As if either it is not sending any data.

esion-net, 2021-04-22:
Is exact the same behaviour for me too, @Lefuneste83 

The light stops after some time though. 

I really would like to help and get the new version up and running but as I've mentioned in another thread already - I was not able to properly debug cause none of the tutorials for adding new tuya devices acutally work. 

@Koenkk Can you please help out? :)

Koenkk, 2021-04-23:
@kirovilya do you know the reason for this?

zbuh, 2021-04-23:
I'm also available for help in this. I have a working sniffer if needed.

kirovilya, 2021-04-24:
you need to sniff traffic with a native gateway. I made for my (old) version of the siren (now I no longer have it).

dfrome, 2021-04-27:
Hope the other guys can do it. 
I have no native tuya gateway, else I would have sniffed already.
I bought this hardware because it was listed as supported in HA, so I was confident that I don't need a Tuya gateway.

leroyloren, 2021-04-29:
I wrote to the manufacturer, maybe it will provide us with more technical information :-)

dfrome, 2021-05-04:
> I'm also available for help in this. I have a working sniffer if needed.

hello zbuh,
what did you mean with this. Do you have a sniffer AND a Tuya gateway ? ( I understand the sniffer alone would not help, as it will not show the commandos expected by the device. )
regards.

zbuh, 2021-05-04:
No I don’t have the Tuya gateway.

kostas-gr, 2021-05-08:
Hello,
I also have the same issue. Very interested if you have made any progress yet

pepsinio, 2021-05-11:
Any update on this one? I looked into the logs, and get the feeling that the pairing process does not succeed correctly.

Boxter83, 2021-05-18:
Hi! I have the same problem with my Neo NAS-AB02B0, successful pairing but cant`t do anything. 
 
Looks like i also have V3

[image]

Enjoy-Combi, 2021-05-19:
+1, same issue wirh v3

yjp909, 2021-05-19:
The same issue...Pairs perfectly, but only sends linkquality, No any operration !

yjp909, 2021-05-19:
[image]

dfrome, 2021-05-20:
There are more and more people having this issue indeed.
@pepsinio @Boxter83 @Enjoy-Combi @yjp909 anyone has a tuya gateway, or know someone who has one ? Our HA specialists can help us only if someone can sniff the messages between our device and a tuya gateway.

Enjoy-Combi, 2021-05-20:
Hi,

Is this one could help ? https://fr.aliexpress.com/item/1005001585660474.html?spm=a2g0o.productlist.0.0.1a5050cbMqmusI&algo_pvid=null&algo_expid=null&btsid=0bb0622a16215001006558699e7122&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

I can buy it.

nmaster2042, 2021-05-20:
+1 same issue here.

dfrome, 2021-05-20:
> Hi,
> 
> Is this one could help ? https://fr.aliexpress.com/item/1005001585660474.html?spm=a2g0o.productlist.0.0.1a5050cbMqmusI&algo_pvid=null&algo_expid=null&btsid=0bb0622a16215001006558699e7122&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_
> 
> I can buy it.

I would say yes, this one should work : "Le hub vous permet de contrôler jusqu'à 32 appareils Tuya ZigBee".

BerndK, 2021-06-02:
As long as this might a duplicate to #7407 I want to point to my latest post: https://github.com/Koenkk/zigbee2mqtt/issues/7407#issuecomment-853120732
I'm now able to analyze the traffic. Please let me know if there is something missing.
To check if we have duplicated issues: my siren keeps blinking after pairing to zigbee2mqtt, it does stop bloinking when pairing it to tuya hub. Is this the same behavior as here?

nmaster2042, 2021-06-03:
@BerndK : yes it's same issue.

BerndK, 2021-06-03:
OK, how to go on?
As mentioned in my [post](https://github.com/Koenkk/zigbee2mqtt/issues/7407#issuecomment-853120732) I started analyzing the  messages paired with the original hub. On the first view (only tested some functions) there seems to be no difference. I will do a full test when pairing is really done.
Can someone confirm that the pairing is not complete (even when zigbee2mqtt reporst 'successful pairing', but blinking does not stop)? Are we on the right track? Is it the right idea, that the device 'waits' for answering command 0x24?
Or starts the problem a bit earlier (looking at the wireshark logs)? There are a bunch of (6) identical writes starting no 174-204. This is followed by 'Request Key' 228 and 'Verify Key' 242 - can't find these when pairing to zigbee2mqtt!?
Perhaps @Koenkk or someone else can assist or give some hints!?
I bet I can do more tests or change the code, but currently can't see clear what to do?

Sverto, 2021-06-03:
Regarding the pairing of the device, I can confirm it keeps blinking for a while after completing the process so it seems you're on the right track. Yet it might be important to note that in my case I'm using ZHA (yet the behavior and issues seem the same). I suppose ZHA is using the same Quirk as Zigbee2Mqtt is.

dfrome, 2021-06-04:
hello, I use Zigbee2mqtt and not ZHA. 
I can confirm as well: the device keeps blinking for a long time after completing the pairing process.

papanirual, 2021-06-07:
Hi, I have the same issue with this device. The device would be very handy for serveral tasks so it would be really great if it could be persuaded to correctly communicate with z2m.  :-)

Zigbee2MQTT version 1.19.1-dev (commit #9eb1fe61)
zigbee-herdsman (0.13.114)
CC26X2R1, Firmware: 20201017

Behaviour is the same as described above - blinking for a long time and even though it seems to be paired and its sensors and switches are discovered, they cannot be accessed.

Thanks in advance!

ayenz, 2021-06-08:
Have same version 3 and same issues as you guys described above 
[image]

jabber77, 2021-06-10:
Hello,
has anyone found a solution because like everyone else, this siren is not fully paired
thanks

marto68, 2021-06-12:
I too have a version 3, and same problem.

berndvos, 2021-06-17:
same here, I can pair, but the lights keep flashing, it says supported, but all values are n/a. Haven't opened the casing, but I guess I also have the V3 version.

michaelroussel2494, 2021-06-17:
Hello

Same issues here:

debug No converter available for 'NAS-AB02B0' with cluster 'manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":1536}'
debug No converter available for 'NAS-AB02B0' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,2,17,0,7,68],"type":"Buffer"}'

thanks

ivv72, 2021-06-22:
HELLO

I bought two of these sirenes from different places and both of them refuse to pair correctly
also please help
oh and with the tuya hub it works great you can even make a scene in the app and activate it from homeassistant so it does work a little, but not as it should.
thanks

kostas-gr, 2021-06-23:
Hello all,
Koenkk in order to support our request has requested someone with a Tuya
hub to sniff the messages with the siren.
Can someone who has a hub perform this action?
Thanks

ivv72, 2021-06-23:
+1 same issue here, I bought twoo v3's and they both dont work(bind) properly.
I hope you all can help me.
because it would be the cherry on top of my alarm system I want to make

igorpuchkovcom, 2021-06-23:
Same issue. Version 3.

dfrome, 2021-06-24:
> Hello all, Koenkk in order to support our request has requested someone with a Tuya hub to sniff the messages with the siren. Can someone who has a hub perform this action? Thanks

hello @kostas-gr ,
@BerndK has done exactly this and documented logs in #7407 .
I fear @Koenkk has no time for this issue.
I guess other people have knowledge about how to solve our problems, but I don't know how to raise their awareness.
Anybody with an idea ?

GrimDeclaw, 2021-06-25:
Hi,

I'm trying to understand @BerndK sniffing, and might have a lead! See my last post in #7407, not sure but might be something!

Tuan1984, 2021-07-06:
Just received mine! same thing pairs but no data.. let's get this done! lol

marto68, 2021-07-07:
Being Tuya, I'm to scared to connect mine to wifi just in case it updates the firmware negatively somehow.
Using the Tuya app, you would need to create scenes to activate it.
Which is not great.
I don't want my device to rely on the cloud.
Hopefully it will be added soon.

GrimDeclaw, 2021-07-07:
> Using the Tuya app, you would need to create scenes to activate it.

Hi,
Do you mean that the Tuya GW automatically creates scenes when you pair with it?
If so, that might be what's missing ... but still can't be sure....

marto68, 2021-07-07:
I mean connecting it to the Tuya app via wifi, 2.5ghz. Like you would lights and other devices, then you can create scenes that HASS finds. 
But I could be wrong, you might need Tuya Zigbee gateway to connect it.
As I said, I'm too scared to try

GrimDeclaw, 2021-07-07:
OK, I understand what you means. Thanks!

esion-net, 2021-07-15:
Just got a gateway today, might be able to run some tests and do some sniffing. Could use some help/guidance  though.

adrianofox, 2021-07-21:
Hi,
I have the same problem with this siren (version 3), it pairs with the z2m, completes the interview, but then it keeps blinking, sends only the linkquality.
The strangest thing is that it had already worked on z2m before (I had to pair it with tuya's gateway first, so it worked), but I changed my HA hardware and then it didn't work anymore.
Thanks.

GrimDeclaw, 2021-07-22:
> The strangest thing is that it had already worked on z2m before (I had to pair it with tuya's gateway first, so it worked), but I changed my HA hardware and then it didn't work anymore.
Hi,
You mean that a v3 siren worked with z2m before your upgrade right? Do you just change the hardware and restart with a backup or do you also update z2m? Maybe an unpredicted regression then, and might be worth trying oldest version, and then find what the gw send to the device

adrianofox, 2021-07-22:
Hi,

> You mean that a v3 siren worked with z2m before your upgrade right? 

Exactly.

>Do you just change the hardware and restart with a backup or do you also update z2m? 
>Maybe an unpredicted regression then, and might be worth trying oldest version, and then find what the gw send to the device

It crashed when I tried to pair it again with the new HA hardware but on the same z2m. I haven't tried to get an old backup back onto new hardware, it might be a good try.

michalpulda, 2021-07-22:
> Hi,
> I have the same problem with this siren (version 3), it pairs with the z2m, completes the interview, but then it keeps blinking, sends only the linkquality.
> The strangest thing is that it had [...]

I have seen this kind of behavior of a Tuya device described here https://github.com/zigpy/zha-device-handlers/issues/862#issuecomment-864629183

What has been observed is that after a device has been paired first to a Tuya gateway it is reporting fine when connected to other coordinator.

Pedder007, 2021-07-26:
Hi all, coming from the other ‚case‘, reading here, I now tend to buy a Tuya GW to get my multi-sensors (also  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4790) at least working with a ConBee stick. Is there any preference for the wireless or the wired (LAN) version of the Tuya GW? - just to tackle these pair-issues once per affected device...

landish-2021, 2021-07-26:
I confirm that after pairing the siren to the Tuya gateway and disconnecting it from the gateway, without resetting to factory settings, the siren is paired with the SS2652Z Zigbee2MQTT cordinator and outputs data on its sensors to the home assistant.

GrimDeclaw, 2021-07-26:
> I confirm that after pairing the siren to the Tuya gateway and disconnecting it from the gateway, without resetting to factory settings, the siren is paired with the SS2652Z Zigbee2MQTT cordinator and outputs data on its sensors to the home assistant.

Well, Good to know!!
Just to be sure, how do you pair the siren without factory resetting it? The only way I am aware of is "long press the button til the bip", which I'm guessing is the factory reset.

So the problem now is : what do the GW do on pairing that z2m doesn't?
@BerndK Do you think that your tuya GW sniffing contains what we need? I can't do any sniffing as I don't own a tuya GW :(

landish-2021, 2021-07-26:
After I pair the siren from the beginning to the Tuya gateway and then disconnect it from the Tuya gateway, without resetting to factory settings (just select turn off the siren), then the siren is paired with another coordinator, (we do not press the pairing button on the siren in any case) just turn on the stick for pairing devices, I have this SS2652R with Zigbee2MQTT, and the siren then normally pairs and gives correct data on its sensors in the home assistant.

adrianofox, 2021-07-26:
> After I pair the siren from the beginning to the Tuya gateway and then disconnect it from the Tuya gateway, without resetting to factory settings (just select turn off the siren), then the siren is [...]

You are right. I turned off the siren in the tuya app and then the z2m paired. Perfect. Thanks.

Pedder007, 2021-07-26:
Thanks a lot for the confirmation - so it seems to be regardless which Tuya GW I use. I will give it a try then to pair it with ConBee :-)

GrimDeclaw, 2021-07-26:
> You are right. I turned off the siren in the tuya app and then the z2m paired. Perfect. Thanks.

Just for information : the siren stop blinking after tuya gw pairing right? Did it starts blinking again when your turn it back on and if so, did it correctly stops blinking again after z2m pairing done?

adrianofox, 2021-07-26:
The siren starts blinking when you turn it off in the Tuya app. So when you activate the pairing mode on the z2m it appears on the HA (without pressing the button on the siren).

RaffoNin, 2021-07-27:
I feel like the problem and the workaround (use of gateway) should be on a separate issue so the developer can see it. It's possible other Tuya devices could have the same problem in the future.

[68 more replies]

closed 2021-11-13 by Koenkk