Koenkk/zigbee2mqtt #7052: Device Xiaomi WXCJKG13LM does not work as described

Sportslite, 2021-04-12:
## What happened
The device Xiaomi WXCJKG13LM (6 Buttons) does not send the correct action.
(The device Xiaomi WXCJKG12LM (4 Buttons) works like expected.)

I'm using HA with zigbee2mqtt plugin

## What did you expect to happen
The action property should contain e.g. button_1_double, button_1_triple, button_1_hold ...
but I always got button_1_single

thunder-62, 2021-04-16:
Hi,
I am using these switches without any issues. 

However, what I do: once one of those switches joins the network, I send {"operation_mode":"event"} to the switch to set it in the correct operation mode.

Sportslite, 2021-04-17:
> Hi,
> I am using these switches without any issues.
> 
> However, what I do: once one of those switches joins the network, I send {"operation_mode":"event"} to the switch to set it in the correct operation mode.

Maybe thats the problem, but I'm wondering why the 4 buttons switch works out of the box.
I'm using the node red plugin to set everything. How did you sent the operation mode to the switch?

thunder-62, 2021-04-17:
[{"id":"bae0bc3f.1273f","type":"mqtt in","z":"5c24e810.0110d8","name":"","topic":"zigbee2mqtt/bridge/event","qos":"0","datatype":"json","broker":"31fe20ce.d546e","x":170,"y":1820,"wires":[["192740c6.31a3cf","314564a0.030e8c"]]},{"id":"192740c6.31a3cf","type":"switch","z":"5c24e810.0110d8","name":"","property":"payload.type","propertyType":"msg","rules":[{"t":"eq","v":"device_announce","vt":"str"}],"checkall":"true","repair":false,"outputs":1,"x":350,"y":1820,"wires":[["c03a448e.d7a008","6ef638d.c29d6c8"]]},{"id":"314564a0.030e8c","type":"debug","z":"5c24e810.0110d8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":380,"y":1740,"wires":[]},{"id":"c03a448e.d7a008","type":"switch","z":"5c24e810.0110d8","name":"","property":"payload.data.friendly_name","propertyType":"msg","rules":[{"t":"regex","v":"Rollo.*","vt":"str","case":true}],"checkall":"true","repair":false,"outputs":1,"x":490,"y":1820,"wires":[["ba1700ed.6d633","9415787d.3f59f8"]]},{"id":"6ef638d.c29d6c8","type":"debug","z":"5c24e810.0110d8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":600,"y":1740,"wires":[]},{"id":"ba1700ed.6d633","type":"debug","z":"5c24e810.0110d8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":650,"y":1780,"wires":[]},{"id":"9415787d.3f59f8","type":"change","z":"5c24e810.0110d8","name":"","rules":[{"t":"set","p":"topic","pt":"msg","to":"\"zigbee2mqtt/\" & payload.data.friendly_name & \"/set\"","tot":"jsonata"},{"t":"set","p":"payload","pt":"msg","to":"{\"operation_mode\":\"event\"}","tot":"json"}],"action":"","property":"","from":"","to":"","reg":false,"x":660,"y":1820,"wires":[["98fc3d62.24f04","c59bb853.f79b18"]]},{"id":"98fc3d62.24f04","type":"mqtt out","z":"5c24e810.0110d8","name":"","topic":"","qos":"0","retain":"false","broker":"31fe20ce.d546e","x":860,"y":1820,"wires":[]},{"id":"c59bb853.f79b18","type":"debug","z":"5c24e810.0110d8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":880,"y":1780,"wires":[]},{"id":"31fe20ce.d546e","type":"mqtt-broker","z":"","name":"","broker":"192.168.10.88","port":"1883","clientid":"","usetls":false,"compatmode":false,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""}]

;-)

Sportslite, 2021-04-18:
Now its working fine. I managed it via the zigbee2mqtt plugin itself. There are two buttons "command" and "event". After pressing the event button, it works. Thanks.

thunder-62, 2021-04-18:
I am not sure whether the setting is persistent. Thus I added this in a rode-red flow... Maybe worth to check by remove /readd battery

Sportslite, 2021-04-19:
How did you sent it via node red? What node are you using?

thunder-62, 2021-04-21:
apologies for the late reply... I have pasted the flow I use here: https://github.com/Koenkk/zigbee2mqtt/issues/7052#issuecomment-821801691 just copy it, open the three bar menu on the top right in node red and import it...

Sportslite, 2021-04-25:
The import doesn't work:
[image]

closed 2021-05-04 by Sportslite