Koenkk/zigbee2mqtt #7056: input brightness value of color.HSB doesn't work correctly

baku104788, 2021-04-12:
To begin I was under the assumption that the brightness portion of HSB should be on a scale of 0-100. Let me know if this is incorrect. I did also experience issues attempting to scale from my usual 0-100 input to the 0-254 input used by the general brightness value.

When using the new color.hsb option to control the bulb I've noticed that the bulb will not turn on with the brightness value at 1 or 2.  It only starts working at 3.  I also attempted scaling it to be on 0-254 but still the first few values on the low end did not cause the bulb to come on and instead I had to stop using color.b and send the brightness attribute to the brightness attribute to get things working as expected.

sjorge, 2021-04-13:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/2453

This pending PR might fix your issue

baku104788, 2021-05-15:
Keeping this active until the next release when we can test the PR above is merged into release.

closed 2021-06-23 by github-actions[bot]