Koenkk/zigbee2mqtt #7063: Trust ZPIR-8000 Motion sensor is always Occupied

kadiusone, 2021-04-13:
## What happened
After adding Trust ZPIR-8000 to the network is reporting all the time to be Occupied. It is getting updates from the sensor when it detects motion but it is never set to Clear

## What did you expect to happen
Occupancy attribute should change to false after some time (60sec) when there is no motion detected and the status of the sensor is changed to Clear

## How to reproduce it (minimal and precise)

- Add ZPIR-8000 to the network
- Detect motion with a sensor
- leave it idle

## Debug info
Zigbee2MQTT version:  1.18.1-1
Adapter hardware: CC2530
Adapter firmware version: 20201127
Home Assistant: 2021.4.3

Sensor information:
Firmware build date: 2017-08-04
Firmware version: v0.1.0
Vendor: Trust
Model: ZPIR-8000

Kryzek, 2021-04-13:
I have this device with different firmware version and it's working as expected.
Firmware build date: 2017-09-12
Firmware version: v0.1.4

Wondering if the official Trust hub does provide firmware updates for Trust zigbee devices...

kadiusone, 2021-04-13:
I was expecting that this could be an issue. They do have OTA updates but using their coordinator. But still, I do see the update coming in from the sensor I need a way just to st it to Clear after some time.

Kryzek, 2021-04-13:
You could always try repairing (or reconfiguring) the device. It should give a update automatically if there is no motion after 30 (or so) seconds, just like it gives update when it detects motion.

```
Zigbee2MQTT:debug 2021-04-13 14:22:47: Received Zigbee message from 'Sensor_M_FD', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":34}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-04-13 14:22:47: MQTT publish: topic 'zigbee/Sensor_M_FD', payload '{"battery":70,"battery_low":false,"elapsed":5426,"last_seen":"2021-04-13T14:22:47+03:00","linkquality":51,"occupancy":true,"tamper":false}'

Zigbee2MQTT:debug 2021-04-13 14:23:07: Received Zigbee message from 'Sensor_M_FD', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-04-13 14:23:07: MQTT publish: topic 'zigbee/Sensor_M_FD', payload '{"battery":70,"battery_low":false,"elapsed":20100,"last_seen":"2021-04-13T14:23:07+03:00","linkquality":51,"occupancy":false,"tamper":false}'
```

kadiusone, 2021-04-13:
@Kryzek I would love to that but I need at least a hint on how I can do that with my minimal programming skills. 
I tried the Xiaomi Aquara fix but as this is a different device it simply does not work
`'0x00158d000195dc7f':
  friendly_name: '0x00158d000195dc7f'
  no_occupancy_since:
    - 10
    - 60
  occupancy_timeout: 60`

closed 2021-05-22 by github-actions[bot]