Koenkk/zigbee2mqtt #7064: Add symlink to active logging directory

daxy01, 2021-04-13:
Would it be possible to add a symlink in the log directory to the directory where the active logs are stored?
e.g. When starting z2m it creates the directory "2021-04-12.15-53-33". After a few restarts (due to testing for example), it becomes somewhat harder to find the right directory.

It would be ideal if, when z2m starts, **log/2021-04-12.15-53-33** is still be created, but then also a symlink is added for **log/current** which points to log/2021-04-12.15-53-33. Of course any existing symlink must be removed first. 

This way one can easily review logs under log/current/log.txt 😀

Unfortunately I am not able to change lib/util/logger.js myself to add this capability else I would have submitted a PR myself.
Thanks for considering!

daxy01, 2021-04-22:
I played around a bit with lib/util/logger.js and I believe this code should do the trick:

```
--- logger.js-orig	2021-04-22 12:21:30.530151710 +0200
+++ logger.js	2021-04-22 12:28:41.347002792 +0200
@@ -15,10 +15,14 @@
 const timestamp = moment(Date.now()).format('YYYY-MM-DD.HH-mm-ss');
 const directory = settings.get().advanced.log_directory.replace('%TIMESTAMP%', timestamp);
 const logFilename = settings.get().advanced.log_file.replace('%TIMESTAMP%', timestamp);
+const current = settings.get().advanced.log_directory.replace('%TIMESTAMP%', 'current');
+const actual = './' + timestamp;

 // Make sure that log directoy exsists when not logging to stdout only
 if (output.includes('file')) {
+    fs.unlink(current, function(err, result) { if(err) console.log('error', err); });
     fx.mkdirSync(directory);
+    fs.symlink(actual, current, function(err, result) { if(err) console.log('error', err); });
 }

 const z2mToWinstonLevel = (level) => level === 'warn' ? 'warning' : level;
```

I haven't used Github to submit a PR before, could someone check if this is ok and perhaps submit one on my behalf?

daxy01, 2021-04-22:
Just did my first PR 😀

Koenkk, 2021-04-25:
Merged, thanks and congratz with your first PR :smile:

closed 2021-04-25 by Koenkk