Koenkk/zigbee2mqtt #7065: Window open (0x4000) reporting stopped working between 1.18.1 -> 1.18.2

Zeddy-, 2021-04-13:
## What happened
Noticed window reporting stopped updating after updating from 1.18.1 to 1.18.2

Checked the reporting settings on the frontend and realized it was suddenly displaying Viessman related attribute names.

[image]

And i looked up this Viessmanncustom0 in the code and found this:
src/zcl/definition/cluster.ts
            StelproOutdoorTemp: {ID: 0x4001, type: DataType.int16, manufacturerCode: ManufacturerCode.Stelpro},
            viessmannCustom0: {ID: 0x4000, type: DataType.enum8, manufacturerCode: ManufacturerCode.VIESSMAN_ELEKTRO},

Is it now mapping my reporting incorrectly with the Viessman manufacturer code?

## What did you expect to happen

## How to reproduce it (minimal and precise)
Upgrade 1.18.1 to 1.18.2 / 1.18.2-dev

## Debug info
Zigbee2MQTT version: 1.18.2 commit: abd8a092
Adapter hardware: CC2531, CC2530, CC26X2R1
Adapter firmware version: varies.

Zeddy-, 2021-04-13:
I removes the viessman lines from cluster.ts and re-compiled. 

Looks like this in the reporting tab:
[image]

It started working immediately 
topic 'zigbee2mqtt/eTRV0100-67', payload '{"linkquality":81,"window_open_internal":1}'

Koenkk, 2021-04-13:
Could you provide the herdsman debug logging when readding the lines to cluster.ts and receiving a window open report?

bbayszczak, 2021-04-18:
Hi,

Looks like we're meeting a similar reporting issue with Eurotronics Spirit Zigbee #6995

Zeddy-, 2021-05-05:
> Could you provide the herdsman debug logging when readding the lines to cluster.ts and receiving a window open report?
> 
> See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

It's not only window open report that is failing, the viessman code is taking over my Danfoss thermostats almost completely.. 
Received Zigbee message from 'eTRV0100-1', type 'readResponse', cluster 'hvacThermostat', data '{"viessmannCustom10":1,"viessmannCustom7":4,"viessmannCustom8":660,"viessmannWindowOpenForce":0}' from endpoint 1 with groupID 0

There is clearly a bug in cluster.ts but i have no idea how to fix it other than just removing everything viessmann related.

Zeddy-, 2021-05-07:
I just tested and it seems my devices stop reporting in completely now. I can still request / read values from them, but changing the set point (occupied heating set point) manually, causes no traffic whatsover, even when running zigbee2mqtt with the herdsman debug activated. 

Fresh installation of zigbee2mqtt 1.18.2 and power cycled thermostats looks like this, so as you can see, there are some viessman variables in there that i am also not able to remove?

[image]

Zeddy-, 2021-05-15:
Hi @Koenkk 

turns out i had corrupt allys which caused alot of random behavior, now i factory reset them and got back to the main issue. Which is the invalid attribute names.

As you can see the Danfoss Ally is returning "viessmannWindowOpenForce":1

This is not the name documented for this device, it should be window_open_external

`Zigbee2MQTT:debug 2021-05-15 13:00:36: Received MQTT message on 'zigbee2mqtt/eTRV0100-2/get' with data '{"window_open_external":""}'
Zigbee2MQTT:debug 2021-05-15 13:00:36**: Publishing get 'get' 'window_open_external' to 'eTRV0100-2'**
  zigbee-herdsman:controller:endpoint Read 0xccccccfffe7c930c/1 hvacThermostat([16387], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4678,"transactionSequenceNumber":null,"writeUndiv":false}) +277ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,30,216] +6s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,30,216] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,30] - 216 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":30} +30s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,30,68,129,0,0,1,2,231,169,1,1,0,99,0,172] +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,30,68,129,0,0,1,2,231,169,1,1,0,99,0,172] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [49,201,0,0,10,12,70,18,11,1,3,64,0,16,1,231] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,30,68,129,0,0,1,2,231,169,1,1,0,99,0,172,49,201,0,0,10,12,70,18,11,1,3,64,0,16,1,231] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [169,29,248] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,30,68,129,0,0,1,2,231,169,1,1,0,99,0,172,49,201,0,0,10,12,70,18,11,1,3,64,0,16,1,231,169,29,248] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 30 - 2 - 4 - 129 - [0,0,1,2,231,169,1,1,0,99,0,172,49,201,0,0,10,12,70,18,11,1,3,64,0,16,1,231,169,29] - 248 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":513,"srcaddr":43495,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":99,"securityuse":0,"timestamp":13185452,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[12,70,18,11,1,3,64,0,16,1]}} +18ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":11,"manufacturerCode":4678,"commandIdentifier":1},"Payload":[{"attrId":16387,"status":0,"dataType":16,"attrData":1}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":43495,"endpoint":1,"linkquality":99,"groupID":0,"wasBroadcast":false}' +32s
Zigbee2MQTT:debug 2021-05-15 13:00:42: Received Zigbee message from 'eTRV0100-2', type 'readResponse', cluster 'hvacThermostat', data '{**"viessmannWindowOpenForce":1**}' from endpoint 1 with groupID 0
  zigbee-herdsman:controller:endpoint DefaultResponse 0xccccccfffe7c930c/1 513(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +6s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +13ms
`

Zeddy-, 2021-05-15:
Remove all viessman attributes from cluster.js causes the response to look like this instead:

```
Debug Publishing 'set' 'window_open_external' to 'eTRV0100-1'
Debug Received Zigbee message from 'eTRV0100-1', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":2811}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/eTRV0100-1', payload '{"linkquality":120,"local_temperature":28.11}'
Debug Received Zigbee message from 'eTRV0100-1', type 'attributeReport', cluster 'hvacThermostat', data '{"16387":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/eTRV0100-1', payload '{"linkquality":120,"window_open_external":true}'
```

Zeddy-, 2021-05-15:
Fixed the issue for myself by just renaming these and removing the manufacturer code (which doens't seem to do anything):

`            window_open_internal: { ID: 0x4000, type: dataType_1.default.enum8 },
            viessmannCustom1: { ID: 0x4001, type: dataType_1.default.int16, manufacturerCode: manufacturerCode_1.default.VIESSMAN_ELEKTRO },
            viessmannCustom2: { ID: 0x4002, type: dataType_1.default.int16, manufacturerCode: manufacturerCode_1.default.VIESSMAN_ELEKTRO },
            window_open_external: { ID: 0x4003, type: dataType_1.default.boolean },
            viessmannCustom4: { ID: 0x4007, type: dataType_1.default.boolean, manufacturerCode: manufacturerCode_1.default.VIESSMAN_ELEKTRO },
            viessmannCustom5: { ID: 0x4008, type: dataType_1.default.uint16, manufacturerCode: manufacturerCode_1.default.VIESSMAN_ELEKTRO },
            viessmannCustom6: { ID: 0x4009, type: dataType_1.default.uint16, manufacturerCode: manufacturerCode_1.default.VIESSMAN_ELEKTRO },
            viessmannCustom7: { ID: 0x4010, type: dataType_1.default.enum8, manufacturerCode: manufacturerCode_1.default.VIESSMAN_ELEKTRO },
            viessmannCustom8: { ID: 0x4011, type: dataType_1.default.uint16, manufacturerCode: manufacturerCode_1.default.VIESSMAN_ELEKTRO },
            etr_mounting: { ID: 0x4012, type: dataType_1.default.boolean },
            viessmannCustom10: { ID: 0x4020, type: dataType_1.default.uint8, manufacturerCode: manufacturerCode_1.default.VIESSMAN_ELEKTRO },
            viessmannCustom11: { ID: 0x4021, type: dataType_1.default.uint16, manufacturerCode: manufacturerCode_1.default.VIESSMAN_ELEKTRO },
            viessmannCustom12: { ID: 0x404b, type: dataType_1.default.int8, manufacturerCode: manufacturerCode_1.default.VIESSMAN_ELEKTRO },
            viessmannCustom13: { ID: 0x404c, type: dataType_1.default.enum8, manufacturerCode: manufacturerCode_1.default.VIESSMAN_ELEKTRO },
            viessmannCustom14: { ID: 0x404d, type: dataType_1.default.bitmap8, manufacturerCode: manufacturerCode_1.default.VIESSMAN_ELEKTRO },
            viessmannCustom15: { ID: 0x404e, type: dataType_1.default.bitmap8, manufacturerCode: manufacturerCode_1.default.VIESSMAN_ELEKTRO },`

Koenkk, 2021-05-15:
@Zeddy- thanks for your good investigation. If implemented a possible fix, can you check if it works in the latest dev branch now? (you can discard your changes)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

robertalexa, 2021-05-16:
OFFTOPIC - Apologies, but i have a question for a Danfoss Ally owner in order to decide if i need to do a PR for this device or not.

@Zeddy- I am currently looking at changing the Hive Radiator Valve based on the Danfoss Ally (which you own) and I need some info from a Danfoss Ally owner please (because it suspect the temperature range might be wrong in code). Could you please tell me when you physically turn the temperature on the TRV, what is the range that the device allows? Is it 6 to 28 degrees Celsius or 5 to 32?

Thank you kindly and sorry to barge in.

@Koenkk not sure if this will help you, but it might. While trying to do the Hive TRV support I have swapped to the dev branch where you said you have deployed a fix for the window open issue. Using the Danfoss Ally code (from danfoss.js) after pairing z2m crashes saying that attribute can't be found in cluster 513. This is for attr 16402 and 16384. I am not seeing this behaviour with the same code but in the master branch. But because I am seeing this while already messing up with code on a different device, I am unsure if this is actually a problem or a side effect. The device I am working on is Hive TRV001 UK7004240. Hope this helps you

robertalexa, 2021-05-17:
@Koenkk I am not sure your bug fix is necessary, but I am not here to contest you. I am currently working on this and the device seems to be a bit misconfigured, and using attributes from a different device (the veisman - they share the addresses so it is fine) but the output is not matched

Have a look at this commit please and tell me if I am wrong. https://github.com/robertalexa/zigbee-herdsman-converters/commit/dfccce4774f6356560927363be55d537aabcd6cd

I plan on doing a PR with all my changes to Danfoss Ally and Hive TRV as soon as I am done. I am following the data spec of Danfoss but I do not have a Danfoss devie to test, just the Hive (which uses a different firmware but so far most of the commands match)

Koenkk, 2021-05-17:
@robertalexa that solution will also work. But I believe my fix is better since things will not break again when another attribute is added with the same ID.

robertalexa, 2021-05-17:
@Koenkk I am having more than 1 attribute hex id behaving the same way, failing to get the data. However if I use the name it corrects them all. Am I good to keep this fix for this device or would you rather have me remove it.
```
if (typeof msg.data['viessmannCustom0'] == 'number') {
                result[postfixWithEndpointName('window_open_internal', msg, model)] = (msg.data['viessmannCustom0']);
            }
            if (typeof msg.data['viessmannWindowOpenForce'] == 'number') {
                result[postfixWithEndpointName('window_open_external', msg, model)] = (msg.data['viessmannWindowOpenForce'] == 0x01);
            }
if (typeof msg.data['viessmannAssemblyMode'] == 'number') {
                result[postfixWithEndpointName('mounted_mode_active', msg, model)] = (msg.data['viessmannAssemblyMode']==1);
            }
if (typeof msg.data['viessmannCustom10'] == 'number') {
                result[postfixWithEndpointName('algorithm_scale_factor', msg, model)] = msg.data['viessmannCustom10'];
            }
```
The above is an extract of just the ones that needed changes. The data in clusters is correct, the Danfoss and Hive use the same addresses as the viessmann, but whoever first implemented this didn't do proper naming conventions, so it is all messy now.

Offtopic: could you please help me with 2 things in my testing:
1. Is there a nice way of symlinking zigbee-herdsman-converters as part of z2h so I can "live" test my fork instead of editing node_modules code? I have tried symlinking but running npm i in the fork broke everything.
2. How can I test my fork on homeassistant using the z2m addon but my fork of converters?

Thanks,
Rob

Koenkk, 2021-05-17:
@robertalexa can you provide the data/database.db entry of your thermostat?

robertalexa, 2021-05-17:
@Koenkk 
```
"id":2,"type":"EndDevice","ieeeAddr":"0x842e14fffe203c4c","nwkAddr":59431,"manufId":4678,"manufName":"Danfoss","powerSource":"Battery","modelId":"TRV001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,10,32,513,516,2821],"outClusterList":[0,25],"clusters":{"genBasic":{"attributes":{"modelId":"TRV001","manufacturerName":"Danfoss","powerSource":3,"zclVersion":3,"appVersion":33,"stackVersion":8,"hwVersion":69,"dateCode":"20200402","swBuildId":"02.33.0008 02.33"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":106}},"hvacThermostat":{"attributes":{"16403":0,"16432":1,"16433":0,"16458":0,"localTemp":2471,"viessmannWindowOpenForce":0,"viessmannCustom7":4,"viessmannCustom8":660,"viessmannCustom10":1,"occupiedHeatingSetpoint":2100,"pIHeatingDemand":0,"viessmannCustom14":0,"viessmannCustom12":0,"viessmannCustom0":0,"viessmannAssemblyMode":0}},"hvacUserInterfaceCfg":{"attributes":{"keypadLockout":0}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e32e48","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e32e48","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":60,"maxRepIntval":43200,"repChange":1},{"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":600,"repChange":25},{"cluster":513,"attrId":8,"minRepIntval":0,"maxRepIntval":600,"repChange":1},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":600,"repChange":25},{"cluster":513,"attrId":16402,"minRepIntval":0,"maxRepIntval":600,"repChange":1},{"cluster":513,"attrId":16384,"minRepIntval":0,"maxRepIntval":3600,"repChange":1}],"meta":{}}},"appVersion":33,"stackVersion":8,"hwVersion":69,"dateCode":"20200402","swBuildId":"02.33.0008 02.33","zclVersion":3,"interviewCompleted":true,"meta":{"configured":4},"lastSeen":1621268211449}
```

This is for Hive TRV. This is the issue I am currently on (trying to fix everything about this device and it's sister Danfos Ally) https://github.com/Koenkk/zigbee2mqtt/issues/7461

Koenkk, 2021-05-17:
@robertalexa I see the manfucaturerCode is different. Can you provide the herdsman debug log **on the latest dev branch** when receiving these attributes?

robertalexa, 2021-05-17:
@Koenkk The manufacturer code is different but the physical device is 100% the same as Danfoss Ally. It is running a custom closed firmware by Hive.

With regards to the logs, I will have to clone the repo again and reconfigure everything, but I am not sure I will see what you want. Without my PR most of those properties will not work (that was on z2m master).

Koenkk, 2021-05-17:
@robertalexa I see but with https://github.com/Koenkk/zigbee-herdsman/commit/3fd91d14810cc9f5541f56fd91dd5a1e0415f4ce in this won't work anymore if the manufacturer code is different. Therefore I ask you to test and provide logs on the latest dev

robertalexa, 2021-05-17:
@Koenkk This device is not properly supported, nothing like the Danfoss. So if you want me to see those attributes I have to bring code from my PR into the dev branch.

Here is how the device looks like with the current code 
[image]

That is nothing like Danfoss Ally.
[herdsman.log](https://github.com/Koenkk/zigbee2mqtt/files/6496523/herdsman.log)

I am attaching the log but those attributes are not present.

Let me configure the device using the Danfoss js but without any of my changes and I will come back with fresh logs.
Rob

robertalexa, 2021-05-17:
[herdsman2.log](https://github.com/Koenkk/zigbee2mqtt/files/6496558/herdsman2.log)

Just trying that.

As mentioned to you before, on the dev branch, with the Dafoss Ally code, with no further changes npm throws an error after interview because key is not found

```
Zigbee2MQTT:info  2021-05-17 20:58:27: Successfully configured '0x842e14fffe203c4c'
/home/rob/Projects/z2h-dev/node_modules/zigbee-herdsman/dist/zcl/utils.js:84
            throw new Error(`Cluster '${name}' has no attribute '${key}'`);
                  ^

Error: Cluster 'hvacThermostat' has no attribute '16402'
    at Object.getAttribute (/home/rob/Projects/z2h-dev/node_modules/zigbee-herdsman/dist/zcl/utils.js:84:19)
    at /home/rob/Projects/z2h-dev/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:87:37
    at Array.map (<anonymous>)
    at Endpoint.get configuredReportings [as configuredReportings] (/home/rob/Projects/z2h-dev/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:83:43)
    at /home/rob/Projects/z2h-dev/lib/extension/bridge.js:602:60
    at Array.map (<anonymous>)
    at Bridge.publishDevices (/home/rob/Projects/z2h-dev/lib/extension/bridge.js:574:50)
    at EventBus.<anonymous> (/home/rob/Projects/z2h-dev/lib/extension/bridge.js:52:18)
    at EventBus.emit (node:events:378:20)
    at EventBus.emit (/home/rob/Projects/z2h-dev/lib/eventBus.js:23:15)
npm ERR! code 1
npm ERR! path /home/rob/Projects/z2h-dev
npm ERR! command failed
npm ERR! command sh -c node index.js

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/rob/.npm/_logs/2021-05-17T19_58_27_868Z-debug.log
```

In this case 16402 is 0x4012 which matches attribute viessmannAssemblyMode

Hope this helps

robertalexa, 2021-05-17:
Here is another log from fresh.
[herdsman3.log](https://github.com/Koenkk/zigbee2mqtt/files/6496625/herdsman3.log)

I am unable to send load the device on the Frontend because of the attributes error.

Here is a screenshot of the device with my own fork for comparison, to understand how different my device is between the production code and my fork (based on danfoss)
[image]

If I may make a suggestiong @Koenkk , would it be worth you undoing your bug fix, we review my PR and by that my device will be properly supported, the the Danfoss Ally will also be fully working. From there we can review the code and see what we can change?

As it stands now on dev, your change is breaking the danfoss completely by the looks of it, as in apart from that, the danfoss code will need revision. This is why I am suggesting the above, then at least we start from a working state.

Let me know how would you like us to proceed, my end result is to get the Hive TRV supported as that is what I am using in my house.

Rob

Koenkk, 2021-05-18:
@robertalexa 

> As mentioned to you before, on the dev branch, with the Dafoss Ally code, with no further changes npm throws an error after interview because key is not found

Pushed a fix for this in the dev branch. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

> If I may make a suggestiong @Koenkk , would it be worth you undoing your bug fix, we review my PR and by that my device will be properly supported, the the Danfoss Ally will also be fully working. From there we can review the code and see what we can change?

I believe my fix is better for the future, better go for a nice solution now otherwise this will probably break again in the future.

robertalexa, 2021-05-18:
@Koenkk Sounds good. I will see how your changes impact my code and make the necessary amends. Hopefully you can then review my PR. I would really appreciate your help with a command.

Also, would you be kind to answer my 2 questions:
1. Is there a nice way of symlinking zigbee-herdsman-converters as part of z2h so I can "live" test my fork instead of editing node_modules code? I have tried symlinking but running npm i in the fork broke everything.
2. How can I test my fork on homeassistant using the z2m addon but my fork of converters?

Koenkk, 2021-05-18:
@robertalexa for this case I recommend to use an external converter, https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#2-adding-your-device e.g. the advanced example: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/externalConvertersExample/freepad_ext.js

You need to adjust the zigbeeModel temporarily, otherwise this won't work. So change it in `database.db` to e.g. `my_custom_thermostat` and use the following in the external converter: `zigbeeModel: ['my_custom_thermostat'],`

robertalexa, 2021-05-18:
@Koenkk Yeah I am aware of that option, but was hoping for something else. But that might work, I will play with it when I get there.

Your new bug fix on dev DOES fix this issue. So all good on this side of things.

I have just gone through my changes and pushed a few more commits to my PR.

Can we please continue our chat there to improve the Danfoss Ally and properly support the Hive TRV.

Rob

robertalexa, 2021-05-22:
@Zeddy- Fixes and significant improvements to Danfoss Ally are now available on the dev branch of z2m.

Would you be kind to run some tests and if everything goes well, close this issue?

Thanks,
Rob

closed 2021-06-30 by github-actions[bot]