Koenkk/zigbee2mqtt #7067: battery_low and tamper never published for SJCGQ11LM

McGiverGim, 2021-04-13:
## What happened

The SJCGQ11LM sensor (Xiaomi water leak) defines `battery_low` and `tamper` attributes, but it seems they are never published. I think this device does not publish this values, so it seem they must be removed.
Or at least my version does not publish them. I don't know if there are variants and some of them does it.

## What did you expect to happen

This attributes have some value or not exist at all.

## How to reproduce it (minimal and precise)

Look at the UI:
[image]
[image]

## Debug info
Zigbee2MQTT version: Hassio addon 1.18.2-1
Adapter hardware: CC2531
Adapter firmware version: zStack12.20201128

closed 2021-04-13 by Koenkk

VOVAiS, 2021-04-17:
I do not agree
the sensor sends data
but sends them only if it finds a leak
[image]
[image]

McGiverGim, 2021-04-17:
Ups, I didn't notice it... this is closed, I don't know if @Koenkk will see an close issue. Maybe is better to open a new issue referencing this one.

VOVAiS, 2021-04-17:
Do not open a new Issue
  I think @Koenkk will see

Koenkk, 2021-04-17:
@VOVAiS thanks, reverted!