Koenkk/zigbee2mqtt #7074: Button actions missing in HA

derekoharrow, 2021-04-14:
I've some Xiaomi Aqara WXKG12LM buttons that have been working fine for ages now. In the last day or so, the HA scripts for them have stopped working. On investigation, it appears that in the Home Assistant Automations, the Single, Double, Hold and Shake Device Triggers are not appearing in Home Assistant. It only shows the "battery level changes" and "voltage changes" triggers, and not any triggers . 

Everything else seems to be behaving as expected - I just can't select the Device Triggers when creating an automation as they're not listed. I've had to resort to using MQTT Triggers instead of Device Action triggers for these switches.

I've tried un-pairing and re-pairing these devices and they still don't show the device action triggers in HA.

Here's an except of the MQTT payloads:

**zigbee2mqtt/Cabin Button**
`{"battery":80,"last_seen":"2021-04-14T12:25:19+01:00","linkquality":33,"voltage":2965}`

**zigbee2mqtt/Cabin Button/action**
`single`

**homeassistant/sensor/0x00158d000429e66a/action/config**
`{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Cabin Button/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000429e66a"],"manufacturer":"Xiaomi","model":"Aqara wireless switch (with gyroscope) (WXKG12LM)","name":"Cabin Button","sw_version":"Zigbee2MQTT 1.18.2"},"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/Cabin Button","name":"Cabin Button action","state_topic":"zigbee2mqtt/Cabin Button","unique_id":"0x00158d000429e66a_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}`

**homeassistant/sensor/0x00158d000429e66a/click/config**
`{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Cabin Button/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000429e66a"],"manufacturer":"Xiaomi","model":"Aqara wireless switch (with gyroscope) (WXKG12LM)","name":"Cabin Button","sw_version":"Zigbee2MQTT 1.18.2"},"icon":"mdi:toggle-switch","json_attributes_topic":"zigbee2mqtt/Cabin Button","name":"Cabin Button click","state_topic":"zigbee2mqtt/Cabin Button","unique_id":"0x00158d000429e66a_click_zigbee2mqtt","value_template":"{{ value_json.click }}"}`

**homeassistant/sensor/0x00158d000429e66a/battery/config**
`{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Cabin Button/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000429e66a"],"manufacturer":"Xiaomi","model":"Aqara wireless switch (with gyroscope) (WXKG12LM)","name":"Cabin Button","sw_version":"Zigbee2MQTT 1.18.2"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/Cabin Button","name":"Cabin Button battery","state_topic":"zigbee2mqtt/Cabin Button","unique_id":"0x00158d000429e66a_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}`

**homeassistant/sensor/0x00158d000429e66a/voltage/config**
`{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Cabin Button/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000429e66a"],"manufacturer":"Xiaomi","model":"Aqara wireless switch (with gyroscope) (WXKG12LM)","name":"Cabin Button","sw_version":"Zigbee2MQTT 1.18.2"},"device_class":"voltage","json_attributes_topic":"zigbee2mqtt/Cabin Button","name":"Cabin Button voltage","state_topic":"zigbee2mqtt/Cabin Button","unique_id":"0x00158d000429e66a_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}`

And here's the device in HA:

**sensor.cabin_button_action**
```
battery: 80
click: ''
last_seen: '2021-04-14T13:08:13+01:00'
linkquality: 30
voltage: 2965
friendly_name: Cabin Button action
icon: 'mdi:gesture-double-tap'
hide_attributes:
  - templates
  - icon_color
```

## Debug info
Zigbee2MQTT version: 1.18.2 commit: abd8a09
Adapter hardware: CC26X2R1
Adapter firmware version: zStack3x0 / 20210120
Home Assistant version: 2021.4.4

derekoharrow, 2021-04-14:
I think I've found what's causing this, but I'm not sure why...

To get around the errors being logged in home assistant for missing attributes, I added the following to my configuration.yaml

```
device_options:
  homeassistant:
    linkquality: null
    illuminance_lux: null
```

This seemed to prevent the pushing of the **homeassistant/device_automation** topic in MQTT.

I've commented-out these lines now, and the device triggers are back once again!

Koenkk, 2021-04-14:
Think I found the bug, can you check it?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

derekoharrow, 2021-04-19:
I've just checked it out and it does now create the device_automation topics even if I include the device_options specified above, so it does look like it's fixed it. Thanks

closed 2021-04-19 by derekoharrow

drumstick77, 2022-03-17:
Hiya, @derekoharrow @Koenkk  - I'm still having the same issue. I've installed the DEV and I'm not getting any option for WXKG12LM. I had the normal one before installing the dev which performed the same behaviour. any thoughts? thanks

wuast94, 2022-04-01:
i have the same problem, i dont see any sensor in home assistant from buttons, i see the link quality and battery etc but no sensor where i can see the button presses

Tipalo, 2022-04-05:
[image]

Same issue here.
My systems drop list does not show any action property.   On another system with the same adapter and version of Zigbee2MQTT there is a drop list with a complete list of buttons.

drumstick77, 2022-04-05:
I've managed to sort this guys. What I did was, once I paired it, before I add it into any automation, I press the button in all its varieties to register them into HA. Single press, double, hold and release. As soon as I've done them, I then could s we them available. 
Hope that helps.

dany547, 2022-05-06:
> I've managed to sort this guys. What I did was, once I paired it, before I add it into any automation, I press the button in all its varieties to register them into HA. Single press, double, hold and release. As soon as I've done them, I then could s we them available. Hope that helps.

Yus! Thanks! It worked for me

rchovan, 2022-08-11:
this error is still available in the latest releases. I have action working correctly for several days, but now I'm missing not all, but some actions. For example off-press for Philips dimmer 

[image]

In Z2M I can see action 
'zigbee2mqtt/guestroom_dimmer', payload '{"action":"off-press","action_duration":null,"battery":54.5,"brightness":191,"counter":1,"duration":0,"linkquality":255,"update":{"state":"idle"},"update_available":false}'

haffi78, 2022-08-21:
I have same errors all my action Enties for my hue Dimmers just "poofed" and they do not come back, I have tried deleteing the device, rebooting. Nothing seams to work.
The Z2M is poisting "action":"off-press" and such

EpicLPer, 2022-10-20:
Not sure why this has been closed? Same issue here, tho with new devices added.

mayorgergich, 2022-11-11:
This is still an issue as of today. None of the above fixes seem to work for me.

EpicLPer, 2022-11-11:
> This is still an issue as of today. None of the above fixes seem to work for me.

For me being in the Home Assistant Automation interface and then pressing all possible button combinations seems to solve it, at least after a refresh of the page.

svrooij, 2023-04-12:
I'm also having this issue.
```plain
Home Assistant 2023.4.2
Frontend 20230309.1 - latest
Zigee2mqtt 1.30.3
```
Device: https://www.zigbee2mqtt.io/devices/E2123.html#ikea-e2123

Can I provide you with some info to debug @Koenkk ?

LemonMeringuin, 2024-01-18:
I know that this is an old thread, but I am now experiencing the same issue. I solved it for about two days by removing devices and reconnecting them, but it is impacting all of my inovelli blue switches and the aqara buttons I was testing. All press and hold actions are replaced with unknown action. This results in me having to factory reset the inovelli switches each time.

Core 2024.1.3
Supervisor 2023.12.1
Operating System 11.4
Frontend 20240104.0

Z2M 1.35.1
Coordinator EZSP v12
Coordinator Rev 7.3.1.0 build 176

LemonMeringuin, 2024-01-22:
It's annoying that I can't define triggers by device, but typing mqtt topics, I am able to get the same result.

Example code for one of my Inovelli Blues is below.

```
alias: 1st Bath - Set Timer Manual On
description: ""
trigger:
  - platform: mqtt
    topic: zigbee2mqtt/1st Bath - Fan/action
    payload: up_single
condition: []
action:
  - service: timer.start
    data: {}
    target:
      entity_id: timer.1st_bath_timer_manual_on
mode: single
```

ravindrajob, 2024-04-17:
Hi there 😊

I'm experiencing a similar issue as described in this thread. I'm using a SONOFF Zigbee dongle with Zigbee2MQTT and have integrated both Aqara buttons and Moes knobs. In my Home Assistant setup, I have noticed that triggers for these devices, especially the 'single press' action, are intermittently disappearing from the automation UI, leaving only the battery state visible. This happens quite frequently and seems to resolve itself after restarting Zigbee2MQTT and Home Assistant, but the issue reoccurs after some time.

My setup is running in Docker containers, and I keep all components updated. The intermittent nature of this problem makes it particularly challenging to troubleshoot. I wanted to share my experience to see if it might shed some light on this issue and perhaps contribute to finding a resolution.

Thanks for looking into this, and I appreciate any assistance or guidance you can provide.

ravindrajob, 2024-04-17:
> Hi there 😊
>
> I'm experiencing a similar issue as described in this thread. I'm using a SONOFF Zigbee dongle with Zigbee2MQTT and have integrated both Aqara buttons and Moes knobs. In my Home [...]

Thank @LemonMeringuin for the workaround shared! I've implemented it in my Home Assistant automations, and it works like a charm. Here’s how I integrated it into my setup:

I added this MQTT trigger directly to my automation script:

trigger:  
  - platform: mqtt  
    topic: "zigbee2mqtt/bouton_knob_desk/action"  
    payload: 'single'  
 
And here's the complete automation that is now successfully responding to the 'single press' event:

- id: smart_knob_single_press  
  alias: smart_knob_single_press  
  trigger:  
    - platform: device  
      domain: mqtt  
      device_id: MyID  
      type: action  
      subtype: single  
    - platform: mqtt  
      topic: "zigbee2mqtt/bouton_knob_desk/action"  
      payload: 'single'  
  action:  
    - service: light.toggle  
      entity_id: light.ledstrip_lidl_desk  
  mode: single  

[image]

David-65, 2024-05-20:
I just moved from ZHA to Z2M, and I'm experiencing the same thing.
Never had any issues using ZHA.
The buttons I'm using are (two) Hue remotes, and (six) Third Reality buttons.

The only why I can get them to work again, is to do a Force Remove. Then add each one back, making sure to use the same name and check the box update ID.
After that, all the automation will show the single, double, long press functions again, and will start working just fine.

I'm not sure why the single, double actions for the buttons all stop showing up at one time. I'm not sure if it's happening every time I pair a new device in Z2M?

Rachdingue, 2025-01-05:
> It's annoying that I can't define triggers by device, but typing mqtt topics, I am able to get the same result.
> 
> Example code for one of my Inovelli Blues is below.

Thanks bro. My HA scripts stopped and I realized my switches actions entities were gone. I use sample of your code and it works like a charm again.