Koenkk/zigbee2mqtt #7080: [Saswell SEA801-Zigbee/SEA802-Zigbee] Please add support for weekly schedule

soeren-a, 2021-04-14:
Please add support for weekly schedule settings for the already supported [Saswell SEA801-Zigbee/SEA802-Zigbee](https://www.zigbee2mqtt.io/devices/SEA801-Zigbee_SEA802-Zigbee.htm) radiator valve.
## What happened
With the current implementation there is no way to change or set the weekly schedule of the device. This is possible for other Tuya radiator valves but not for this one. There is no way to set the schedule on the valve.

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2538
Adapter firmware version:

soeren-a, 2021-05-21:
It would be great if you could add the ability to change / adjust the weekly schedule. Thanks.

JhnRabbit, 2021-06-09:
I have eceived this thing yesterday and have made the following for the schedule.

What you can do is the following:
Topic: zigbee2mqtt/<friendlyname>/set
Payload: see  attachment

[thermostaatjson.txt](https://github.com/Koenkk/zigbee2mqtt/files/6624094/thermostaatjson.txt)
dayofweek 1-7, 1=sunday
transitionTime: minutes after midnight

closed 2021-07-18 by github-actions[bot]

kr0ner, 2021-09-07:
> I have eceived this thing yesterday and have made the following for the schedule.
> 
> What you can do is the following:
> Topic: zigbee2mqtt/yourdevice/set
> Payload: see  attachment
> 
> 
> [thermostaatjson.txt](https://github.com/Koenkk/zigbee2mqtt/files/6624094/thermostaatjson.txt)
> dayofweek 1-7, 1=sunday
> transitionTime: minutes after midnight

Did this really work for you? I've tried it with the SEA802 and it does not seem to do anything :/

JohnRabbit1, 2021-09-08:
Yes sure it works. I have a couple of 801/802 thermostats here and programmed all of them that way.
If you are using nodered: I have attached a little flow wich does the trick for a complete week.
[flows.txt](https://github.com/Koenkk/zigbee2mqtt/files/7127294/flows.txt)
Rename flows.txt to flows.json (I cannot upload a json file)

kr0ner, 2021-09-08:
> Yes sure it works. I have a couple of 801/802 thermostats here and programmed all of them that way.
> If you are using nodered: I have attached a little flow wich does the trick for a complete week. [...]

That is good to know :) thanks for the hints 👍

geigervibe, 2021-11-09:
I tried to set more than 4 transitions per day but without luck.

I got the following error in the logs and I wonder, if this is really a limitation of device or if the z2m-integration is limited:

`Zigbee2MQTT:error 2021-11-09 09:12:06: Publish 'set' 'weekly_schedule' to 'zb_thermostat_t2' failed: 'Error: Too more transitions provided. Provided: 6 but supports only 6 for device zb_thermostat_t2'`

when I publishing the following:

```
{
  "weekly_schedule": {
    "2": {
      "dayofweek": 2,
      "mode": 1,
      "numoftrans": 6,
      "transitions": [
        {
          "heatSetpoint": "15.5",
          "transitionTime": 480
        },
        {
          "heatSetpoint": "9.0",
          "transitionTime": 540
        },
        {
          "heatSetpoint": "25.0",
          "transitionTime": 600
        },
        {
          "heatSetpoint": "20.5",
          "transitionTime": 660
        },
        {
          "heatSetpoint": "22.5",
          "transitionTime": 670
        },
        {
          "heatSetpoint": "7.5",
          "transitionTime": 720
        }
      ]
    }
  }
}
```

Has somebody an idea about this?

kr0ner, 2021-11-09:
> I tried to set more than 4 transitions per day but without luck.
>
> I got the following error in the logs and I wonder, if this is really a limitation of device or if the z2m-integration is limited: [...]

It always has 4 slots. If you provide e.g. 2, 3 and 4 will be set to 2;)

drzony, 2021-11-09:
@geigervibe The log message is incorrect, the device supports only 4 slots

geigervibe, 2021-11-09:
Thanks for clarification... While I was googling for it, I saw some examples of more than 4 slots for other thermostats. So I was unsafe about this model. Thanks...

iksrd, 2021-11-12:
I have a similar problem. I send the schedule, but nothing happens. Tell me what I'm doing wrong

{
  "weekly_schedule": {
    "1": {
      "dayofweek": 7,
      "mode": 1,
      "numoftrans": 4,
      "transitions": [
        {
          "heatSetpoint": "21",
          "transitionTime": 240
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 480
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 1080
        },
        {
          "heatSetpoint": "21",
          "transitionTime": 1320
        }
      ]
    },
    "2": {
      "dayofweek": 7,
      "mode": 1,
      "numoftrans": 4,
      "transitions": [
        {
          "heatSetpoint": "21",
          "transitionTime": 240
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 480
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 1080
        },
        {
          "heatSetpoint": "21",
          "transitionTime": 1320
        }
      ]
    },
    "3": {
      "dayofweek": 7,
      "mode": 1,
      "numoftrans": 4,
      "transitions": [
        {
          "heatSetpoint": "21",
          "transitionTime": 240
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 480
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 1080
        },
        {
          "heatSetpoint": "21",
          "transitionTime": 1320
        }
      ]
    },
    "4": {
      "dayofweek": 7,
      "mode": 1,
      "numoftrans": 4,
      "transitions": [
        {
          "heatSetpoint": "21",
          "transitionTime": 240
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 480
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 1080
        },
        {
          "heatSetpoint": "21",
          "transitionTime": 1320
        }
      ]
    },
    "5": {
      "dayofweek": 7,
      "mode": 1,
      "numoftrans": 4,
      "transitions": [
        {
          "heatSetpoint": "21",
          "transitionTime": 240
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 480
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 1080
        },
        {
          "heatSetpoint": "21",
          "transitionTime": 1320
        }
      ]
    },
    "6": {
      "dayofweek": 7,
      "mode": 1,
      "numoftrans": 4,
      "transitions": [
        {
          "heatSetpoint": "21",
          "transitionTime": 240
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 480
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 1080
        },
        {
          "heatSetpoint": "21",
          "transitionTime": 1320
        }
      ]
    },
    "7": {
      "dayofweek": 7,
      "mode": 1,
      "numoftrans": 4,
      "transitions": [
        {
          "heatSetpoint": "21",
          "transitionTime": 240
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 480
        },
        {
          "heatSetpoint": "22",
          "transitionTime": 1080
        },
        {
          "heatSetpoint": "21",
          "transitionTime": 1320
        }
      ]
  }
}
}

kr0ner, 2021-11-12:
> I have a similar problem. I send the schedule, but nothing happens. Tell me what I'm doing wrong
>
> {
>   "weekly_schedule": {
>     "1": {
>       "dayofweek": 7,
>       "mode": 1,
>       "numoftrans": [...]

 "dayofweek": 7
Copy & Paste error 😅

gomble, 2021-12-29:
hello did someone find a solution ? Will this be in the gui in the future ?

hannah52, 2022-01-14:
> hello did someone find a solution ? Will this be in the gui in the future ?

You could try [proposed solution](https://github.com/Koenkk/zigbee2mqtt/issues/7080#issuecomment-857700403).
In my case I can set only 3 days. Thus I put 3+3+1 days into `zigbee2mqtt/[device Friendly Name]` topic. 

I suggest you to try with 1 day of week and check if schedule works for you.

JohnRabbit1, 2022-01-14:
Hi,
Yesterday, I tried to change my weekly schedule in version 1.22.2. Yes you are right: something has changed. The days are shifted somehow. Previously day 1 was monday. Now day 1 is tuesday. And so on. day 7 is monday. I don't have problems to set 7 days in one time.

gomble, 2022-01-14:
> 

yes the same problem. Is it not possible to configure this on the gui ?

JohnRabbit1, 2022-01-14:
No, not at this moment. I program the device in node-red with z2m node. It is a simple JSON structure which is injected  in a z2m node.

KardinalReusen, 2022-10-04:
> I have eceived this thing yesterday and have made the following for the schedule.
>
> What you can do is the following: Topic: zigbee2mqtt/yourdevice/set Payload: see attachment [...]

Is it possible to deactivate / delete the weekly schedule of the device instead of changing it? And how can I do it?

JohnRabbit1, 2022-10-04:
You can set system_mode to off.

KardinalReusen, 2022-10-04:
I would prefer operating in system_mode auto (for being able to regulate the temperature) without having a built-in schedule because it is disturbing the settings I set in HA manually . Is that possible, too?

JohnRabbit1, 2022-10-04:
I am not familiar with HA. But what you can try is the following: there is a button on the device with tekst: auto/manu. Select manu: you will disable the built-in schedule. HA can change settings then. Never tried that. Reason fort hat is that I want all my devices operate independent from the automation system. In my case: when Openhab/Nodered fails (cann’t remember when that was the case)  the buit-in schedule continues. So i am using the automation system purely for programmig the devices and switching on/off

KardinalReusen, 2022-10-04:
Maybe I found a solution here: [Link](https://community.home-assistant.io/t/how-to-delete-factory-set-temp-schedule-from-t0l0-tuya-sea802-z01-radiator-valve/390286). I will try it as soon as I get home.

drzony, 2022-10-04:
@KardinalReusen In Home Assistant:
Auto - Based on temperature, but temperature is set based on schedule
Heat - Based on temperature, no schedule
Off - Idle

JohnRabbit1, 2022-10-04:
Good to hear. You can also try this:   system_mode: heat

KardinalReusen, 2022-10-04:
> @KardinalReusen In Home Assistant: Auto - Based on temperature, but temperature is set based on schedule Heat - Based on temperature, no schedule Off - Idle Below is SEA802 set to maintain 20*C at all times [image]

Now I finally understand how it works! Many thanks. I will set the mode to heat.