Koenkk/zigbee2mqtt #7083: Permit join for a single device not working after switched to new Electrolama zig-a-zig-ah! (zzh!)

costiboshca, 2021-04-15:
I recently update my coordinator to  Electrolama zig-a-zig-ah! (zzh!)  and I am not able to pair devices to a specific router device. 
This was possible with old CC2531 device. 
Steps: 
- pair more routerspairing
- disable permit join for all 
- permit join for a specific router
- NOT 
Using Zigbee2MQTT version 1.18.2 and coordinator firmware version with revision 20210120 and type zStack3x0

juslex, 2021-04-15:
> I recently update my coordinator to Electrolama zig-a-zig-ah! (zzh!) (CC2652R) and I am not able to pair devices to a specific router device.
> This was possible with old CC2531 device.
> Steps:
> 
> * pair more routers
> * disable permit join for all
> * permit join for a specific router
> * NOT working
>   Using Zigbee2MQTT version 1.18.2 and coordinator firmware version with revision 20210120 and type zStack3x0

What’s the router device?

golles, 2021-04-16:
I noticed the same with the Ikea E1603/E1702
https://www.zigbee2mqtt.io/devices/E1603_E1702.html

MattWestb, 2021-04-16:
The question is what is the device you is trying paring.
All IKEA devices is today Zigbee 3 and updating the link key (The old CWS and old motion sensor is still ZLL) and many older devices that is not following the Zigbee standard cant handle that.

I have 2 different version of Aqara weather sensors and one model cant joining to IKEA ZB3 routers but after being in the network they can moving to them and working very well.

And if the router cant handling real ZB3 joining IKEA remotes can joining thru them.

I dont knowing if zzh! is storing updated link keys but CC-2531 is not updating link keys and is running on the  default well known key and not have all ZB3 problems with comparability then its not working in real ZB3 mode (and having all security problems that is fixed in ZB3).

costiboshca, 2021-04-16:
Any router that I have is not working. 
Xiaomi ZNCZ04LM, Blitzwolf BW-SS7_1gang. 
On CC2531 it worked. 
When I reconfigured I also change the default network_key  and pan_id.

hellcry37, 2021-04-18:
this happens for me too

juslex, 2021-04-18:
> Any router that I have is not working.
> Xiaomi ZNCZ04LM, Blitzwolf BW-SS7_1gang.
> On CC2531 it worked.
> When I reconfigured I also change the default network_key and pan_id.

i had a similar problem but involving zemismart switch.

https://github.com/Koenkk/zigbee2mqtt/issues/5386#event-4384267455

costiboshca, 2021-04-19:
It's pairing to routers but not when I want to pair only one router that I specify.

juslex, 2021-04-19:
> It's pairing to routers but not when I want to pair only one router that I specify.

Same situation in my mentioned issue.

hellcry37, 2021-05-20:
up

MattWestb, 2021-05-21:
If i remember right is some TI coordinator firmware not opening the trust center for joining then only opening on router for joining. ZHA have patching so also the coordinator is open for joining  then open one router so the trust center is can joining the device from the router.

Its on bug in  the TI Zigbee stack.

closed 2021-06-29 by github-actions[bot]