Koenkk/zigbee2mqtt #7087: Z2M does not start after NPM upgrade to version 7.9.0

Josov, 2021-04-15:
Platform Raspberry Pi 4
npm version 7.9.0 upgraded from version 6.14.12 
adapter: CC2530 (but the same issue with CC2652R1)

## Debug info
pi@Domotica:/opt/zigbee2mqtt $ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.16.2-dev start
> node index.js

Zigbee2MQTT:debug 15-04-2021 11:47:14: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  15-04-2021 11:47:14: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-04-15.11-47-13' filename: log.txt
Zigbee2MQTT:debug 15-04-2021 11:47:14: Removing old log directory '/opt/zigbee2mqtt/data/log/2021-03-08.13-42-26'
(node:1827) UnhandledPromiseRejectionWarning: Error: EACCES: permission denied, unlink '/opt/zigbee2mqtt/data/log/2021-03-08.13-42-26/log.txt'
    at Object.unlinkSync (fs.js:1136:3)
    at rimrafSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:303:17)
    at /opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:336:39
    at Array.forEach (<anonymous>)
    at rmkidsSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:336:26)
    at rmdirSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:329:7)
    at Function.rimrafSync [as sync] (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:301:9)
    at /opt/zigbee2mqtt/lib/util/logger.js:113:20
    at Array.forEach (<anonymous>)
    at DerivedLogger.cleanup (/opt/zigbee2mqtt/lib/util/logger.js:111:21)
(node:1827) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:1827) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Zigbee2MQTT:debug 15-04-2021 11:52:14: Saving state to file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:error 15-04-2021 11:52:14: Failed to write state to '/opt/zigbee2mqtt/data/state.json' (EACCES: permission denied, open '/opt/zigbee2mqtt/data/state.json')
**<^C entered here, this followed>**
Zigbee2MQTT:error 15-04-2021 11:58:32: Failed to call 'OnEvent' 'stop' (TypeError: Cannot read property 'getDevices' of undefined
    at Zigbee.getClients (/opt/zigbee2mqtt/lib/zigbee.js:138:30)
    at OnEvent.stop (/opt/zigbee2mqtt/lib/extension/onEvent.js:23:42)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:381:44)
    at async Controller.stop (/opt/zigbee2mqtt/lib/controller.js:189:9))
Zigbee2MQTT:debug 15-04-2021 11:58:32: Saving state to file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:error 15-04-2021 11:58:32: Failed to write state to '/opt/zigbee2mqtt/data/state.json' (EACCES: permission denied, open '/opt/zigbee2mqtt/data/state.json')
Zigbee2MQTT:error 15-04-2021 11:58:32: Not connected to MQTT server!
Zigbee2MQTT:error 15-04-2021 11:58:32: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
Zigbee2MQTT:info  15-04-2021 11:58:32: Disconnecting from MQTT server
(node:1827) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'end' of undefined
    at MQTT.disconnect (/opt/zigbee2mqtt/lib/mqtt.js:93:21)
    at async Controller.stop (/opt/zigbee2mqtt/lib/controller.js:193:9)
(node:1827) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 3)

The system works fine again after returning to NPM version 6.14.12 !!!

Koenkk, 2021-04-15:
`EACCES: permission denied, unlink ` looks like a permission issue, does `sudo chmod 777 -R /opt/zigbee2mqtt` fix it?

Josov, 2021-04-16:
I tested the change on one Pi, here is the result. 
This time I get further and it looks like the trouble moved to the serial port:

pi@Domotica:/opt/zigbee2mqtt $ sudo DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.16.2-dev start
> node index.js

Zigbee2MQTT:debug 16-04-2021 08:40:12: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  16-04-2021 08:40:12: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-04-16.08-40-11' filename: log.txt
Zigbee2MQTT:debug 16-04-2021 08:40:12: Removing old log directory '/opt/zigbee2mqtt/data/log/2021-04-16.08-30-24'
Zigbee2MQTT:info  16-04-2021 08:40:12: Starting Zigbee2MQTT version 1.16.2-dev (commit #47d26c8)
Zigbee2MQTT:info  16-04-2021 08:40:12: Starting zigbee-herdsman (0.13.39)
Zigbee2MQTT:debug 16-04-2021 08:40:12: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6759},"serialPort":{"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[13,14,5,8,12,9,7,5,10,6,11,4,1,2,3,0],"panID":6759,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
Zigbee2MQTT:error 16-04-2021 08:40:12: Error while starting zigbee-herdsman
Zigbee2MQTT:error 16-04-2021 08:40:12: Failed to start zigbee
Zigbee2MQTT:error 16-04-2021 08:40:12: Exiting...
Zigbee2MQTT:error 16-04-2021 08:40:12: **Error: Error while opening serialport 'Error: Error: Permission denied, cannot open /dev/ttyACM0'**
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
pi@Domotica:/opt/zigbee2mqtt $

Koenkk, 2021-04-17:
Check https://www.zigbee2mqtt.io/information/FAQ.html#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port

Josov, 2021-04-17:
Solved, I had to change zigbee2mqtt.service. I had User=root and changed it to User=pi. Back to how it used to be before.

closed 2021-04-17 by Josov