Koenkk/zigbee2mqtt #7088: LED Driver TuYa TS0503B  RGB UNSUPPORTED_ATTRIBUTE when try restore color

shirou93, 2021-04-15:
## What happened
Zigbee2MQTT:error 2021-04-15 12:55:56: Publish 'get' 'color' to '0x60a423fffe69dc24' failed: 'Error: Read 0x60a423fffe69dc24/1 lightingColorCtrl(["currentX","currentY","enhancedCurrentHue","currentSaturation"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
## What did you expect to happen
Restore color on z2m UI
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.18.2-1
Adapter hardware: CC2652RB
Adapter firmware version: CC2652RB_coordinator_20210120.zip

https://www.zigbee2mqtt.io/devices/TS0503B.html#tuya-ts0503b

shirou93, 2021-04-15:
@sjorge this is same issue as: https://github.com/Koenkk/zigbee2mqtt/issues/6921?

sjorge, 2021-04-15:
> @sjorge this is same issue as: #6921?

Likely, can you try the same read as in #6921 ?

shirou93, 2021-04-15:
Same log when i try restore in ui

[image]

sjorge, 2021-04-15:
I mean the manual read of only the one attribute as mentioned here https://github.com/Koenkk/zigbee2mqtt/issues/6921#issuecomment-812929925

shirou93, 2021-04-15:
[image]

sjorge, 2021-04-15:
PR opened to disable enhancedHue for this device.

sjorge, 2021-04-15:
Merged

shirou93, 2021-04-29:
@sjorge  TuYa TS0505B https://www.zigbee2mqtt.io/devices/TS0505B.html#tuya-ts0505b same issue :)

closed 2021-05-01 by shirou93