Koenkk/zigbee2mqtt #7093: Wrong identification of device

ambitzgi, 2021-04-15:
## What happened
My new device Moes ZK-CH-2U is identified as "Nue / 3A Smart Zigbee 3.0 light controller (HGZB-01)"
It should be treated the same way as ZK-EU-2U. I think the only difference is the physical socket / plug layout. 
swiss socket  vs german socket

## What did you expect to happen
That the device get recognised as ZK-CH-2U or ZK-EU-2U
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: right now I don't know exact version, installed 17.2.2021, most probably 1.17 OR 1.17.1
Adapter hardware: CC2531, 
Adapter firmware version: CC2531_DEFAULT_20201127

from the database file:
{"id":13,"type":"Router","ieeeAddr":"0x086bd7fffe5b73ba","nwkAddr":30876,"manufId":4478,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"LXN59-1S7LX1.0","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"LXN59-1S7LX1.0","manufacturerName":"3A Smart Home DE","powerSource":0,"zclVersion":3,"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20200519"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e3303d","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20200519","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1618519657855}

ambitzgi, 2021-04-15:
It is a duplicate of #6354

Maybe one could add a comment on the supported devices page, that (newer) version of the ZK-??-2U are from a technical point of view a "Nue / 3A HGZB-01" and should be treated as such.

Koenkk, 2021-04-17:
There is no way we can distinguish this device from a HGZB-01, as you can see in your database the device even advertises itself as being from 3A: `"manufName":"3A Smart Home DE"`. I've added this device as a whitelabel now so it will be listed as such on https://www.zigbee2mqtt.io/information/supported_devices.html after the next z2m release.

closed 2021-04-17 by Koenkk