Koenkk/zigbee2mqtt #7096: No converter available for 'local_temperature' () TS0601

copystring, 2021-04-16:
Hi,

sending {"local_temperature": ""} to my TS0601_thermostat results in: No converter available for 'local_temperature' ()

Help? :)

The thermostat does send the temperature on it's own but it is not possible to directly get the current value via /get
Looking at https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html this should work fine.

My exact model: _TZE200_ckud7u2l

Zigbee2MQTT version: 1.18.2-dev commit: 93c0a8a
Adapter hardware: zzh
Adapter firmware version: 20210120

[image]

copystring, 2021-05-17:
Still an issue even with latest version.

Koenkk, 2021-05-17:
`/get` on `local_temperature` is not supported for TuYa devices, updated the docs.

closed 2021-05-17 by Koenkk