Koenkk/zigbee2mqtt #7097: Zemismart tuya ZigBee Smart Curtain Switch

Burki24, 2021-04-17:
sorry, but I'm not familiar with coding.
So I tried to use your instructions to add a new device, but I was miserably failed. :-(
I tried to search details about the device on Zemismart-Webpage, didn't find anything. 

## Information about the device + link

Name:              Zemismart tuya Smart curtain switch
Vendor:            Zemismart
ID:                    ZW-EC-01
Device-Type:    Router
ZigBee-Model: LXX60-CS27LX1.0
Dump:
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100,"tuyaMovingState":0}' from endpoint 1 with groupID 0
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0,"tuyaMovingState":1}' from endpoint 1 with groupID 0
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100,"tuyaMovingState":0}' from endpoint 1 with groupID 0
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0,"tuyaMovingState":1}' from endpoint 1 with groupID 0
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0,"tuyaMovingState":1}' from endpoint 1 with groupID 0
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0,"tuyaMovingState":1}' from endpoint 1 with groupID 0
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
> Received Zigbee message from '0xbc33acfffe56fc1a', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100,"tuyaMovingState":0}' from endpoint 1 with groupID 0

Hopefully, there is anyone, who can help me, thanks in advance

Burkhard

Burki24, 2021-04-19:
The following Informations I got from the Seller:

`/**
 *
 *	Copyright 2019 SmartThings
 *
 *	Licensed under the Apache License, Version 2.0 (the "License"); you may not use this file except
 *	in compliance with the License. You may obtain a copy of the License at:
 *
 *		http://www.apache.org/licenses/LICENSE-2.0
 *
 *	Unless required by applicable law or agreed to in writing, software distributed under the License is distributed
 *	on an "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied. See the License
 *	for the specific language governing permissions and limitations under the License.
 */
import physicalgraph.zigbee.zcl.DataType

metadata {
    definition(name: "Nue ZigBee Blind Switch", namespace: "3A", author: "Kevin", ocfDeviceType: "oic.d.blind", mnmn: "SmartThings", vid: "generic-shade") {
        capability "Actuator"
        capability "Configuration"
        capability "Refresh"
        capability "Window Shade"
        capability "Health Check"
        capability "Switch Level"

        command "pause"

        fingerprint profileId: "0104", inClusters: "0000, 0003, 0004, 0102", outClusters: "0019", model: "E2B0-KR000Z0-HA", deviceJoinName: "SOMFY Blind Controller/eZEX" // SY-IoT201-BD
        fingerprint profileId: "0104", inClusters: "0000, 0202, 0003, 0004, 0005, 0008, 0102", outClusters: "000A, 0019", manufacturer: "Feibit Co.Ltd", model: "FB56+CUR17SB2.2", deviceJoinName: "Nue Smart Blind Switch"
        fingerprint profileId: "0104", inClusters: "0000, 0202, 0003, 0004, 0005, 0008, 0102", outClusters: "000A, 0019", manufacturer: "Feibit Co.Ltd", model: "FB56+CUR17SB1.9", deviceJoinName: "Nue Smart Blinds"
    }

    tiles(scale: 2) {
        multiAttributeTile(name:"windowShade", type: "generic", width: 6, height: 4) {
   
            tileAttribute("device.windowShade", key: "PRIMARY_CONTROL") {
                attributeState "open", label: 'Close', action: "close", icon: "http://www.ezex.co.kr/img/st/window_open.png", backgroundColor: "#00A0DC", nextState: "closing"
                attributeState "closed", label: 'Open', action: "open", icon: "http://www.ezex.co.kr/img/st/window_close.png", backgroundColor: "#ffffff", nextState: "opening"
               // attributeState "partially open", label: 'Close', action: "close", icon: "http://www.ezex.co.kr/img/st/window_open.png", backgroundColor: "#d45614", nextState: "closing"
               // attributeState "opening", label: 'Open', action: "open", icon: "http://www.ezex.co.kr/img/st/window_open.png", backgroundColor: "#00A0DC", nextState: "partially open"
               // attributeState "closing", label: 'Open', action: "open", icon: "http://www.ezex.co.kr/img/st/window_close.png", backgroundColor: "#ffffff", nextState: "partially open"
                attributeState "partially close", label: 'Close', action: "close", icon: "http://www.ezex.co.kr/img/st/window_open.png", backgroundColor: "#d45614", nextState: "closed"
                attributeState "partially open", label: 'Open', action: "open", icon: "http://www.ezex.co.kr/img/st/window_open.png", backgroundColor: "#d45614", nextState: "open"
                attributeState "opening", label: 'Opening', action: "pause", icon: "http://www.ezex.co.kr/img/st/window_open.png", backgroundColor: "#00A0DC", nextState: "partially open"
                attributeState "closing", label: 'Closing', action: "pause", icon: "http://www.ezex.co.kr/img/st/window_close.png", backgroundColor: "#ffffff", nextState: "partially close"
            }
        }  

        standardTile("shadeOpen", "device.windowShade", width:2, height:2) {
            state "default", label:'open', icon:"st.doors.garage.garage-open", backgroundColor:"#FFFFFF", action:"open", nextState: "opening", defaultState:true
            state "open", label:'open', icon:"st.doors.garage.garage-open", backgroundColor:"#E86D13", action:"open"
            state "opening", label:'opening', icon:"st.doors.garage.garage-opening", backgroundColor:"#E86D13", action:"pause"
        }
        standardTile("shadeClose", "device.windowShade", width:2, height:2) {
            state "default", label:'close', icon:"st.doors.garage.garage-closed", backgroundColor:"#FFFFFF", action:"close", nextState: "closing", defaultState:true
            state "close", label:'close', icon:"st.doors.garage.garage-closed", backgroundColor:"#00A0DC", action:"close"
            state "closing", label:'closing', icon:"st.doors.garage.garage-closing", backgroundColor:"#00A0DC", action:"pause"
        }      
           
		standardTile("shadeStop", "device.refresh", width:2, height:2) {
            state "default", label:'stop', icon:"st.Electronics.electronics13", backgroundColor:"#FFFFFF", action:"pause"
        }
        standardTile("contPause", "device.switch", inactiveLabel: false, decoration: "flat", width: 2, height: 2) {
            state "Pause", label:"", icon:'st.sonos.pause-btn', action:'pause', backgroundColor:"#cccccc"
        }
        standardTile("refresh", "device.refresh", inactiveLabel: false, decoration: "flat", width: 2, height: 1) {
            state "default", label:"", action:"refresh.refresh", icon:"st.secondary.refresh"
        }
        valueTile("shadeLevel", "device.level", width: 4, height: 1) {
[... 82 lines cut ...]
 * PING is used by Device-Watch in attempt to reach the Device
 * */
def ping() {
    return refresh()
}

def refresh() {
    log.info "refresh()"
[... cut ...]
    // Device-Watch allows 2 check-in misses from device + ping (plus 2 min lag time)
    log.info "configure()"
    sendEvent(name: "checkInterval", value: 2 * 60 * 60 + 2 * 60, displayed: false, data: [protocol: "zigbee", hubHardwareId: device.hub.hardwareID])
    log.debug "Configuring Reporting and Bindings."
    zigbee.configureReporting(CLUSTER_WINDOW_COVERING, ATTRIBUTE_POSITION_LIFT, DataType.UINT8, 0, 600, null)
}
`
Is that helpfull to implement this device?

Many thanks in advance

Burkhard

closed 2021-05-27 by github-actions[bot]

jimmylefarfadet, 2022-05-18:
Bonjour,

Je relance le sujet car je suis dans le même cas et je n'arrive pas à paramétrer mon interrupteur pour volet.

Si quelqu'un peut m'aider.

Merci