Koenkk/zigbee2mqtt #7104: Docker: Non-root Zigbee2MQTT Container fails to access USB

Moltey, 2021-04-17:
Hi @all,

thank you for providing your zigbee2mqtt project with the community.

During my attempts to improve the security layers within my docker environment, I tried to run zigbee2mqtt as non-root user - however, I failed. My final goal would be, to run the docker daemon as non-root user, whereas also the containers would also be executed as non-root users.

Therefore, I would like to ask, if anyone was successful accessing USB devices in docker containers executed by non-root users and could provide any advice, how to proceed.

Thank you for for advice in advance!

## What happened

If I start the docker container as non-root user, access to the USB device (Conbee II) fails, although (imho) the user has appropriate access privileges. Zigbee-herdsman (0.13.88) throws: 

`Error: Error while opening serialport 'Error: Error: Permission denied, cannot open /dev/ttyACM0'`

## What did you expect to happen

If I start the docker container as root user (leaving out "--user 1001:1001"), the USB device (Conbee II) is passed through as expected.

## How to reproduce it (minimal and precise)

Start parameters of the zigbee2mqtt docker container:
`# sudo docker run \ `
`   -it \`
`   --name=zigbee2mqtt \`
`   -v /home/tstusr/zigbee2mqtt/data:/app/data \`
`   -e TZ=Europe/Berlin \`
`   -v /run/udev:/run/udev:ro \`
`   --device=/dev/ttyACM0 \`
`   --privileged \`
`   --user 1001:1001 \`
`   koenkk/zigbee2mqtt`

User ID and Group ID is 1001:
`User: tstusr ID: 1001`
`Group: tstusr ID: 1001`

The user should have access to the USB device:
`# sudo usermod -a -G dialout tstusr`

`# sudo -u tstusr test -w /dev/ttyACM0 && echo success || echo failure`
`success`

The data directory is in the home folder of tstusr:
`# mkdir -o tstusr -g tstusr /home/tstusr/zigbee2mqtt`

Double check: All files are owned by the User (after first run):
`# chown -R tstusr:tstusr /home/tstusr/zigbee2mqtt`

## Debug info
Docker version: 20.10.5, build 55c4c88
Zigbee2MQTT version: 1.18.2
Adapter hardware: Conbee II @ /dev/
Adapter firmware version: ?

Moltey, 2021-04-17:
Update:
1. The device seems to be passed through into the container with similar device ownership (root:dialout) compared to the host:
 `# docker exec -it zigbee2mqtt ls -l /dev/ttyACM*`
`crw-rw----    1 root     dialout   166,   0 Apr 17 16:48 /dev/ttyACM0`
2. However, the access test is not successful in the container:
`# docker exec -it zigbee2mqtt test -w /dev/ttyACM0 && echo success || echo failure`
`failure`

Any suggestion is welcome ... thx!

Moltey, 2021-04-18:
I found the solution myself and proposed an update to the README.md (see above).

closed 2021-04-18 by Moltey

segaura, 2023-07-29:
> I found the solution myself and proposed an update to the README.md (see above).

Cool, thank you.
In docker-compose.yml the same can be obtained with:

```
  image: ...
  user: ...
  group_add:
  - dialout
  volumes: ...
```