Koenkk/zigbee2mqtt #7106: After z2m restart all lights turn on on full brightness

Mr-Groch, 2021-04-17:
## What happened
When I restart zigbee2mqtt and turn on any light it will turn on on 100% brightness instead of last stored brightness in state.json

## What did you expect to happen
Lights should always turn on with stored last brightness if I'm not telling it directly to set brightness value

## How to reproduce it (minimal and precise)
1. Turn off any light at any brightness value below 100%
2. Restart zigbee2mqtt
3. Turn on light without brightness value (just state: ON)
4. Observe that light turn's on with 100% brightness

## Debug info
Zigbee2MQTT version: 1.18.2
Adapter hardware: CC26X2R1 zzh!
Adapter firmware version: 20210120

Koenkk, 2021-04-18:
Can you provide the herdsman debug logging of this?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Mr-Groch, 2021-04-18:
Here you go, herdsman debug log before restart:

http://groch.cloud/misc/herdsman-before_restart.log

And after restart:

http://groch.cloud/misc/herdsman-after_restart.log

1. Before restart, I've turned light 7cb03eaa00a9f923 on with brigtness around 25%
2. Next i've turned that light off
3. Restarted zigbee2mqtt addon container
4. After restart I've turned 7cb03eaa00a9f923 on without brightness value
5. Light turns on on 100% brightness

Koenkk, 2021-04-19:
I see the issue indeed, I also need to send MQTT commands, can you provide the debug log mixed with the herdsman one?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

Mr-Groch, 2021-04-19:
Debug log with herdsman debug, befora and after restart. Same steps, same light - 7cb03eaa00a9f923 (friendly name: Taśma LED RGBW - zewnętrzna 1)

http://groch.cloud/misc/z2m_herdsman_debug_log_20210419.7z

Koenkk, 2021-04-20:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

titus-kruse, 2021-04-22:
I've checked out latest zigbee2mqtt docker image today and made changes of commit 9b9edce3c7c7ce636bf79c21baee51f2ad98f75c manually. After restarting container, brightness of my lights is on same state as before. Even after restarting docker at all. Thank you so much.

Koenkk, 2021-04-22:
Cool 😄

closed 2021-04-22 by Koenkk

Mr-Groch, 2021-04-22:
Yes, I can confirm its fixed, thanks!