Koenkk/zigbee2mqtt #7112: Xiaomi LLKZMK11LM not updating certain exposes after firmware update

snippem, 2021-04-18:
## What happened

Updated the switch when new firmware was reported by zigbee2mqtt
Going from version 9-20-2018 to 8-6-2020.

## What did you expect to happen

To work better with the new firmware :-)
But now the power meter works but the energy meter , temperature meter and voltage readout doesn't work anymore.
Also noted when the update completed some of the switches sometimes turn on and off without telling them to do so.
I overcome this problem by resetting the switch.
This doesn't had any effect on the missing exposes.

## Debug info
Zigbee2MQTT version: latest developer version.
Adapter hardware: Slae.sh cc2652rb
Adapter firmware version: Couldn't find it

McGiverGim, 2021-04-26:
@snippem maybe can you put your logs in debug mode and paste it here? Maybe it shows something about this problem.

snippem, 2021-04-27:
> @snippem maybe can you put your logs in debug mode and paste it here? Maybe it shows something about this problem.

Okay here is the debug log:

Debug Received MQTT message on 'zigbee2mqtt/Daan plafondlamp/l1/set' with data 'ON'
Debug Publishing 'set' 'state' to 'Daan plafondlamp'
Info MQTT publish: topic 'zigbee2mqtt/Daan plafondlamp', payload '{"energy":null,"interlock":true,"linkquality":45,"power":0.06,"state_l1":"ON","state_l2":"OFF","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null}'
Debug Received Zigbee message from 'Daan plafondlamp', type 'attributeReport', cluster 'genOnOff', data '{"61440":248397568,"onOff":0}' from endpoint 2 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Daan plafondlamp', payload '{"energy":null,"interlock":true,"linkquality":45,"power":0.06,"state_l1":"ON","state_l2":"OFF","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null}'
Debug Received Zigbee message from 'Daan plafondlamp', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440745,"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Daan plafondlamp', payload '{"energy":null,"interlock":true,"linkquality":45,"power":0.06,"state_l1":"ON","state_l2":"OFF","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null}'
Debug Received Zigbee message from 'Daan plafondlamp', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":2.3317694664001465}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Daan plafondlamp', payload '{"energy":null,"interlock":true,"linkquality":45,"power":2.33,"state_l1":"ON","state_l2":"OFF","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null}'
Debug Received Zigbee message from 'Daan plafondlamp', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":10.66430950164795}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Daan plafondlamp', payload '{"energy":null,"interlock":true,"linkquality":57,"power":10.66,"state_l1":"ON","state_l2":"OFF","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null}'

McGiverGim, 2021-04-28:
It seems it receives only the on/off switch information and one analog that z2m maps to power. Nothing more appears. At least in this log.

I don't know if when pairing it can give more information. If you have time do the pairing or the device again with debug enabled.

Which clusters show the UI of Z2M for you? This are the clusters of the original device, without OTA:
[image]

snippem, 2021-04-28:
Hopefully will find some time to add the device and log these are the cluster with the update[image]

leroyloren, 2021-05-04:
Hi, the same issue since OTA 8-6-2020 does not show energy related values.

McGiverGim, 2021-05-04:
@leroyloren are you able to take some debug logs with the device repairing/joining the network? To see if something appears there...

Here we have two options:
- The firmware file is not correct for the device
- With the update the information of the device is sent in a different manner. I think this information is obtained only when joining the network.

McGiverGim, 2021-05-04:
I've seen that the original post that started the Xiaomi updates https://github.com/Koenkk/zigbee2mqtt/issues/6906, there are four older firmware versions: 
``` 
+---lumi.relay.c2acn01
|   +---0.0.0_0036_8691F18B_e92ad148ac2de53568745a44a594007f
|   |       OTA_lumi_relay_c2acn01_V1.3.36.20190308_94007F.20191009160743.ota
|   |
|   +---0.0.0_0042_6CD27E92_33ff98239d745a64cbe3e0408768389c
|   |       OTA_lumi_relay_c2acn01_V1.3.42.20191230_68389C.20200429113233.ota
|   |
|   +---30_0000001E_c0e78b9c_86ffab33d58a29a8cee396c363870c94
|   |       OTA_withCRC_LMES_DualController_V1.3.30_20170801.20181024095320.ota
|   |
|   \---35_3b451bb3_2b39856873feffe89987e517ce06c78a
|           OTA_lumi_relay_c2acn01_V1.3.35.20180920.20181008170031.ota`
```
I don't know if they can be installed "manually" to see if some of them fixes the problem. But I will do that with caution, maybe the result is worse...

leroyloren, 2021-05-04:
```
Debug Received Zigbee message from 'Zebrik', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Zebrik","transaction":"433su-1"}'
Info Checking if update available for 'Zebrik'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Zebrik'","meta":{"device":"Zebrik","status":"checking_if_available"},"type":"ota_update"}'
Debug Check if update available for '0x00158d00062694c3' (lumi.relay.c2acn01)
Debug Using endpoint '1'
Debug Received Zigbee message from 'Zebrik', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":46,"imageType":257,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Debug Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":257,"fileVersion":46}'
Debug Is new image available for '0x00158d00062694c3', current '{"fieldControl":0,"manufacturerCode":4447,"imageType":257,"fileVersion":46}', latest meta '{"fileVersion":46,"fileSize":179502,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20201218113852_lumi.relay.c2acn01_0.0.0_0046_20201216_6BB0FD.ota","sha512":"ea9f9d74c65b2828e6417b525502bfc9d2f75c220b2504927e2b515c340503bc03e2612ad4c2b2ee8d1ff75d1328db8efa743c07c21a82f4ebd15f8ddac40a0b"}'
Debug Update available for '0x00158d00062694c3': NO
Info No update available for 'Zebrik'
```

McGiverGim, 2021-05-04:
This does not seem a "join" or "repair" to the network. Maybe it is an OTA check?
What I ask for is enable join the zigbee network, and press the relay button five seconds to join the network. Like the first time it join the network. Is then when the device publishes all this capabilities if I'm not wrong. I don't know if you have the device "more or less" accesible.

leroyloren, 2021-05-04:
```
Device 'Zebrik' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Zebrik","ieee_address":"0x00158d00062694c3"},"type":"device_joined"}'
Info Starting interview of 'Zebrik'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Zebrik","ieee_address":"0x00158d00062694c3","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"Zebrik"},"type":"device_connected"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"Zebrik"},"type":"pairing"}'
Debug Device 'Zebrik' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Zebrik","ieee_address":"0x00158d00062694c3"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Zebrik"},"type":"device_announced"}'
Debug Device 'Roleta pokoj 2' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Roleta pokoj 2","ieee_address":"0x5c0272fffec20a57"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Roleta pokoj 2"},"type":"device_announced"}'
Debug Received Zigbee message from 'Zebrik', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.relay.c2acn01"}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Zebrik', type 'raw', cluster 'genBasic', data '{"data":[28,95,17,0,10,1,255,66,68,3,40,0,5,33,25,0,8,33,46,18,9,33,0,1,100,16,0,101,16,0,110,32,0,111,32,0,148,32,25,149,57,10,215,3,61,150,57,123,248,21,69,151,57,136,229,200,59,152,57,240,108,192,62,155,33,0,0,156,32,1,10,33,0,0,12,40,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Zebrik', type 'attributeReport', cluster 'genBasic', data '{"appVersion":1,"modelId":"lumi.relay.c2acn01"}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Zebrik', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Zebrik', type 'readResponse', cluster 'genBasic', data '{"powerSource":4}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Zebrik', type 'readResponse', cluster 'genBasic', data '{"zclVersion":0}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Zebrik', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Zebrik', type 'readResponse', cluster 'genBasic', data '{"hwVersion":18}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Zebrik', type 'readResponse', cluster 'genBasic', data '{"dateCode":"8-6-2020"}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Zebrik', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Info Successfully interviewed 'Zebrik', device has successfully been paired
Info Device 'Zebrik' is supported, identified as: Xiaomi Aqara wireless relay controller (LLKZMK11LM)
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara wireless relay controller","exposes":[{"access":5,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"endpoint":"l1","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l1","name":"state","property":"state_l1","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l2","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l2","name":"state","property":"state_l2","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":3,"description":"Enabling prevents both relais being on at the same time","name":"interlock","property":"interlock","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"LLKZMK11LM","supports_ota":true,"vendor":"Xiaomi"},"friendly_name":"Zebrik","ieee_address":"0x00158d00062694c3","status":"successful","supported":true},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Aqara wireless relay controller","friendly_name":"Zebrik","model":"LLKZMK11LM","supported":true,"vendor":"Xiaomi"},"type":"pairing"}'
```

McGiverGim, 2021-05-04:
I thought that the debug will show other clusters, but only genBasic is shown... I think I can't help much more.
@Koenkk some idea about this? If the problem is with the firmware file, a lot of users will update using the OTA function to this version and lost some features... is a very popular device.

Koenkk, 2021-05-04:
Have you already tried reconfiguring the device by clicking on the device in the frontend and pressing the yellow refresh icon?

snippem, 2021-05-04:
> Have you already tried reconfiguring the device by clicking on the device in the frontend and pressing the yellow refresh icon?

Hi tried that and is say's the device cannot be reconfigured.

Koenkk, 2021-05-04:
Can someone check if energy metering still works with the original gateway when on this firmware?

leroyloren, 2021-05-04:
[image]
[image]

If you really stop showing this information in the latest firmware, is there any way to downgrade?

McGiverGim, 2021-05-05:
The screen captures are from your relay (the one with the problem) with the official hub? The accessory ID seems the MAC address and is different from the one of your logs.

leroyloren, 2021-05-05:
yeah, it's Aqara hub from a friend.

pilmrc, 2021-05-11:
Doing more investigation with log debug mode on.
In reality the device is sending data.
every minute (quite regular) is sending the following:

> debug 2021-05-11 19:04:40: Received Zigbee message from 'switch1', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0

And  every 2-5 minutes, there is no fixed time data like the following

> debug 2021-05-11 19:03:29: Received Zigbee message from 'switch1', type 'raw', cluster 'genBasic', data '{"data":[28,95,17,1,10,1,255,66,68,3,40,32,5,33,42,0,8,33,46,18,9,33,0,2,100,16,1,101,16,1,110,32,1,111,32,1,148,32,2,149,57,10,215,35,58,150,57,126,200,22,69,151,57,240,41,164,59,152,57,0,65,205,60,155,33,0,0,156,32,1,10,33,0,0,12,40,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0

> debug 2021-05-11 19:08:54: Received Zigbee message from 'switch1', type 'raw', cluster 'genBasic', data '{"data":[28,95,17,8,10,1,255,66,68,3,40,32,5,33,42,0,8,33,46,18,9,33,0,2,100,16,1,101,16,1,110,32,1,111,32,1,148,32,6,149,57,10,215,35,58,150,57,113,38,23,69,151,57,240,41,164,59,152,57,96,11,163,61,155,33,0,0,156,32,1,10,33,0,0,12,40,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0

At this point I suppose the upgrade is changing the way the data is formatted. Not expert on this.
Can I help in any other way? Pairing the device in debug mode can give more details?

Koenkk, 2021-05-12:
@McGiverGim good point, disabled the OTA

Looking at the logs of @pilmrc it seems Xiaomi changed to yet another proprietary format, currently have no clue how to decode this.

snippem, 2021-05-21:
> @McGiverGim good point, disabled the OTA
> 
> Looking at the logs of @pilmrc it seems Xiaomi changed to yet another proprietary format, currently have no clue how to decode this.

Is it a option to set the previous working firmware as an update.
This so my switches have all the functions that i had with the older firmware?
And also because my switches with the new firmware sometimes turn on and off all happening within a second.

Koenkk, 2021-05-21:
@snippem we don't have to OTA file of the previous firmware, otherwise it should be possible.

snippem, 2021-05-21:
Oke is there any chance the new firmware will have the same functionality as the old firmware?

Koenkk, 2021-05-21:
@snippem someone has to figure out how the new data should be parsed, but I currently have no clue how to do it.

ALaDoffe, 2021-05-22:
@Koenkk 
Hello since OTA update was removed for this device, i get this in HA logs :

WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'update_available' when rendering '{{ value_json.update_available}}'

More on FrontEnd, i have still references of Ota Update: 

[image]

I tested : 
     remove/rejoin the devices,
     remove mosquitto retain messages

Thx

Koenkk, 2021-05-22:
@ALaDoffe done!

snippem, 2021-05-22:
> @snippem someone has to figure out how the new data should be parsed, but I currently have no clue how to do it.

Just interested.
With the previous firmware the data has to be parsed to I presume? How did they do it with the previous firmware version .
And this new firmware version would this be shipped with new devices ?

stickpin, 2021-06-02:
I don't know if anyone interested, but for me personally, new firmware solved a very annoying issue.

With old firmware Zigbee Groups was not working for L2, only L1 was working.
After the upgrade, everything is working as expected.
Yes, I've also lost the energy meter, temperature meter and voltage, which I anyway never used with those relays, but I've got properly working Zigbee groups. 

@Koenkk I don't know if it's possible but maybe to allow people to do an update manually somehow?

McGiverGim, 2021-07-03:
Please don't close, I think this will bite us later if new products start to come with this firmware...

closed 2021-08-11 by github-actions[bot]

Galorhallen, 2021-08-21:
I've the same problem, with another device Aquara smart plug EU. A similar issue is #7745

olylak, 2021-12-21:
I'm facing the same issue with DLKZMK11LM.
But my device was not working OOTB in terms of null values like energy, temp and voltage. I assume it was already flashed with a newer firmare that couses those issues. 
Any updates on this topic?

olylak, 2022-01-10:
Any solution?

egolfeld, 2022-01-21:
how to manually update?

McGiverGim, 2022-02-03:
@egolfeld To do a manual upgrade you have this instructions: https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files

To all the other, there is a new firmware available, but the OTA function is disabled in the device until verified if this new firmware works. Maybe you want to upgrade manually to see if it fixes the issues.

leroyloren, 2022-02-03:
No, still same.

[image]

stickpin, 2022-02-03:
For those who use this relay with Zigbee Group, after an update to version 9-22-2021 Zigbee groups still functioning as expected.
Energy, Voltage and Temperature are still not present.

McGiverGim, 2022-02-03:
Someone has tested if others, like ZHA, have the same problem?

leroyloren, 2022-02-03:
@Koenkk
Would it not be possible to display otherwise current FW versions for Aqara devices? At the moment, there is a date that has nothing to do with the release of the update, or strings like "2020www." For Aqara, build  as fileVersion versions are listed everywhere.

`debug 2022-02-03 20:50:04: Device 'Zebrik' requested OTA
debug 2022-02-03 20:50:04: Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":257,"fileVersion":47}'
debug 2022-02-03 20:50:04: ZigbeeOTA: downloaded main index
debug 2022-02-03 20:50:04: Is new image available for '0x00158d0006c8fb74', current '{"fieldControl":0,"manufacturerCode":4447,"imageType":257,"fileVersion":47}', latest meta '{"fileVersion":47,"fileSize":182206,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/lumi.relay.c2acn01_20211201_v0047.ota","sha512":"66380db41c6f9cc8d4c74809caff46c0cb1b691046c3e490fe8d98f2fa2a5c6acceb1d27456a3d7efd50cde92ef150c5adbc0287538a7218540400385bd2956d"}'
`

For example from Aqara Home app:
[image]

vs Z2M
[image]

Koenkk, 2022-02-04:
Added another possible fix, can someone check if it works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

stickpin, 2022-02-04:
> Added another possible fix, can someone check if it works?
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Nothing is changed on my side.
Looks exactly the same.

[image]

leroyloren, 2022-02-05:
It is strange that the Aqara app offers only (see. picture), no voltage can be found anywhere. Is it not possible that they simply removed these values?

[image]

z-x-f, 2022-02-05:
> It is strange that the Aqara app offers only (see. picture), no voltage can be found anywhere. Is it not possible that they simply removed these values?

aqara app does not provide voltage display.

Koenkk, 2022-02-06:
@z-x-f but could it be that aqara entirely removed it in the latest fw?

leroyloren, 2022-02-06:
Sure, but min. it shows the total consumption and Z2M also does not provide it.

stickpin, 2022-02-06:
> Sure, but min. it shows the total consumption and Z2M also does not provide it.

@leroyloren Z2M showing current consumption, exactly as Aqara app:
[image]

z-x-f, 2022-02-06:
> Sure, but min. it shows the total consumption and Z2M also does not provide it.
@Koenkk  Ah, i mean in aqara app, any device doesn't provide voltage display.
As for this device, I might buy one in the future to test if it doesn't display properties like voltage.

leroyloren, 2022-02-06:
@stickpin i think total energy consuption.

localbuddha, 2022-02-15:
@Koenkk Hey! Just want let you know my [LLKZMK11LM](https://www.zigbee2mqtt.io/devices/LLKZMK11LM.html#xiaomi-llkzmk11lm) relay was updated successfully to 9-22-2021, and it shows power, temperature and voltage.

[image]

leroyloren, 2022-02-15:
Hi, it's interesting, I have the latest FW via Aqara Hub (0.0.0_47) and it doesn't work. Interestingly, the Z2M still displays the version02-27-2019 and no updates availability when trying to OTA over Z2M.
So where does the "version" of the firmware in Z2M come from, when, according to Aqara, the last one is listed 9-22-2021?
I don't see anything about the version when pairing.

`Device '' is supported, identified as: Xiaomi Aqara wireless relay controller (LLKZMK11LM)
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara wireless relay controller","exposes":[{"access":5,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"endpoint":"l1","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l1","name":"state","property":"state_l1","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l2","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l2","name":"state","property":"state_l2","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":3,"description":"Enabling prevents both relais being on at the same time","name":"interlock","property":"interlock","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"LLKZMK11LM","options":[{"access":2,"description":"Number of digits after decimal point for temperature, takes into effect on next report of device.","name":"temperature_precision","property":"temperature_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the temperature value (absolute offset), takes into effect on next report of device.","name":"temperature_calibration","property":"temperature_calibration","type":"numeric"}],"supports_ota":true,"vendor":"Xiaomi"},"friendly_name":"","ieee_address":"","status":"successful","supported":true},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Aqara wireless relay controller","friendly_name":"","model":"LLKZMK11LM","supported":true,"vendor":"Xiaomi"},"type":"pairing"}'
Debug Received Zigbee message from '', type 'raw', cluster 'genBasic', data '{"data":[28,95,17,3,10,1,255,66,62,3,40,0,5,33,34,0,8,33,47,18,9,33,0,9,100,16,0,101,16,0,110,32,0,111,32,0,148,32,1,150,57,46,4,22,69,151,57,240,41,164,59,152,57,0,0,0,0,155,33,0,0,156,32,1,10,33,0,0,12,40,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/', payload '{"energy":null,"interlock":null,"last_seen":"2022-02-15T18:29:09+01:00","linkquality":178,"power":0,"state_l1":null,"state_l2":null,"temperature":null,"update":{"state":null},"update_available":null,"voltage":null}'
Debug Received Zigbee message from '', type 'attributeReport', cluster 'genBasic', data '{"65521":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/', payload '{"energy":null,"interlock":null,"last_seen":"2022-02-15T18:29:10+01:00","linkquality":185,"power":0,"state_l1":null,"state_l2":null,"temperature":null,"update":{"state":null},"update_available":null,"voltage":null}'
Debug Received Zigbee message from '', type 'attributeReport', cluster 'genBasic', data '{"65521":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/', payload '{"energy":null,"interlock":null,"last_seen":"2022-02-15T18:29:14+01:00","linkquality":167,"power":0,"state_l1":null,"state_l2":null,"temperature":null,"update":{"state":null},"update_available":null,"voltage":null}'
Debug Received Zigbee message from '', type 'attributeReport', cluster 'genBasic', data '{"65521":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/', payload '{"energy":null,"interlock":null,"last_seen":"2022-02-15T18:29:22+01:00","linkquality":174,"power":0,"state_l1":null,"state_l2":null,"temperature":null,"update":{"state":null},"update_available":null,"voltage":null}'
Debug Received Zigbee message from '', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/', payload '{"energy":null,"interlock":null,"last_seen":"2022-02-15T18:30:02+01:00","linkquality":171,"power":0,"state_l1":null,"state_l2":null,"temperature":null,"update":{"state":null},"update_available":null,"voltage":null}'`

[image]

[image]

localbuddha, 2022-02-15:
> Hi, it's interesting, I have the latest FW via Aqara Hub (0.0.0_47) and it doesn't work. Interestingly, the Z2M still displays the version02-27-2019 and no updates availability when trying to OTA [...]

Just to be sure: are you on Zegbee2MQTT Edge (DEV)? I'm updating all my relays one by one on Edge (DEV), not on release.

leroyloren, 2022-02-15:
So I take it back, I "forced" the firmware version update from the cluster via the console, it is now displayed correctly, but even so, the energy, etc. is not displayed.

v.1.23.0-dev commit: [6394d0f](https://github.com/Koenkk/zigbee2mqtt/commit/6394d0f)

database.db
`{"id":23,"type":"Router","ieeeAddr":"0x","nwkAddr":60836,"manufId":4151,"manufName":"LUMI","powerSource":"DC Source","modelId":"lumi.relay.c2acn01","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,1,2,10,6,16,2820,12],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65521":0,"modelId":"lumi.relay.c2acn01","appVersion":47,"zclVersion":0,"stackVersion":2,"hwVersion":18,"manufacturerName":"LUMI","dateCode":"9-22-2021","powerSource":4}},"genAnalogInput":{"attributes":{"presentValue":0}},"genOnOff":{"attributes":{"61440":218103832,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":257,"inClusterList":[6,16,4,5],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"61440":250455228,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":47,"stackVersion":2,"hwVersion":18,"dateCode":"9-22-2021","zclVersion":0,"interviewCompleted":true,"meta":{},"lastSeen":1644950524325,"defaultSendRequestWhen":"immediate"}`

leroyloren, 2022-02-15:
@[localbuddha](https://github.com/localbuddha) Can you please send your device from database.db?

localbuddha, 2022-02-15:
> @[localbuddha](https://github.com/localbuddha) Can you please send your device from database.db?

`{"id":4,"type":"Router","ieeeAddr":"0x00158d0004285246","nwkAddr":13370,"manufId":4151,"manufName":"LUMI","powerSource":"DC Source","modelId":"lumi.relay.c2acn01","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,1,2,10,6,16,2820,12],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65281":{"3":38,"5":179,"7":"0x0000000000000000","8":4644,"9":256,"100":0,"101":0,"110":0,"111":0,"148":25,"149":0.004062499850988388,"150":2255.032470703125,"151":0.0035429000854492188,"152":0.20125150680541992,"155":0},"65521":0,"modelId":"lumi.relay.c2acn01","appVersion":1,"manufacturerName":"LUMI","powerSource":4,"zclVersion":0,"stackVersion":2,"hwVersion":18,"dateCode":"9-22-2021"}},"genOnOff":{"attributes":{"61440":53754414,"onOff":0}},"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":257,"inClusterList":[6,16,4,5],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"61440":117440552,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":18,"dateCode":"9-22-2021","zclVersion":0,"interviewCompleted":true,"meta":{},"lastSeen":1644948514019,"defaultSendRequestWhen":"immediate"}`

leroyloren, 2022-02-15:
@localbuddha Thx. 👍

@Koenkk , can it be related, please?

**localbuddha**
`"65281": {"3": 38, "5": 179, "7": "0x0000000000000000", "8": 4644, "9": 256, "100": 0, "101": 0, " 110 ": 0," 111 ": 0," 148 ": 25," 149 ": 0.004062499850988388," 150 ": 2255.032470703125," 151 ": 0.0035429000854492188," 152 ": 0.20125150680541992," 155 ": 0}`

**vs mine:**
`{"65521": 0,`

localbuddha, 2022-02-16:
Little upd: 4 of 19 relays failed to update:

`Update of 'Relay_7' failed (Update failed with reason: 'invalid image')`

Koenkk, 2022-02-16:
@leroyloren  what coordinator/firmware version are you using? Make sure to use the latest and after flashing it repair the Xiaomi device

leroyloren, 2022-02-16:
Zzh! 20211217

Koenkk, 2022-02-16:
@leroyloren and did you repair the device after updating to this fw? To make sure, try to repair it now.

leroyloren, 2022-02-16:
Yes, several times. Last time yesterday, when we wrote about it in this discussion.

https://github.com/Koenkk/zigbee2mqtt/issues/7112#issuecomment-1040571911

Pairing log.

dinamik1488, 2022-02-16:
does energy monitoring work?

localbuddha, 2022-02-16:
I've just checked my setup, guys, and I need to say "sorry". My Sonoff Zigbee 3.0 USB Dongle Plus looks faulty, and all firmware updates for relays was completed on old CC2531 dongle v. 20190608

[26 more replies]