Koenkk/zigbee2mqtt #7113: Philips Hue Garnea - Not Supported Error

beasthouse-au, 2021-04-18:
Trying to pair my Philips Hue Garnea lights with z2m. A couple have worked, however the others are saying that they are unsupported, even though the [documentation would indicate otherwise](https://www.zigbee2mqtt.io/devices/5110131H5.html).

I get the following error code in the logs:

`Failed to interview '0x0017880108d4474b', device has not successfully been paired`

Then it does appear in devices:

<blockquote class="imgur-embed-pub" lang="en" data-id="pUwoF1Z"><a href="https://imgur.com/pUwoF1Z">View post on imgur.com</a></blockquote><script async src="//s.imgur.com/min/embed.js" charset="utf-8"></script>

<blockquote class="imgur-embed-pub" lang="en" data-id="xVgKv5d"><a href="https://imgur.com/xVgKv5d">View post on imgur.com</a></blockquote><script async src="//s.imgur.com/min/embed.js" charset="utf-8"></script>

All devices were updated in the Hue app prior to bringing across to z2m

Zigbee2MQTT version: 1.18.2
Adapter hardware: CC2652R
Adapter firmware version: CC2652R_coordinator_20210120

danieledwardgeorgehitchcock, 2021-04-20:
Could you capture a debug log during one of the failed devices pairing, and post it up here?

beasthouse-au, 2021-04-20:
Waited 24 hours, tried again and they worked fine. No idea what caused it, go Zigbee!

closed 2021-04-20 by beasthouse-au