Koenkk/zigbee2mqtt #7114: PIR Sensor (ZM-35ZH-Q)

JayChun, 2021-04-18:
Always thanks for z2m!!
I bought some PIR sensors which is really similar to Konke PIR Sensor.
But the sensors don't pair to z2m. Please check the device information below to add a new device.

## Information about the device + link
https://www.aliexpress.com/item/1005002327925412.html?spm=a2g0s.9042311.0.0.20d34c4dkZFXMg
It is really like to Konke PIR Senor. but i slightly more bigger than Konke.

## data/database.db entry of the device
{"id":28,"type":"Unknown","ieeeAddr":"0x00124b000f6099ce","nwkAddr":28798,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1618740078502}

philippfabian, 2021-04-20:
Same here, would be great to have this in z2m.

Device: PIR Sensor
Model: ZM-35ZH-Q
URL : [https://www.aliexpress.com/item/1005002324940947.html?spm=a2g0s.9042311.0.0.48d84c4dnD8a32](url)

**data/database.db entry of the device**
{"id":9,"type":"EndDevice","ieeeAddr":"0x00124b00026003c8","nwkAddr":55015,"manufId":0,"epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280],"outClusterList":[0,3,1,1280],"clusters":{"ssIasZone":{"attributes":{"zoneStatus":1}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1618938428133}

chrisrock1984, 2021-05-03:
same for me - does not pair

chrisrock1984, 2021-05-07:
getting following messages in the log:
Debug Received Zigbee message from '0x00124b00025565a0', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x00124b00025565a0', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing

haverland, 2021-05-13:
I'd added debug to herdsman-controller. On pairing:
igbee2mqtt    | Zigbee2MQTT:debug 2021-05-14 00:04:20: Received Zigbee message from '0x00124b0006296020', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:warn  2021-05-14 00:04:20: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
zigbee2mqtt    | Zigbee2MQTT:warn  2021-05-14 00:04:20: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt    | 2021-05-13T22:04:21.522Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":55,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":2,"dataType":33,"attrData":1}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":1690,"endpoint":1,"linkquality":105,"groupID":0,"wasBroadcast":false}'
zigbee2mqtt    | Zigbee2MQTT:debug 2021-05-14 00:04:21: Received Zigbee message from '0x00124b0006296020', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0

bootuz-dinamon, 2021-05-25:
https://github.com/Koenkk/zigbee2mqtt/issues/7564   

look here.
They are one and the same device.

converter in the same place

closed 2021-07-03 by github-actions[bot]

SinnoTech, 2021-10-21:
Has this issue been resolved?

actabi, 2021-12-03:
I also have the same problem. Do you have any news?

DaHopi, 2021-12-09:
Same here.. bought two of this devices.. but paring not possible.. what do you guys need?
BR

Koenkk, 2021-12-09:
Make sure you are on z2m 1.22.1

casamontico51, 2022-01-03:
i am on version 1.22.2 and is not pairing.

JayChun, 2022-02-01:
longtime...
Just now I checked the sensor. but it still can't.
z2m 1.22.2

evlo, 2022-06-26:
offtopic, but i can't find info about this anywhere - how does this sensor measure light they mention on aliexpress? Is it only dark/light or is it like 1 to 1000 lux?

JayChun, 2022-06-26:
> offtopic, but i can't find info about this anywhere - how does this sensor measure light they mention on aliexpress? Is it only dark/light or is it like 1 to 1000 lux?

The sensor was supported by Z2M maybe officailly after 1.22.1.
And please confirm this screenshot you need. 
it can't detect light motion or illumination, just have PIR detection. 

[image]

evlo, 2022-06-26:
Nevermind i figured Ir out, there Is vsrsion od this sensor package with And wo lux meter