Koenkk/zigbee2mqtt #7128: Error on MOES Zigbee+RF curtain module (TS130F?) calibration

FaBRiK74, 2021-04-19:
## What happened
When I try to calibrate the module it gives error (see logs below)

## What did you expect to happen
Able to calibrate:)

I also have a TS130F based curtain module from Lonsonho (QS-Zigbee-C01) which works perfectla and I am able to calibrate (just tried again and was OK).
This MOES module opens and closes, shows the percentage, I can switch the rmotor reversal but not able to switch on the calibration.
There is also a "backlight_mode" in exposes but I'm pretty sure t should not be there :)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2652RB (Slaesh's)

Log:
Error Publish 'set' 'calibration' to 'Zsolti teraszredőny' failed: 'Error: Write 0x60a423fffe6f0548/1 closuresWindowCovering({"tuyaCalibration":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug Error: Write 0x60a423fffe6f0548/1 closuresWindowCovering({"tuyaCalibration":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:196:23) at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:225:26) at Generator.next (<anonymous>) at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58)
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'calibration' to 'Zsolti teraszredőny' failed: 'Error: Write 0x60a423fffe6f0548/1 closuresWindowCovering({\"tuyaCalibration\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"Zsolti teraszredőny"},"type":"zigbee_publish_error"}'

FaBRiK74, 2021-04-25:
Is there any suggestion how to solve?

Khau, 2021-05-03:
I'm also interested. I bought this module but without the calibration it's not usable.

thierry-rhone, 2021-05-04:
Hi have this module.
To calibrate it https://www.zigbee2mqtt.io/devices/TS130F.html#tuya-ts130f
1 It's important to start procedure curtain opened. If you start curtain closed, it invert opened closed
2 do not chang Backlight_mode. it has a bug. If you chang it, the stop button lights up at the same time as open or close button.

FaBRiK74, 2021-05-04:
Hi Thierry,

As I wrote, the Calibrate button in Exposes under Z2M for this module is inactive. I can't switch on. When I try to switch on, I get error message. See my original post.

thierry-rhone, 2021-05-04:
Hi Fabrik,
sorry, I did not understand correctly
I have this switch, https://www.aliexpress.com/item/1005002060255177.html?spm=a2g0s.9042311.0.0.5bba4c4dAF05w3
It is recognized as TS130F. It works.

Koenkk, 2021-05-04:
`failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"Zsolti teraszredőny"},"type":"zigbee_publish_error"}'` indicates that your specific device does not support calibration.

Khau, 2021-05-04:
If we look on the vendor page: https://www.moeshouse.com/products/zigbee-diy-rf433-smart-curtain-switch-module-for-electric-motorized-roller-blinds-shutter-motor-2mqtt-available?_pos=2&_sid=5fe5449ce&_ss=r
It is mention that we can calibrate it via the mobile app. Maybe it's an another attribute than for the Lonsonho QS-Zigbee-C01.
For me it means we should maybe use a different implementation than TS130F.

FaBRiK74, 2021-05-05:
But it definitely supports calibration. Also the manual details the calibration method in application. 
The exact product: https://www.aliexpress.com/item/1005002382743227.html?spm=a2g0s.9042311.0.0.27424c4dq0XV0s

Khau, 2021-05-07:
I commented the device configuration with properties zigbeeModel: ['TS130F'], model: 'TS130F', vendor: 'TuYa' because our Moes module use that configuration because there is no manufacturerName. Then I paired my device again and I got these logs:
```
Successfully interviewed '0x60a423fffeff1ce3', device has successfully been paired
Warning Device '0x60a423fffeff1ce3' with Zigbee model 'TS130F' and manufacturer name '_TZ3000_1dd0d5yi' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x60a423fffeff1ce3","ieee_address":"0x60a423fffeff1ce3","status":"successful","supported":false},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x60a423fffeff1ce3","supported":false},"type":"pairing"}'
Debug Received Zigbee message from '0x60a423fffeff1ce3', type 'attributeReport', cluster 'closuresWindowCovering', data '{"61443":100}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model 'TS130F' and manufacturer name '_TZ3000_1dd0d5yi'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Debug Received Zigbee message from '0x60a423fffeff1ce3', type 'attributeReport', cluster 'closuresWindowCovering', data '{"tuyaMovingState":1}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model 'TS130F' and manufacturer name '_TZ3000_1dd0d5yi'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Debug Received Zigbee message from '0x60a423fffeff1ce3', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model 'TS130F' and manufacturer name '_TZ3000_1dd0d5yi'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Debug Received Zigbee message from '0x60a423fffeff1ce3', type 'attributeReport', cluster 'closuresWindowCovering', data '{"tuyaMotorReversal":1}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model 'TS130F' and manufacturer name '_TZ3000_1dd0d5yi'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Debug Received Zigbee message from '0x60a423fffeff1ce3', type 'attributeReport', cluster 'genOnOff', data '{"20480":0}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model 'TS130F' and manufacturer name '_TZ3000_1dd0d5yi'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Debug Received Zigbee message from '0x60a423fffeff1ce3', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model 'TS130F' and manufacturer name '_TZ3000_1dd0d5yi'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Debug Received Zigbee message from '0x60a423fffeff1ce3', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model 'TS130F' and manufacturer name '_TZ3000_1dd0d5yi'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```
So we can se that the manufacturer name is '_TZ3000_1dd0d5yi'. Then I copied the device configuration of QS-Zigbee-C01 which is similar to the moes one. I having now a dedicate config for my module.

Next the calibration. The module use the function tuya_cover_calibration but it doens't work:
```
Publish 'set' 'calibration' to '0x60a423fffeff1ce3' failed: 'Error: Write 0x60a423fffeff1ce3/1 closuresWindowCovering({"tuyaCalibration":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug Error: Write 0x60a423fffeff1ce3/1 closuresWindowCovering({"tuyaCalibration":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:196:23) at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:225:26) at Generator.next (<anonymous>) at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58)
```
Is there a way to debug this device and adapt the calibration function?

FaBRiK74, 2021-05-12:
My Tuya Zigbee Hub just arrived and tested this MOES Zigbee+RF Curtain Switch Moule (identidfied as TS130F). I could calibrate it via the Smart Life app, but still cannot calibrate in Z2M. Gives the same error.

klio-klio, 2021-05-19:
i have the same problem. it is recognized incorrectly. i can't calibrate. The time for opening and closing is too short. can't you enter that directly? Sorry ... I'm new here.

agonbar, 2021-06-06:
@FaBRiK74 There´s this manual:

https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Could you maybe try to sniff the message sent when using the app to calibrate? Right now I have to be offline until the 16th, but after that I could help you with the process. If not I can try and order some parts to Aliexpress and do it myself, but it´s going take quite some time.

vampywiz17, 2021-06-27:
Hello there, 

Interest this device, but i don't know, what product is the theme in this issue?

this one:

https://www.banggood.com/MoesHouse-ZigBee+RF433-Smart-Curtain-Switch-Module-Motorized-Roller-Shutter-Blinds-Motor-2MQTT-Tuya-Smart-APP-Alexa-Google-Home-p-1835998.html?cur_warehouse=CN&ID=6292509

or this one:

https://www.aliexpress.com/item/1005002060255177.html?spm=a2g0s.9042311.0.0.5bba4c4dAF05w3

The first one is the interesting to me, because it is a really good Shelly 2.5 alternative (the + RF control a really good "fail-safe" control mode)

If i read correctly the relay is possible to calibrate to the % based control right? It is only bugged now?

aelias-eu, 2021-06-28:
Hi,
@vampywiz17  - i have the first one - By default, it's identified as TS130F, but the **calibration does not work** (from zigbee2mqtt). By default the opening/closing times are about 10 seconds.   I have contacted the moes support about technical details, but  their answer is not surprising:
```
Dear, 
Nice day!
Hope my email finds you health and wealth.

This is Zoe with Moes and I am here to serve you.Thanks so much for your email.
No dear,for this is our tuya smart device and it is highly compatible with tuya zigbee
hub for the functions in Tuya smart App.
Thanks so much for your precious time and kindness.

Follow our Facebook account for more promotions:https://www.facebook.com/moessmart/
Looking forward to your reply.

Best regards.
Yours sincerely
```

So I don't know what to think, but i know that they will not help us :) and probably the only possible way is to sniff it.

@FaBRiK74  - if you calibrate the switch via Tuya hub, is it possible to disconnect it from Tuya and connect into zigbee2mqtt by preserving new calibration data? Or will the calibration be lost by module reset? Could you please confirm/verify? Thanks

FaBRiK74, 2021-06-29:
@aelias-eu unfortunately after I calibrate with Tuya hub and I link to HA's stick it looses the calibration

aelias-eu, 2021-06-29:
@FaBRiK74  - that is exactly what i was afraid of :) I'll have to wait for my tuya hub to arrive and then I will try to sniff the communication between hub & the Moe's module. I'm looking forward to it as this will be my first Zigbee sniffing attempt (I'm not counting my today's successful  PoC). 
I will report back here...

vampywiz17, 2021-06-29:
@FaBRiK74 

You able to try to bind the Zigbee RF relay with a remote? It possible?

aelias-eu, 2021-06-29:
Rf pairing works fine.

FelR, 2021-06-30:
@aelias-eu Did you manage to calibrate the switch using the RF remote? In my case binding with the remote worked well, too, but it only moves the blinds in the standard calibration boundaries. Haven’t found a way yet to initiate calibration using the remote

aelias-eu, 2021-06-30:
@FelR no, if it's possible, I don't know how - RF remote and also the hardware inputs (S1, S2) are both just executing the movement up/down and this is respecting calibration boundaries. Maybe there is a combination of S1,S2 and/or the reset button, but that would require firmware reverse engineering (or moes documentation leak). I will just wait for the zigbee scan when my tuya hub arrives. Seems to me as the easiest path for now. I already have 3 pcs of these devices, so it's quite important to me...

vampywiz17, 2021-07-02:
@aelias-eu 

I think is not possible to calibrate it via RF. The only way is the Tuya app and (hope so) the Zigbee2mqtt, if we able to solve this issue :)

Akkalabeth, 2021-07-09:
I'm hoping for a solution too, as currently it is quite unworkable with my sunscreen. I've connected it to an Osram Smart Mini remote, with "up" button setting the screen to 100% and inverting the motor, "down" is back to 0% and inverting motor. So when pressing "up", it moves about 50cm before reaching 100%, then you can press "down" again to move it another 50 (by inverting the motor it keeps going in the same direction)

As I want to control the screen by "holding" the up/down button, i'm actually considering getting a 2 gang module, connect those to the in & out wire and having it remote controlled this way. Only worry I got is that there might be a situation (kids playing with the remote) where both in & out to the motor are powered on, I'm currently checking if this would do any damage to the motor or whether it has a build in system to prevent these situations.

vampywiz17, 2021-07-09:
@Akkalabeth 

You bind the remote or group it? So it possible this device? It works if you turn off coordiantor?

Akkalabeth, 2021-07-09:
No I don't think so, I just used Automations for now, still playing around learning the whole system :)

SkyByte, 2021-07-09:
Hello,
i can confirm this issue. The 'MOES Zigbee + RF Curtain Switch Module' cannot being calibrated in Z2M, because for this switch (the version including 433MHz RF!) Z2M sends an 'UNSUPPORTED_ATTRIBUTE'.
That's too bad, because without calibration this switch is absolutely useless, unless your curtain needs only 10 sec. (=preset) for opening/closing.  ;-)
Hoping this issue will be fixed soon.

`
Error:
Publish 'set' 'calibration' to '0x60a423fffeff2bc7' failed: 'Error: Write 0x60a423fffeff2bc7/1 closuresWindowCovering({"tuyaCalibration":0}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Info:
MQTT publish: topic 'tele/zigbee/bridge/log', payload '{"message":"Publish 'set' 'calibration' to '0x60a423fffeff2bc7' failed: 'Error: Write 0x60a423fffeff2bc7/1 closuresWindowCovering({\"tuyaCalibration\":0}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"0x60a423fffeff2bc7"},"type":"zigbee_publish_error"}'
`

SkyByte, 2021-07-17:
> @aelias-eu unfortunately after I calibrate with Tuya hub and I link to HA's stick it looses the calibration

I can not confirm this.

Today, my wireless Tuya Zigbee Gateway arrived. I added the Curtain Switch Module to the Smart Life app and was able to enter the amount of seconds for switching length to an individual value. In my case i successfully changed it to 60 seconds.
After that, i did not disconnect the module from Tuya, but i joined the module to Z2M pressing the joining button for about 10 seconds as usual. The module is then automatically shown as 'offline' in the Tuya Smart Life app.

And what a surprise!! The setting has **not** been lost. Sure, the calibration command still doesn't work in Z2M, but the time period is still at 60 seconds, even if you delete this device in Tuya Smart Life app later.

So i would recommend to buy or borrow a Tuya Zigbee Gateway to set the amount of seconds once to a sufficient value, until this issue will be fixed in Z2M.

aelias-eu, 2021-07-17:
Thanks for the information. I'm still waiting for the hub to arrive :(

vampywiz17, 2021-07-17:
@SkyByte 

You able to do this?

https://github.com/Koenkk/zigbee2mqtt/issues/7128#issuecomment-855426841

aelias-eu, 2021-08-04:
Sorry for the delay, but still no luck. The first gateway did not arrive, I hlave ordered a second one from another seller, but it looks like this one will follow the same pattern :(

aelias-eu, 2021-08-10:
OK  people, it's working, the gateway came yesterday, today i did a bit of sniffing and here is the result:

New device needs to be added in file **zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/moes.js** : 
```
    {
        fingerprint: [{modelID: 'TS130F', manufacturerName: '_TZ3000_1dd0d5yi'}],
        model: 'MS-108ZR',
	vendor: 'Moes',
        description: 'Zigbee + RF Curtain Switch Module',
        supports: 'open, close, stop, position',
        fromZigbee: [fz.tuya_cover_options, fz.cover_position_tilt ],
        toZigbee: [tz.cover_state, tz.moes_cover_calibration, tz.cover_position_tilt, tz.tuya_cover_reversal ],
	exposes: [e.cover_position(), exposes.numeric('calibration_time'), exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']),
            exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF')], 
    },

```
next you have to edit the _tuya_cover_options_ section in **zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js**, you need to insert the moesCalibrationTime
```
    tuya_cover_options: {
        cluster: 'closuresWindowCovering',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('tuyaMovingState')) {
                const value = msg.data['tuyaMovingState'];
                const movingLookup = {0: 'UP', 1: 'STOP', 2: 'DOWN'};
                result.moving = movingLookup[value];
            }
            if (msg.data.hasOwnProperty('tuyaCalibration')) {
                const value = msg.data['tuyaCalibration'];
                const calibrationLookup = {0: 'ON', 1: 'OFF'};
                result.calibration = calibrationLookup[value];
            }
            if (msg.data.hasOwnProperty('moesCalibrationTime')) {
                const value = parseFloat(msg.data['moesCalibrationTime']) / 10.0;
                result.calibration_time = value;
            }
            if (msg.data.hasOwnProperty('tuyaMotorReversal')) {
                const value = msg.data['tuyaMotorReversal'];
                const reversalLookup = {0: 'OFF', 1: 'ON'};
                result.motor_reversal = reversalLookup[value];
            }
            return result;
        },
    },

```
next you have to add the _moes_cover_calibration_ support in **zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js**

```
    moes_cover_calibration: {
        key: ['calibration_time'],
        convertSet: async (entity, key, value, meta) => {
            const calibration = value *10;
            await entity.write('closuresWindowCovering', {moesCalibrationTime: calibration});
            return {state: {calibration_time: value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('closuresWindowCovering', ['moesCalibrationTime']);
        },
    },
```

and the last one is to extend the _closuresWindowCovering_ section of **zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/definition/cluster.js** with tne moesCalibrationTime
```
    closuresWindowCovering: {
        ID: 258,
        attributes: {
            windowCoveringType: { ID: 0, type: dataType_1.default.enum8 },
[... 17 lines cut ...]
            windowCoveringMode: { ID: 23, type: dataType_1.default.bitmap8 },
            intermediateSetpointsLift: { ID: 24, type: dataType_1.default.octetStr },
            intermediateSetpointsTilt: { ID: 25, type: dataType_1.default.octetStr },
            ubisysTurnaroundGuardTime: { ID: 0x1000, type: dataType_1.default.uint8, manufacturerCode: manufacturerCode_1.default.Ubisys },
            ubisysLiftToTiltTransitionSteps: { ID: 0x1001, type: dataType_1.default.uint16, manufacturerCode: manufacturerCode_1.default.Ubisys },
            ubisysTotalSteps: { ID: 0x1002, type: dataType_1.default.uint16, manufacturerCode: manufacturerCode_1.default.Ubisys },
            ubisysLiftToTiltTransitionSteps2: { ID: 0x1003, type: dataType_1.default.uint16, manufacturerCode: manufacturerCode_1.default.Ubisys },
            ubisysTotalSteps2: { ID: 0x1004, type: dataType_1.default.uint16, manufacturerCode: manufacturerCode_1.default.Ubisys },
            ubisysAdditionalSteps: { ID: 0x1005, type: dataType_1.default.uint8, manufacturerCode: manufacturerCode_1.default.Ubisys },
            ubisysInactivePowerThreshold: { ID: 0x1006, type: dataType_1.default.uint16, manufacturerCode: manufacturerCode_1.default.Ubisys },
            ubisysStartupSteps: { ID: 0x1007, type: dataType_1.default.uint16, manufacturerCode: manufacturerCode_1.default.Ubisys },
            tuyaMovingState: { ID: 0xf000, type: dataType_1.default.enum8 },
            tuyaCalibration: { ID: 0xf001, type: dataType_1.default.enum8 },
            tuyaMotorReversal: { ID: 0xf002, type: dataType_1.default.enum8 },
	    moesCalibrationTime: { ID: 0xf003, type: dataType_1.default.uint16 }, // aelias 20210810
        },

```

This should be all
Complete files are here:
[moes.tar.gz](https://github.com/Koenkk/zigbee2mqtt/files/6960942/moes.tar.gz)

You can the set the calibration time via mqtt by sending the value in seconds via /set
{"calibration_time":5}

vampywiz17, 2021-08-10:
@aelias-eu 

Great, thanks a lot !
only need a pull req. right? :)

triplecillas, 2021-08-31:
First of all, thanks to @aelias-eu for his work and all the people that helped to solve this issue.
I have one of this devices and want to calibrate it.
As I am new in zigbee2mqtt (an also not experienced in Linux) I cannot find the folders and files mentioned in the solution, so I cannot modify the files.
Can someone of you please explain me how to find the paths and or files?

Danit2, 2021-09-01:
I tried to change the calibration time.
Unfortunately I always get an error with: "No converter available for 'calibration_time' (40)".
Can someone please help me? @aelias-eu?
I have the latest version of zigbee2mqtt: 1.21.1
There the files would have to be adjusted right?

Can someone tell me where I can find the files from zigbee2mqtt? I use Home Assistant OS on a RPI4 with zigbee2mqtt as an add-on.

Thanks a lot.

aelias-eu, 2021-09-02:
Hi, 
I have to test it first, but it really looks like there is something wrong.

the whole **moes_cover_calibration** in *toZigbee.js* is missing
```
moes_cover_calibration: {
        key: ['calibration_time'],
        convertSet: async (entity, key, value, meta) => {
            const calibration = value *10;
            await entity.write('closuresWindowCovering', {moesCalibrationTime: calibration});
            return {state: {calibration_time: value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('closuresWindowCovering', ['moesCalibrationTime']);
        },
    },
```
it's OK in my fork https://github.com/aelias-eu/zigbee-herdsman-converters/blob/master/converters/toZigbee.js
but it's missing in the main repo https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js

I think there is the problem...
I reviewed the history of the pull request and it seems that the toZigbee.js file was not processed by the pull request...

MattWestb, 2021-09-02:
@aelias-eu Github was having problems for some days ago perhaps it was crashing then doing the merger and it was not completed.

Danit2, 2021-09-02:
I found a solution to how it works.
For everyone who uses the frontend at HomeAssistant.

Go to the device and click on "dev console".
[image]

Take the cluster "closures" and the attributt "moesCalibrationTime"
[image]

Then write your time in seconds * 10 in the field an press "write".

This will work at the moment.

triplecillas, 2021-09-02:
Yesss!!! 
@Danit2 it works!!!
Thanks a lot.

cgonzalez509, 2021-09-20:
I could not believe it, but @Danit2 solution worked for me using the dev tool, cluster Closures & attribute moesCalibrationTime=t*10. I have Zigbee2Mqtt version: 1.21.1-1.
Txs a lot!
[image]

Sangoku, 2021-10-30:
Did anyone discover how to set the type of the switch which is used  i opened a bug  here:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/3248

wogn8r, 2022-02-18:
@Danit2 thanks for the solution but I don't have dev console partition. Can you please tell me how to find dev console.

Danit2, 2022-02-19:
You dont need anymore the dev console. 
If you have the newest versuon of Z2M then it works normaly. 

https://www.zigbee2mqtt.io/devices/MS-108ZR.html#calibration-time-numeric

wogn8r, 2022-02-19:
@Danit2 Thank you for your help.  my repostory is out of date.  I updated and installed the latest version.  The device is now working properly.

ikhvostenkov, 2022-06-27:
@aelias-eu Thanks for posting attributes HEX values, that helped me a lot. I appreciate.

BubbaCZ, 2022-10-10:
Hello All, please to wrap it up, so if I use only smartlife app, there is no way how to prelong the running time from 10 seconds to something reasonable, I have to have Zigbee2mqtt and installed Home Assistance and only then there is a chance ?

arashipl, 2022-12-21:
@Koenkk As I understand your fixes should be included in latest hassio? In my instance (all updates installed) Z2MQTT detects my MS-108ZR as TS130F and - as expected - exposes incorrect calibration interface. 
As I understand hassio uses Z2MQTT docker image and I am not familiar with docker, spent most of the day looking for solution and got desperate enough to ask.

Koenkk, 2022-12-21:
@arashipl I don't see any recent fixes I did regarding this issue, but all latest changes are included in the edge version of the addon.

arashipl, 2022-12-21:
@Koenkk I am trying to check if my MS-108ZR is different somehow and if manufacturerName is _TZ3000_1dd0d5yi
Unfortunately I do not see this beeing output to debug (dump below). 
All I can see here that Vendor is TuYa, not Moes.

Is it possible to force set device to a given model?

```
Zigbee2MQTT:info  2022-12-21 15:12:44: Successfully interviewed '0x842e14fffe15cfb8', device has successfully been paired
Zigbee2MQTT:info  2022-12-21 15:12:44: Device '0x842e14fffe15cfb8' is supported, identified as: TuYa Curtain/blind switch (TS130F)
Zigbee2MQTT:info  2022-12-21 15:12:44: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Curtain/blind switch","exposes":[{"features":[{"access":3,"name":"state","property":"state","type":"binary","value_off":"CLOSE","value_on":"OPEN"},{"access":7,"description":"Position of this cover","name":"position","property":"position","type":"numeric","value_max":100,"value_min":0}],"type":"cover"},{"access":1,"name":"moving","property":"moving","type":"enum","values":["UP","STOP","DOWN"]},{"access":7,"name":"calibration","property":"calibration","type":"binary","value_off":"OFF","value_on":"ON"},{"access":7,"name":"backlight_mode","property":"backlight_mode","type":"enum","values":["LOW","MEDIUM","HIGH"]},{"access":7,"name":"motor_reversal","property":"motor_reversal","type":"binary","value_off":"OFF","value_on":"ON"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS130F","supports_ota":false,"vendor":"**TuYa**"},"friendly_name":"0x842e14fffe15cfb8","ieee_address":"0x842e14fffe15cfb8","status":"successful","supported":true},"type":"device_interview"}'
```

Koenkk, 2022-12-21:
@arashipl what z2m version is this?

arashipl, 2022-12-21:
@Koenkk home assistant addon, Current version: 1.18.1-1

Koenkk, 2022-12-21:
That's a very old version. 

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

arashipl, 2022-12-21:
This is embarassing...
Checked for updates a few times, never known about repo switch.
Sorry for taking up your time. Current version works like a charm.