Koenkk/zigbee2mqtt #7129: Add log level to Logs tab on Z2M gui

derekoharrow, 2021-04-19:
A minor feature improvement request.

It's great that you can see the Z2M log file in the web gui, but to change the log file level (from my usual Warn to Info or Debug) you have to go to the Settings tab, then Advanced, and then scroll down until you find the log level (about half way down).

It would make much more sense if this could be presented on the Logs tab so that it is easily and readily accessible.

Also, if possible, it would be great to have a "clear logs" button on that tab too.

McGiverGim, 2021-04-19:
Another thing that I miss is the date/time of the log...

Koenkk, 2021-04-19:
As explained here: https://github.com/Koenkk/zigbee2mqtt/issues/new/choose, please create frontend issues here: https://github.com/nurikk/z2m-frontend/issues

closed 2021-04-19 by Koenkk