Koenkk/zigbee2mqtt #7130: IKEA button/curtain integration broken with Docker 1.18.2

gimlichael, 2021-04-19:
## What happened
When pressing the button to open or close the blinds (both from IKEA) nothing happens.

## What did you expect to happen
I'll expect the curtains to roll up or down.

## How to reproduce it (minimal and precise)
koenkk/zigbee2mqtt:1.18.2

**Works perfectly fine with koenkk/zigbee2mqtt:1.17.1.**

gimlichael, 2021-04-19:
Well, this is awkward .. just took from 1.17.1 --> 1.18.2 one by one .. and now it works.

Sorry .. been a while since I last was messing around with my Zigbee setup.

Keep up the good work, K - and thank you for staying passionate about your project!

gimlichael, 2021-04-22:
Actually, and i discovered this by accident, there is a bug.

Lets say i restart my home-assistant for maintenance, then 1.18.2 - even after restart - will not capture anything from the IKEA devices.

BUT if i start out with a version prior to 1.18.2, then it works.

When i afterwards use 1.18.2 it works.

Yes, it sounds confusing - but this is the pattern i discovered.

vytautasgimbutas, 2021-05-16:
@Koenkk there's something wrong with 1.18.2 release. Maybe with the dependencies that have been updated?
I'm experiencing something similar but with Xiaomi Aqara Curtain Motor. Using latest HomeAssistant version.

1) 1.18.1 - works after homeassistant restart
2) After upgrading to 1.18.2 curtains still work in homeassistant
3) After homeassistant restart curtains don't work anymore

1.18.1:
```
luna@luna:~/docker-compose/zigbee2mqtt$ docker logs -f zigbee2mqtt --tail 500|grep 'zigbee2mqtt/0x00158d00045ab3d6'
Zigbee2MQTT:info  2021-05-16 14:46:09: MQTT publish: topic 'zigbee2mqtt/0x00158d00045ab3d6', payload '{"options":{"hand_open":true,"reverse_direction":true},"position":100,"running":false,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-05-16 14:46:09: MQTT publish: topic 'homeassistant/cover/0x00158d00045ab3d6/cover/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x00158d00045ab3d6/set","device":{"identifiers":["zigbee2mqtt_0x00158d00045ab3d6"],"manufacturer":"Xiaomi","model":"Aqara curtain motor (ZNCLDJ11LM)","name":"0x00158d00045ab3d6","sw_version":"Zigbee2MQTT 1.18.1"},"json_attributes_topic":"zigbee2mqtt/0x00158d00045ab3d6","name":"0x00158d00045ab3d6","position_topic":"zigbee2mqtt/0x00158d00045ab3d6","set_position_template":"{ \"position\": {{ position }} }","set_position_topic":"zigbee2mqtt/0x00158d00045ab3d6/set","unique_id":"0x00158d00045ab3d6_cover_zigbee2mqtt","value_template":"{{ value_json.position }}"}'
Zigbee2MQTT:info  2021-05-16 14:46:09: MQTT publish: topic 'homeassistant/sensor/0x00158d00045ab3d6/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d00045ab3d6"],"manufacturer":"Xiaomi","model":"Aqara curtain motor (ZNCLDJ11LM)","name":"0x00158d00045ab3d6","sw_version":"Zigbee2MQTT 1.18.1"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00158d00045ab3d6","name":"0x00158d00045ab3d6 linkquality","state_topic":"zigbee2mqtt/0x00158d00045ab3d6","unique_id":"0x00158d00045ab3d6_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
```

1.18.3:
```
luna@luna:~/docker-compose/zigbee2mqtt$ docker logs -f zigbee2mqtt --tail 500|grep 'zigbee2mqtt/0x00158d00045ab3d6'
Zigbee2MQTT:info  2021-05-16 14:46:32: MQTT publish: topic 'zigbee2mqtt/0x00158d00045ab3d6', payload '{"linkquality":null,"options":{"hand_open":true,"reverse_direction":true},"position":100,"running":false,"state":null,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-05-16 14:46:33: MQTT publish: topic 'homeassistant/cover/0x00158d00045ab3d6/cover/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x00158d00045ab3d6/set","device":{"identifiers":["zigbee2mqtt_0x00158d00045ab3d6"],"manufacturer":"Xiaomi","model":"Aqara curtain motor (ZNCLDJ11LM)","name":"0x00158d00045ab3d6","sw_version":"Zigbee2MQTT 1.18.3"},"json_attributes_topic":"zigbee2mqtt/0x00158d00045ab3d6","name":"0x00158d00045ab3d6","position_template":"{{ value_json.position }}","position_topic":"zigbee2mqtt/0x00158d00045ab3d6","set_position_template":"{ \"position\": {{ position }} }","set_position_topic":"zigbee2mqtt/0x00158d00045ab3d6/set","unique_id":"0x00158d00045ab3d6_cover_zigbee2mqtt"}'
Zigbee2MQTT:info  2021-05-16 14:46:33: MQTT publish: topic 'homeassistant/sensor/0x00158d00045ab3d6/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d00045ab3d6"],"manufacturer":"Xiaomi","model":"Aqara curtain motor (ZNCLDJ11LM)","name":"0x00158d00045ab3d6","sw_version":"Zigbee2MQTT 1.18.3"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00158d00045ab3d6","name":"0x00158d00045ab3d6 linkquality","state_topic":"zigbee2mqtt/0x00158d00045ab3d6","unique_id":"0x00158d00045ab3d6_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2021-05-16 14:46:33: MQTT publish: topic 'homeassistant/sensor/0x00158d00045ab3d6/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d00045ab3d6"],"manufacturer":"Xiaomi","model":"Aqara curtain motor (ZNCLDJ11LM)","name":"0x00158d00045ab3d6","sw_version":"Zigbee2MQTT 1.18.3"},"icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/0x00158d00045ab3d6","name":"0x00158d00045ab3d6 update state","state_topic":"zigbee2mqtt/0x00158d00045ab3d6","unique_id":"0x00158d00045ab3d6_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Zigbee2MQTT:info  2021-05-16 14:46:33: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d00045ab3d6/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d00045ab3d6"],"manufacturer":"Xiaomi","model":"Aqara curtain motor (ZNCLDJ11LM)","name":"0x00158d00045ab3d6","sw_version":"Zigbee2MQTT 1.18.3"},"json_attributes_topic":"zigbee2mqtt/0x00158d00045ab3d6","name":"0x00158d00045ab3d6 update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x00158d00045ab3d6","unique_id":"0x00158d00045ab3d6_update_available_zigbee2mqtt","value_template":"{{ value_json.update_available}}"}'
```

Maybe it has something to do that `linkquality` and `state` is `null` in >=1.18.2?

[image]

gimlichael, 2021-06-22:
Fixed with 1.19.1 (maybe 1.19 also).

closed 2021-06-22 by gimlichael