Koenkk/zigbee2mqtt #7140: Problem with MQTT "Attribute" Mode, and Ikea Tradfri Brightness and Color Transitions

patmalcolm91, 2021-04-20:
## What happened
I have enabled MQTT "attribute" mode, and whenever I send a brightness command shortly followed by a color temperature command (i.e. within the bulb's configured transition time (0.5s)) to one of my Ikea Tradfri LED1732G11 bulbs, the brightness transition is interrupted by the color temperature command and does not complete. Interestingly, zigbee2mqtt still reports the new desired brightness, despite the bulb itself not matching this brightness.

This appears to be related to #1810, with the difference being that I am using MQTT "attribute" mode.

## What did you expect to happen
When sending brightness and color temperature commands in short succession, both should always be considered.

## How to reproduce it (minimal and precise)
* Enable MQTT "attribute" mode
* Set a bulb's transition time to a large value (e.g. 1 second)
* Publish a brightness command to the bulb
* Within the configured transition time, publish a color temperature command to the bulb

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC2531
Adapter firmware version: 20201127

Koenkk, 2021-04-20:
This is a known bug of the TRADFRI bulbs. The fix is to send both at the same time, but the transition will only be applied to the color temperature.

closed 2021-04-20 by Koenkk

patmalcolm91, 2021-04-20:
How can both attributes be sent at the same time when using MQTT "Attribute" mode? There will always be some small time between the messages. Is there a way to configure zigbee2mqtt to group attribute messages together if they are sent close together?

Koenkk, 2021-04-20:
Even when in attribute mode, you can send payloads like this; topic: `zigbee2mqtt/friendly_name/set` payload `{"brightness": 100, "color_temp": 300}`.

patmalcolm91, 2021-04-20:
> Even when in attribute mode, you can send payloads like this; topic: `zigbee2mqtt/friendly_name/set` payload `{"brightness": 100, "color_temp": 300}`.

Ok, great, in that case I can implement a workaround in my current setup. Thanks for your help!