Koenkk/zigbee2mqtt #7150: Ikea Shortcut Button (E1812) Stops Working After Pairing

idofr, 2021-04-20:
## What happened
I've got two Ikea Shortcut Buttons and a Hue Dimmer Switch, all brand new. All three devices were easily added to Zigbee2MQTT without any problems. The pairing was quick and they're all recognised as the devices they really are, link quality and battery status is also showing.
Yet, I can't seem to get any response from the Ikea buttons after pairing. Using Mqtt Explorer and a very simple NodeRed scheme to show all communications in the network, I only see events coming from the Hue Switch, both Ikea Buttons don't seem to be generating any sort of messages. Sometimes, when I restart Zigbee2MQTT (used as the HomeAssistant plugin), it then show the last message in the queue, but that's it, it doesn't show anything afterwards.
As I said, the Hue Switch works perfectly fine and I see all messages it's generating.

Would very much appreciate some assistance here.

Cheers

## What did you expect to happen
I'd expect to see the messages from both Ikea buttons just the way I see the messages from the Hue switch, i.e. both in MQTT Explorer and in NodeRed.

## How to reproduce it (minimal and precise)
I did nothing really but pairing all three devices. No other devices are connected/paired.

## Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: CC2531
Adapter firmware version: now sure where to find this info.

mpmario, 2021-04-25:
For me exactly the same problem, with 
- IKEA Shortcut Button (IKEA E1812) and
- IKEA Remote Control (E1810)

Zigbee2MQTT version: 1.18.2
Adapter hardware: CC2652RB
Adapter firmware version: revision: 20210120

idofr, 2021-04-26:
In [this](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3495) thread @ebaauw said on 22.12.2020: "It's one of those devices that goes to sleep pretty quickly after pairing, and won't wake up, unless you created a binding within that time."

This might be what we're seeing here.... Not sure how to continue from here though, or even validate that that's indeed the matter

mpmario, 2021-04-26:
@idofr Did you also noticed this error in the log, when adding the device?

Zigbee2MQTT:error 2021-04-25 21:59:28: Failed to configure '0x588e81fffe2b8222', attempt 1 (Error: Bind 0x588e81fffe2b8222/1 genPowerCfg from '901' failed (AREQ - ZDO - bindRsp after 10000ms)
     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:67:35)
     at listOnTimeout (internal/timers.js:554:17)
     at processTimers (internal/timers.js:497:7))

idofr, 2021-04-26:
Pairing looks fine:
What you see here is removing the device, enabling pairing mode, resetting the button and paiting again
```
Warning Device '0x588e81fffe2b4513' left the network
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"ieee_address":"0x588e81fffe2b4513"},"type":"device_leave"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x588e81fffe2b4513"},"type":"device_removed"}'
Info Device '0x588e81fffe2b4513' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x588e81fffe2b4513","ieee_address":"0x588e81fffe2b4513"},"type":"device_joined"}'
Info Starting interview of '0x588e81fffe2b4513'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x588e81fffe2b4513","ieee_address":"0x588e81fffe2b4513","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x588e81fffe2b4513"},"type":"device_connected"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x588e81fffe2b4513"},"type":"pairing"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x588e81fffe2b4513","ieee_address":"0x588e81fffe2b4513"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x588e81fffe2b4513"},"type":"device_announced"}'
Info Successfully interviewed '0x588e81fffe2b4513', device has successfully been paired
Info Device '0x588e81fffe2b4513' is supported, identified as: IKEA TRADFRI shortcut button (E1812)
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"TRADFRI shortcut button","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["on","brightness_move_up","brightness_stop"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E1812","supports_ota":true,"vendor":"IKEA"},"friendly_name":"0x588e81fffe2b4513","ieee_address":"0x588e81fffe2b4513","status":"successful","supported":true},"type":"device_interview"}'
Info MQTT publish: topic 'homeassistant/sensor/0x588e81fffe2b4513/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x588e81fffe2b4513"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x588e81fffe2b4513","sw_version":"Zigbee2MQTT 1.18.1"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0x588e81fffe2b4513","name":"0x588e81fffe2b4513 battery","state_topic":"zigbee2mqtt/0x588e81fffe2b4513","unique_id":"0x588e81fffe2b4513_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x588e81fffe2b4513/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x588e81fffe2b4513"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x588e81fffe2b4513","sw_version":"Zigbee2MQTT 1.18.1"},"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x588e81fffe2b4513","name":"0x588e81fffe2b4513 action","state_topic":"zigbee2mqtt/0x588e81fffe2b4513","unique_id":"0x588e81fffe2b4513_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x588e81fffe2b4513/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x588e81fffe2b4513"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x588e81fffe2b4513","sw_version":"Zigbee2MQTT 1.18.1"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x588e81fffe2b4513","name":"0x588e81fffe2b4513 linkquality","state_topic":"zigbee2mqtt/0x588e81fffe2b4513","unique_id":"0x588e81fffe2b4513_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x588e81fffe2b4513/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x588e81fffe2b4513"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x588e81fffe2b4513","sw_version":"Zigbee2MQTT 1.18.1"},"icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/0x588e81fffe2b4513","name":"0x588e81fffe2b4513 update state","state_topic":"zigbee2mqtt/0x588e81fffe2b4513","unique_id":"0x588e81fffe2b4513_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Info MQTT publish: topic 'homeassistant/binary_sensor/0x588e81fffe2b4513/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x588e81fffe2b4513"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x588e81fffe2b4513","sw_version":"Zigbee2MQTT 1.18.1"},"json_attributes_topic":"zigbee2mqtt/0x588e81fffe2b4513","name":"0x588e81fffe2b4513 update available","payload_off":false,"pay [... cut ...]
Info MQTT publish: topic 'zigbee2mqtt/0x588e81fffe2b4513', payload '{"battery":47,"linkquality":47}'
Info MQTT publish: topic 'zigbee2mqtt/0x588e81fffe2b4513', payload '{"battery":47,"linkquality":47,"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/0x588e81fffe2b4513', payload '{"battery":47,"linkquality":47,"update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/0x588e81fffe2b4513', payload '{"battery":47,"linkquality":47,"update":{"state":"idle"},"update_available":false}'
Info Successfully configured '0x588e81fffe2b4513'
```

lkxnon, 2021-05-15:
Have same issue here. Running latest version of zigbee2mqtt + zzh! stick 20210120 firmware zStack3x0. Pairing seems to be successful, no errors. Buttons are working fine, sending mqtt output in debug log is visible, integration with Node-Red or HA works fine. And next day none of them works. It is paired, visible in devices list but action (toggle or up/down) does not make any action, nothing in debug log.

I have E1524/E1810 remote and E1812 button, new batteries. Very strange behaviour.

Tonio16, 2021-05-17:
Hello

I paired two E1810 recently. Both of then are seen as unsupported. The first one that I paired was recognized as E1810 after 2-3 days. For the second one I am still waiting.

My conf:
Zigbee2MQTT version 1.18.3 commit: f2e39af1
Coordinator type EZSP v8
Coordinator revision 6.9.1.0 build 214
Frontend version 0.4.0

Antoine

idofr, 2021-06-17:
Solved by switching to https://slae.sh/projects/cc2652/

silverl, 2021-07-02:
I'm having this same problem with:

* ConBee II USB
* zigbee2mqtt 1.20.0-1
* Ikea Tradfri E1812
* Ikea SYMFONISK E1744

Just bought the 2 Ikea devices to play around with. They paired fine, didn't notice any issues pairing.

Looks like zigbee2mqtt is seeing battery levels from the devices.

But I'm seeing zero activity in MQTT when using the buttons.

wiuwiu, 2021-07-13:
I have the same situation with Ikea Shortcut Button E1812. This device worked perfectly but after OTA update and restart zigbee2mqtt device don't publish his state/battery/link. 

    Adapter hardware: CC2652RB from slae.sh (firmware: 20210120)
    zigbee2mqtt 1.20.0-1
    Ikea Shortcut Button E1812

MattWestb, 2021-07-14:
@wiuwiu You have using IKEA test feed for OTA and the latest controller firmware is very "test" its not zigbee citified and have problems.

Its not possible doing firmware downgrading then its blocked in IKEA controllers firmware so its only waiting until IKEA is relenting one working firmware in the production feed.

JakubTHOC, 2021-07-14:
Same problem here

tricoos, 2021-07-21:
I don't have that problem, but I have obviously an older firmware version and the OTA check fails which is weird - but the button works:

**Button E1812**
FW date: 20190715
FW version 2.3.015

**Zigbee2MQTT Version**
1.20.0 commit: 41b67fd

**Coordinator**
- CC2652P2
- zStack3x0
- 20210317

**OTA Check Result**
` MQTT publish: topic 'zigbee/bridge/log', payload '{"message":"Failed to check if update available for 'tradfributton1' (Device didn't respond to OTA request)","meta":{"device":"tradfributton1","status":"check_failed"},"type":"ota_update"}'
Info MQTT publish: topic 'zigbee/bridge/response/device/ota_update/check', payload '{"data":{"id":"tradfributton1"},"error":"Failed to check if update available for 'tradfributton1' (Device didn't respond to OTA request)","status":"error","transaction":"520v7-1"}'
Error Failed to check if update available for 'tradfributton1' (Device didn't respond to OTA request)`

Maybe this helps for a cross-check.

tricoos, 2021-07-21:
@willu 
> ```
> Ikea Shortcut Button E1812 (firmware: 2.3.075)
> ```

Weird, my second button with the same firmware version just was logged as no update available:
` MQTT publish: topic 'zigbee/tradfributton2', payload '{"battery":100,"linkquality":30,"update":{"state":"idle"},"update_available":false}'`

And that's with version 2.3.0.15 and you have 2.3.075 so there should be an update available.
Maybe it helps if everyone with the problem here posts his button firmware version?

froogl, 2021-08-17:
One of the issues with this button is that it does not switch it's parent once paired (just like the Xiaomi devices). At least that is my feeling. So you should not move the button to a new location after pairing (do that before pairing).

JHthe4, 2021-09-09:
I'm having a similar issue. After the button failed to send messages, I upgraded zigbee2mqtt, force removed it (regular remove was unresponsive) and paired it again successfully. I also had to manually reconfigure since the first configuration failed. Now it is paired, but all it does is regularly announce itself. No Zigbee messages whatsoever when the button is actually pressed. 

```info  2021-09-09 17:09:11: Device '0xb4e3f9fffe5a9fa6' joined
info  2021-09-09 17:09:11: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffe5a9fa6","ieee_address":"0xb4e3f9fffe5a9fa6"},"type":"device_joined"}'
info  2021-09-09 17:09:11: Configuring '0xb4e3f9fffe5a9fa6'
info  2021-09-09 17:09:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xb4e3f9fffe5a9fa6"},"type":"device_connected"}'
debug 2021-09-09 17:09:13: Device '0xb4e3f9fffe5a9fa6' announced itself
info  2021-09-09 17:09:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffe5a9fa6","ieee_address":"0xb4e3f9fffe5a9fa6"},"type":"device_announce"}'
info  2021-09-09 17:09:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xb4e3f9fffe5a9fa6"},"type":"device_announced"}'
error 2021-09-09 17:09:21: Failed to configure '0xb4e3f9fffe5a9fa6', attempt 1 (Error: Bind 0xb4e3f9fffe5a9fa6/1 genPowerCfg from '901' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
debug 2021-09-09 17:11:03: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '0xb4e3f9fffe5a9fa6'
info  2021-09-09 17:11:03: Configuring '0xb4e3f9fffe5a9fa6'
debug 2021-09-09 17:11:03: Received Zigbee message from '0xb4e3f9fffe5a9fa6', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587290161,"imageType":4550,"manufacturerCode":4476}' from endpoint 1 with groupID 0
debug 2021-09-09 17:11:03: Device '0xb4e3f9fffe5a9fa6' requested OTA
debug 2021-09-09 17:11:03: Check if update available for '0xb4e3f9fffe5a9fa6' (TRADFRI SHORTCUT Button)
debug 2021-09-09 17:11:04: Is new image available for '0xb4e3f9fffe5a9fa6', current '{"fieldControl":1,"manufacturerCode":4476,"imageType":4550,"fileVersion":587290161}', latest meta '{"fileVersion":587290161,"url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.15.055/bin/10054470-TRADFRI-shortcut-button-2.3.015.ota.ota.signed"}'
debug 2021-09-09 17:11:04: Update available for '0xb4e3f9fffe5a9fa6': NO
info  2021-09-09 17:11:04: MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffe5a9fa6', payload '{"linkquality":81,"update":{"state":"idle"},"update_available":false}'
debug 2021-09-09 17:11:07: Received Zigbee message from '0xb4e3f9fffe5a9fa6', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":74}' from endpoint 1 with groupID 0
info  2021-09-09 17:11:07: MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffe5a9fa6', payload '{"battery":74,"linkquality":81,"update":{"state":"idle"},"update_available":false}'
info  2021-09-09 17:11:07: Successfully configured '0xb4e3f9fffe5a9fa6'
debug 2021-09-09 17:30:02: Device '0xb4e3f9fffe5a9fa6' announced itself
info  2021-09-09 17:30:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffe5a9fa6","ieee_address":"0xb4e3f9fffe5a9fa6"},"type":"device_announce"}'
info  2021-09-09 17:30:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xb4e3f9fffe5a9fa6"},"type":"device_announced"}'
debug 2021-09-09 17:32:23: Device '0xb4e3f9fffe5a9fa6' announced itself
info  2021-09-09 17:32:23: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffe5a9fa6","ieee_address":"0xb4e3f9fffe5a9fa6"},"type":"device_announce"}'
info  2021-09-09 17:32:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xb4e3f9fffe5a9fa6"},"type":"device_announced"}'
```

McMornan, 2021-09-11:
Hi, I had the same issues with my ikea dimmer switches and buttons. Also using z2m 1.21.1 and cc2652rb. For me persistence was the key to success. I paired every button several Times (4x click to reset). Before it finally worked I also removed the buttons once. Also continously clicking on the button (not the reset switch) seems to have helped. (to prevent it falling asleep) After a few tries I could see the action messages coming in. Final configuration failed however and I paired again, clicking and clicking.... After 10-15 tries it finally worked. But I admit that pita process is not as it should be 😁

voltagex, 2021-09-15:
Having this issue with a Symfonisk, although I see the following

```
Sep 16 02:34:51 boxodisks npm[3694108]: Zigbee2MQTT:error 2021-09-16 02:34:51: Failed to configure '0x5c0272fffeac3850', attempt 1 (Error: Bind 0x5c0272fffeac3850/1 genPowerCfg from '0x00124b0001575f2a/1' failed (waiting for response TIMEOUT)
```
How can I help debug this?

```
Sep 16 02:42:09 boxodisks npm[3694331]: Zigbee2MQTT:info  2021-09-16 02:42:09: Starting Zigbee2MQTT version 1.21.1 (commit #4a51e0c0)
Sep 16 02:42:09 boxodisks npm[3694331]: Zigbee2MQTT:info  2021-09-16 02:42:09: Starting zigbee-herdsman (0.13.138)
Sep 16 02:42:10 boxodisks npm[3694331]: CREATED DECONZ ADAPTER
Sep 16 02:42:10 boxodisks npm[3694331]: Zigbee2MQTT:info  2021-09-16 02:42:10: zigbee-herdsman started (resumed)
Sep 16 02:42:10 boxodisks npm[3694331]: Zigbee2MQTT:info  2021-09-16 02:42:10: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":112,"product":0,"revision":"0x26700700","transportrev":0},"type":"ConBee2/RaspBee2"}'
```

cpuks, 2021-09-24:
I've got this issue on E1812 - firmware 2.3.075 and CC2652 - pairing is not a problem but then I cannot get anything out of this shortcut. Is it a matter of waiting for new firmware?

ristomatti, 2021-09-25:
> After 10-15 tries it finally worked. But I admit that pita process is not as it should be 😁

I had the same experience as @McMornan described. I bought four E1821 (fw 2.3.015) Shortcut buttons and two E1744 (fw 2.1.024) Symphony dimmers. Initially I tried pairing them with a CC2531 coordinator. 1-2 paired right away but after that didn't got much success.

I then swapped the coordinator to a CC2652R (zzh!) I had already purchased in anticipation of potential issues and upgraded Z2M. This didn't seem to help. I got the devices paired several times but within hours each stopped working. I tried pairing them both to a Trådfree extender/router and to "any device". 

After persistently trying to pair them several times during 2-3 days, suddenly I got all but one Shortcut button paired and they've now worked reliably for 3-4 days in a row, even two which I moved to another room. I remember pairing them multiple times to a BlitzWolf BW-SHP13 socket at that time.

Another noteworthy thing was that 4 of the 6 devices reported 30-70% battery state right out the package and I had to replace them. The actual voltages of the replaced IKEA branded batteries were between 2.5-3.05V.

I'm currently running Zigbee2MQTT v1.21.1 with zStack3x0 rev. 20210708.

thijslimmen, 2021-09-27:
Experiencing the same issues. devices pair successfully, but no press events are received..

MattWestb, 2021-10-09:
@ristomatti what coordinator is you using ?
I think its problems with that the coordinator is accepting that Zigbee 3 devices like updating there trust center link key and is leaving the network because of that.

ristomatti, 2021-10-09:
> @ristomatti what coordinator is you using ? 

Check couple of messages back.

AlesZurek, 2021-10-31:
Hi guys, I wanted to share my update.
As I wrote before I have this issue that I got no action from ikea shortcut button after pairing.

I had coordinator Sonoff CC2351 in my Rapsberry pi 4.
More info
zigbee2mqtt_version
1.21.0 commit: 70891ee
coordinator_type
zStack12
coordinator_revision
20180507
frontend_version
0.5.18

Currently I ordered Sonoff CC2652P coordinator
https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus

Today I received it so I tried it and on first try I was able to add my IKEA shortcut button and it is working.
Sadly I wasn't able to add my second IKEA shortcut button but maybe it can be malfunctional. Not sure about that but the other one is working fine.
I was also able to get working another piece of hardware which wasn't working (no actions received) TuYa TS004F.

That new coordinator have values:
Zigbee2MQTT version
1.21.0 commit: 70891ee
Coordinator type
zStack3x0
Coordinator revision
20210120
Frontend version
0.5.18

So in my case I just need to change my coordinator to get it work.

McMornan, 2021-10-31:
Hi,

thanks for sharing. However in my case its not the pairing - that can be done in the vicinity of the coordinator and retrying several times - it's the STAYING PAIRED which ist the problem for me. So would be great if you could give us an update in a few days if the button is still recognized.

Lars

X52p, 2021-11-01:
> I have the same problem. It worked fine for about a year. Then I upgraded my USB dongle to a CC2652 (and also updated zigbee2mqtt) Some hours later the first of my two shortcut buttons stopped working, and today (about 24h later) the second one also gave up.
> 
> So (I think) it could be changes in Zigbe2mqtt or a problem with the firmware of the dongle.
> 
> If I can provide anything useful (like logs or something) please tell me.

I updated my CC2652 to the current firmware version (20210708) and re-paired the buttons. I also updated the firmware of the buttons to the newest version (OTA update)

Still, after a few hours, they disconnect again.

Theoreet, 2021-11-01:
i'am having some kind of the same issue.
Playing aroud with the E1812, running the 2.3.080/20210727 firmware. The Z2M i tried 1.22.0 and latest Edge version. The USB 2652P runs 20210319 firmware version. Frontend version 0.6.30
Pairing seems to be working after several tries, battery seems to be reporting, but the action stays "null". the other remotes reports okay ( this is the only shortcut button i own)

belkone, 2021-11-02:
@idofr try this: https://github.com/Koenkk/zigbee2mqtt/issues/9439#issuecomment-956456214

alexwala, 2021-11-02:
> i'am having some kind of the same issue. Playing aroud with the E1812, running the 2.3.080/20210727 firmware. The Z2M i tried 1.22.0 and latest Edge version. The USB 2652P runs 20210319 firmware [...]

how did you updated to 2.3.080 (using z2m ota) ? may you can help on https://github.com/Koenkk/zigbee2mqtt/issues/9457

EricV-hue, 2021-11-02:
Just written my experience here:
https://github.com/Koenkk/zigbee2mqtt/issues/9457#issuecomment-957760527

alexwala, 2021-11-03:
finally it is working: using https://github.com/Koenkk/zigbee2mqtt/issues/9457#issuecomment-957760527, multiple pairings and reboots :) a good tip: the device should be close to the coordinator during pairing, otherwise there may be errors
thanks all !

MaciejR89, 2021-11-03:
Binding "LevelCtrl" and "OnOff" to coordinator fix the problem

```
Zigbee2MQTT:debug  Received MQTT message on 'zigbee2mqtt/bridge/request/device/bind' with data '{"clusters":["genOnOff","genLevelCtrl"],"from":ikea_shortcut_button/1","to":"Coordinator","transaction":"y1xu9-4"}'
Zigbee2MQTT:debug  binding cluster 'genOnOff' from 'ikea_shortcut_button' to 'Coordinator'
Zigbee2MQTT:debug binding cluster 'genLevelCtrl' from 'ikea_shortcut_button' to 'Coordinator'
Zigbee2MQTT:info  Successfully bound cluster 'genLevelCtrl' from 'ikea_shortcut_button' to 'Coordinator'
Zigbee2MQTT:info  Successfully bound cluster 'genOnOff' from 'ikea_shortcut_button' to 'Coordinator'
Zigbee2MQTT:info  MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genLevelCtrl","from":"ikea_shortcut_button","to":"Coordinator"},"type":"device_bind"}'
Zigbee2MQTT:debug Received Zigbee message from 'ikea_shortcut_button', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  MQTT publish: topic 'zigbee2mqtt/ikea_shortcut_button', payload '{"action":"on","battery":60,"linkquality":48,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  MQTT publish: topic 'zigbee2mqtt/ikea_shortcut_button/action', payload 'on'
```

tricoos, 2021-11-03:
@MaciejR89 After doing this it wreaks havoc for me (I have a toggle action onClick) as it fires three times now:
```
MQTT publish: topic 'zigbee/tradfributton1', payload '{"action":"on","battery":100,"linkquality":45,"update":{"state":"available"},"update_available":true}'
Info MQTT publish: topic 'zigbee/tradfributton1', payload '{"action":"on","battery":100,"linkquality":45,"update":{"state":"available"},"update_available":true}'
Info MQTT publish: topic 'zigbee/tradfributton1', payload '{"action":"on","battery":100,"linkquality":45,"update":{"state":"available"},"update_available":true}'
```

tricoos, 2021-11-03:
I guess I will throw that thing in the bin and stick a wonderful Aqara button on the wall. Luckily I still have one unused left.

FlatulentCow, 2021-11-07:
Having exactly the same issue.  I have two brand new E 1812 buttons (delivered yesterday) that were working without issue then I saw there was an update in the OTA section.  I updated them both to 2.3.080 and now I cannot get any actions to register.  Battery is being reported but even with fresh batteries they still show low power.  I've tried all the workarounds suggested that I can find and nothing so far has worked.

jonas740, 2021-11-07:
I don't really know if I have succeeded in fixing this issue, don't want to celebrate just yet.

But what I have done so far.

Took my CC2652RB apart and inspected it with microscope and found a solder point so look a bit odd so I resoldered that one. (can't really say if it was a problem but it just did not look "neat")

Reflashed it with latest firmware.

I bought a new USB 2.0 extension cable(Amazon Basic 3 meter) and I placed my CC2652RB as far away from any electronics I could. 

After this it has magically worked.

I had a USB 2.0 extension cable prior to buying a new one but I wanted to take every step I could to make sure that I have removed all the variables that might cause the issue.

Regarding battery status for each device, I believe that to be very inaccurate and not to be trusted.

MattWestb, 2021-11-07:
Sounds great Jonas !!

Sleeping end device is using max 1/10 of the chip max power and most of the time they is using less then 1/100 of the chop max power so if its noisy around they is getting problem sending commands and getting replay back then its parent can here them.

One good thing its having routers around the sleeping end device so they is not needing using "high power" and still can communicating OK.

IKEA implementation of battery status is . . . 
Its reporting OK but always the same until you is using one button then its updating the is sending the "new" battery status and is reporting it until . . . 
And its very "jumpy" !!!

unsounddabbit, 2021-11-17:
I just bought 6 of these guys and upgraded the firmware to all of them.  All worked before the FW update.  Is there a way to roll back firmware?

MattWestb, 2021-11-17:
I have unpacking the last good working firmware for all updated controller that have loosing group binding or is locked in bad firmware.
You need one SWD probe for flashing. https://github.com/zigpy/zigpy/discussions/660#discussioncomment-1565541
And for the E1812 you can do it without distorting the device https://github.com/zigpy/zigpy/discussions/660#discussioncomment-1563280.

If you like keep it on the last good or updating it later if its coming  on good working version.

FlatulentCow, 2021-11-17:
Group bindings is indeed what caused my issues.  They were binding to a default group 901 which didn't actually exist.  I struggled for 3 days trying to bind them to the coordinator.  Must have deleted them and re-added them 30 times.  With a combination of clicking the button when binding whilst simultaneously kicking off an OTA check and pushing the bind to coordinator command, I finally got them bound to the coordinator.  They work now but in hindsight I would return them, get replacements and prevent the auto OTA updates.

unsounddabbit, 2021-11-17:
Mat, Thanks for that, but I'm not skilled enough to pull off your fix. I tried to work with the bindings and nothing worked.  I'm going to return them.  it's such a shame.  I see this thread was created in April.  I'm guessing there won't be a firmware update that fixes this any time soon.

MattWestb, 2021-11-17:
IKEA was having one test feed for firmware (they have closing it and using one that we dont knowing now) and it was some bad beta firmware there that is not possible updating or roiling back therefor i was making the flashing as one last resort for user with real broken not upgrading devices.

If enough users is returning then is IKEA getting one large red light in there system and perhaps somthing is happening but all controllers is very tricky for them redoing then its making all magic in there system.

And in the pipe is blinds and signal repeater and the old CWS 1.0 was certified yesterday (the last ZLL that is getting Zigbee 3) and i think they is rolling update on all lights for fixing some stack bugs but its only speculations from my side.

I think update fixing group binding i think is possible but it was taking very long time for this release after the bad one so i thing its not weeks but mounts before it can being released if they is doing it.

unsounddabbit, 2021-11-17:
Matt,

That's what I was thinking.  They will need to fix this and me doing their job won't make that happen.  I've heard that IKEA firmware can be hit or miss and I've been slow to use their stuff.  Phillips is expensive, but it works.

eljanoc, 2021-11-21:
Same problem with the E1812. Binding is OK, but configuring does not work.
After a lot of tries, I succeeded once in configuring the module, but I did not know exactly how ... So, it answered correctly with short and long click response !
I then deleted the item, to see, and start the association again ..., never managed to correctly configure the module.
I work with a conbee2

chasebolt, 2021-11-29:
I updated from 2.3.015 to 2.3.080 and the button has stopped responding when pressed. zigbee2mqtt sees it come online when i pull the battery and reinsert it.

redhead-iBoy, 2021-11-29:
> I updated from 2.3.015 to 2.3.080 and the button has stopped responding when pressed. zigbee2mqtt sees it come online when i pull the battery and reinsert it.
@chasebolt 
Same problem here. Alle buttons worked fine till the update. Tried different setup, HA instead of OH and a different coordinator. The 2.3.080 firmware just doesn´t seem to work correctly with z2m. 
Pairing was smooth like allways. The shortcut button exposes the battery lvl and the link quality just fine. 
Only the actions aren´t recognized by z2m.

eloo, 2021-11-29:
@chasebolt @Lars383 

the problem has already a workaround which is already mentioned in this issue some comments above.. 
https://github.com/Koenkk/zigbee2mqtt/issues/9439#issuecomment-956456214

you should try to bind the device during the interview to the coordinator
at least this is working for all of my 4 E1812

Hope that helps

Lars383, 2021-11-29:
@eloo Thanks a lot eloo, you saved my day :-) 
Danke Joseph :-)

GrumpyMeow, 2021-12-04:
I regret i updated the firrmware of one my E1524/E1810 remotes to v2.3.080. This as i'm now unable to bind this remote to a group of lights. In contrary to others in this thread, everything else seems to be working alright.
`1.22.0, zStack3x0, 20210120, 0.6.30, SonfOff Zigbee Dongle Plus`

reneklootwijk, 2021-12-05:
> I regret i updated the firrmware of one my E1524/E1810 remotes to v2.3.080. This as i'm now unable to bind this remote to a group of lights. In contrary to others in this thread, everything else seems to be working alright with the remote. I do see actions being triggered. The reason i want this remote to be bound to a group, is because i want two remotes to control the lights. `1.22.0, zStack3x0, 20210120, 0.6.30, SonfOff Zigbee Dongle Plus`

I think firmware 2.3.080 does support binding to a group anymore, you have to bind to the individual devices. At least this worked for my E1524 after upgrading.

reneklootwijk, 2021-12-05:
> Group bindings is indeed what caused my issues. They were binding to a default group 901 (after the OTA update to 2.3.080) which didn't actually exist. I struggled for 3 days trying to bind them to the coordinator. Must have deleted them and re-added them 30 times. With a combination of clicking the button when binding whilst simultaneously kicking off an OTA check and pushing the bind to coordinator command, I finally got them bound to the coordinator. They work now but in hindsight I would return them, get replacements and prevent the auto OTA updates.

For me this also worked, I removed the device, rejoined and during the interview process send the mqtt message to bind to the coordinator. I also updated the zigbee-herdsman-converters version to support the off command by double clicking the switch.

Koenkk, 2021-12-07:
Pushed a fix to correctly configure these remotes after the latest OTA update.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

lbt, 2021-12-20:
I had these problems with a Raspbee (deconz) under zigbee2mqtt
I switched to the dev branch (HEAD is 42569d6c39295b6bcbde788a2db94e0262b434e8) and still don't get events from the E1812

McMornan, 2021-12-20:
The fix will not affect devices which already have the non functional firmware, only ones which are upgraded by zigbee2mqtt with the lates dev branch. You need to manually bind the button to the coordinator after successful pairing. It works. I paired two shortcut buttons just yesterday and they work as intended (only send an "on" message though, which is enough to toggle)

example command on linux command line:
mosquitto_pub -h 192.168.2.5 -p 1883 -t zigbee2mqtt/bridge/request/device/bind -m '{"from":"IKEAButton5","to":"coordinator"}'

lbt, 2021-12-20:
I'm afraid I'm having problems with this:

Here you see the mqtt traffic including my initial request. I tried "coordinator" with lowercase as you said but noted it was in uppercase in the UI so I tried that too.
```
zigbee2mqtt/bridge/request/device/bind = b'{"from":"PushButton","to":"Coordinator"}'        2 {'dup': 0, 'retain': 0}
zigbee2mqtt/bridge/logging = b'{"level":"error","message":"Failed to bind cluster \'genLevelCtrl\' from \'PushButton\' to \'Coordinator\' (Error: Bind 0x0c4314fffe4c09fe/1 genLevelCtrl from \'0x00212effff022d34/1\' failed (waiting for response TIMEOUT))"}'        0 {'dup': 0, 'retain': 0}
zigbee2mqtt/bridge/logging = b'{"level":"info","message":"MQTT publish: topic \'zigbee2mqtt/bridge/log\', payload \'{\\"message\\":{\\"cluster\\":\\"genLevelCtrl\\",\\"from\\":\\"PushButton\\",\\"to\\":\\"Coordinator\\"},\\"type\\":\\"device_bind_failed\\"}\'"}'        0 {'dup': 0, 'retain': 0}
zigbee2mqtt/bridge/log = b'{"message":{"cluster":"genLevelCtrl","from":"PushButton","to":"Coordinator"},"type":"device_bind_failed"}'        0 {'dup': 0, 'retain': 0}
```

The UI already shows a binding:

[image]

(I think I set the onOff)

Clearly the timeout suggests the lack of an RF connection so I tried clicking the button before doing this to get it awake but that didn't help.

I then force removed the device (remove failed) via the UI and reset the button and it re-paired. Here are the logs
```
Dec 20 21:25:52 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:25:52: Device '0x0c4314fffe4c09fe' joined
Dec 20 21:25:52 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:25:52: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe4c09fe","ieee_address":"0x0c4314fffe4c09fe"},"type":"device_joined"}'
Dec 20 21:25:52 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:25:52: Configuring '0x0c4314fffe4c09fe'
Dec 20 21:25:52 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:25:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0c4314fffe4c09fe"},"type":"device_connected"}'
Dec 20 21:25:52 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:25:52: Starting interview of '0x0c4314fffe4c09fe'
Dec 20 21:25:52 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:25:52: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe4c09fe","ieee_address":"0x0c4314fffe4c09fe","status":"started"},"type":"device_interview"}'
Dec 20 21:25:52 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:25:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0c4314fffe4c09fe"},"type":"pairing"}'
Dec 20 21:25:53 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:25:53: Successfully interviewed '0x0c4314fffe4c09fe', device has successfully been paired
Dec 20 21:25:53 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:25:53: Device '0x0c4314fffe4c09fe' is supported, identified as: IKEA TRADFRI shortcut button (E1812)
Dec 20 21:25:53 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:25:53: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"TRADFRI shortcut button","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["on","off","brightness_move_up","brightness_stop"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E1812","options":[{"description":"Simulate a brightness value. If this device provides a brightness_move_up or brightness_move_down action it is possible to specify the update interval and delta.","features":[{"access":2,"description":"Delta per interval, 20 by default","name":"delta","property":"delta","type":"numeric","value_min":0},{"access":2,"description":"Interval duration","name":"interval","property":"interval","type":"numeric","unit":"ms","value_min":0}],"name":"simulated_brightness","property":"simulated_brightness","type":"composite"}],"supports_ota":true,"vendor":"IKEA"},"friendly_name":"0x0c4314fffe4c09fe","ieee_address":"0x0c4314fffe4c09fe","status":"successful","supported":true},"type":"device_interview"}'
Dec 20 21:25:53 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:25:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"TRADFRI shortcut button","friendly_name":"0x0c4314fffe4c09fe","model":"E1812","supported":true,"vendor":"IKEA"},"type":"pairing"}'
Dec 20 21:26:15 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:26:15: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe4c09fe', payload ''
Dec 20 21:26:15 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:26:15: MQTT publish: topic 'zigbee2mqtt/PushButton/availability', payload 'online'
Dec 20 21:26:15 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:26:15: MQTT publish: topic 'zigbee2mqtt/PushButton', payload '{"linkquality":255}'
Dec 20 21:26:15 plum npm[26755]: Zigbee2MQTT:info  2021-12-20 21:26:15: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/rename', payload '{"data":{"from":"0x0c4314fffe4c09fe","homeassistant_rename":false,"to":"PushButton"},"status":"ok","transaction":"00bj4-9"}'
[... cut ...]
zigbee2mqtt/bridge/logging = b'{"level":"info","message":"MQTT publish: topic \'zigbee2mqtt/bridge/log\', payload \'{\\"message\\":{\\"cluster\\":\\"closuresWindowCovering\\",\\"from\\":\\"PushButton\\",\\"to\\":\\"Coordinator\\"},\\"type\\":\\"device_bind\\"}\'"}'        0 {'dup': 0, 'retain': 0}
zigbee2mqtt/bridge/log = b'{"message":{"cluster":"closuresWindowCovering","from":"PushButton","to":"Coordinator"},"type":"device_bind"}'        0 {'dup': 0, 'retain': 0}
```

Which resulted in this:
[image]

Let me know if there are any other tests/debug settings etc that I could do

lbt, 2022-01-20:
I returned to try and debug this and it's now working for me. I didn't do anything at all related to the device :)

Pushes on the button get:

```
info MQTT publish: topic 'zigbee2mqtt/PushButton', payload '{"action":"on","battery":16,"linkquality":255,"update":{"state":"idle"},"update_available":false}'
```
Holding gets:
```
info MQTT publish: topic 'zigbee2mqtt/PushButton', payload '{"action":"on","battery":16,"linkquality":255,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/PushButton', payload '{"action":"brightness_move_up","action_rate":83,"battery":16,"linkquality":255,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/PushButton', payload '{"action":"brightness_move_up","action_rate":83,"battery":16,"linkquality":255,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/PushButton', payload '{"action":"brightness_move_up","action_rate":83,"battery":16,"linkquality":255,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/PushButton', payload '{"action":"brightness_move_up","action_rate":83,"battery":16,"linkquality":255,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/PushButton', payload '{"action":"brightness_stop","battery":16,"linkquality":255,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/PushButton', payload '{"action":"brightness_stop","battery":16,"linkquality":255,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/PushButton', payload '{"action":"on","battery":16,"linkquality":255,"update":{"state":"idle"},"update_available":false}'
info
```
The Rotary produces

```
info MQTT publish: topic 'zigbee2mqtt/Rotary', payload '{"action":"play_pause","battery":60,"brightness":97,"linkquality":255,"rate":195,"update":{"state":"available"},"update_available":true}'
```

```
info MQTT publish: topic 'zigbee2mqtt/Rotary', payload '{"action":"rotate_left","battery":60,"brightness":1,"linkquality":239,"rate":195,"update":{"state":"available"},"update_available":true}'
info MQTT publish: topic 'zigbee2mqtt/Rotary', payload '{"action":"rotate_left","battery":60,"brightness":1,"linkquality":239,"rate":195,"update":{"state":"available"},"update_available":true}'
info MQTT publish: topic 'zigbee2mqtt/Rotary', payload '{"action":"rotate_stop","battery":60,"brightness":1,"linkquality":239,"rate":195,"update":{"state":"available"},"update_available":true}'
```

```
info MQTT publish: topic 'zigbee2mqtt/Rotary', payload '{"action":"rotate_right","battery":60,"brightness":1,"linkquality":199,"rate":195,"update":{"state":"available"},"update_available":true}'
info MQTT publish: topic 'zigbee2mqtt/Rotary', payload '{"action":"rotate_right","battery":60,"brightness":21,"linkquality":199,"rate":195,"update":{"state":"available"},"update_available":true}'
info MQTT publish: topic 'zigbee2mqtt/Rotary', payload '{"action":"rotate_stop","battery":60,"brightness":1,"linkquality":239,"rate":195,"update":{"state":"available"},"update_available":true}'
```

CWempe, 2022-01-20:
@idofr 

It would be very useful for other users if you would add a link to the workaround (https://github.com/Koenkk/zigbee2mqtt/issues/9439#issuecomment-956456214) to the initial post.

I read all of this long issue just to see there is an easy workaround available. 😄

closed 2022-01-28 by github-actions[bot]