Koenkk/zigbee2mqtt #7155: Tuya TS0601 One Phase switch

Diddlik, 2021-04-21:
## Information about the device + link
Zigbee Tuya Einphasig 65A Din Schiene Zigbee Smart Energy Meter timer Monitor kWh Meter Wattmeter 110V 220V 50/60Hz

https://de.aliexpress.com/item/1005002409588154.html?spm=a2g0s.9042311.0.0.61194c4dsTY832

## data/database.db entry of the device

FROM database:
`
{"id":60,"type":"Router","ieeeAddr":"0x60a423fffe0da3d1","nwkAddr":42659,"manufId":4098,"manufName":"_TZE200_byzdayie","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_byzdayie","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1618998828975}
`

FROM Log:
```

 '0x60a423fffe0da3d1', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":5}' from endpoint 1 with groupID 0
warn  2021-04-21 11:41:45: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_byzdayie'
warn  2021-04-21 11:41:45: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

Diddlik, 2021-04-22:
So, I have created an external converter and I would able to get a dump. See below. But at the moment I dont not know what to do the next.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const device = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_byzdayie'}],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Zigbee Tuya smart energy meter DDS238-2 Zigbee',
    fromZigbee: [
        //fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    meta: {
        configureKey: 1,
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
};
module.exports = device;

```

der Dump sieht so aus:
```

1619071976798 0x60a423fffe0da3d1 00 8f 13 02 00 00 00 00 00
1619071977790 0x60a423fffe0da3d1 00 8f 14 02 00 00 00 09 18
1619072004787 0x60a423fffe0da3d1 00 8f 01 01 00 01
1619072005799 0x60a423fffe0da3d1 00 8f 12 02 00 00 00 00 00
1619072006789 0x60a423fffe0da3d1 00 8f 13 02 00 00 00 00 00
1619072007789 0x60a423fffe0da3d1 00 8f 14 02 00 00 00 09 16
1619072034783 0x60a423fffe0da3d1 00 8f 01 01 00 01
1619072035806 0x60a423fffe0da3d1 00 8f 12 02 00 00 00 00 00
1619072036786 0x60a423fffe0da3d1 00 8f 13 02 00 00 00 00 00
1619072037790 0x60a423fffe0da3d1 00 8f 14 02 00 00 00 09 15
1619072064781 0x60a423fffe0da3d1 00 8f 01 01 00 01
1619072065787 0x60a423fffe0da3d1 00 8f 12 02 00 00 00 00 00
1619072066807 0x60a423fffe0da3d1 00 8f 13 02 00 00 00 00 00
1619072067787 0x60a423fffe0da3d1 00 8f 14 02 00 00 00 09 14
1619072094789 0x60a423fffe0da3d1 00 8f 01 01 00 01
1619072095794 0x60a423fffe0da3d1 00 8f 12 02 00 00 00 00 00
1619072096786 0x60a423fffe0da3d1 00 8f 13 02 00 00 00 00 00
1619072097791 0x60a423fffe0da3d1 00 8f 14 02 00 00 00 09 14
1619072124778 0x60a423fffe0da3d1 00 8f 01 01 00 01
1619072125806 0x60a423fffe0da3d1 00 8f 12 02 00 00 00 00 00
1619072126792 0x60a423fffe0da3d1 00 8f 13 02 00 00 00 00 00
1619072127785 0x60a423fffe0da3d1 00 8f 14 02 00 00 00 09 16
1619072154797 0x60a423fffe0da3d1 00 8f 01 01 00 01
1619072155787 0x60a423fffe0da3d1 00 8f 12 02 00 00 00 00 00
1619072156786 0x60a423fffe0da3d1 00 8f 13 02 00 00 00 00 00
1619072157844 0x60a423fffe0da3d1 00 8f 14 02 00 00 00 09 16
1619072184785 0x60a423fffe0da3d1 00 8f 01 01 00 01
1619072185783 0x60a423fffe0da3d1 00 8f 12 02 00 00 00 00 00
1619072186782 0x60a423fffe0da3d1 00 8f 13 02 00 00 00 00 00
1619072187784 0x60a423fffe0da3d1 00 8f 14 02 00 00 00 09 15
[... 313 lines cut ...]
1619074525692 0x60a423fffe0da3d1 00 93 12 02 00 00 00 00 00
1619074526700 0x60a423fffe0da3d1 00 93 13 02 00 00 00 00 00
1619074527706 0x60a423fffe0da3d1 00 93 14 02 00 00 00 09 06
1619074554782 0x60a423fffe0da3d1 00 93 01 01 00 01
1619074554892 0x60a423fffe0da3d1 00 93 01 01 00 01
1619074555696 0x60a423fffe0da3d1 00 93 12 02 00 00 00 00 00
1619074556690 0x60a423fffe0da3d1 00 93 13 02 00 00 00 00 00
1619074557709 0x60a423fffe0da3d1 00 93 14 02 00 00 00 09 04
1619074584690 0x60a423fffe0da3d1 00 93 01 01 00 01
1619074585688 0x60a423fffe0da3d1 00 93 12 02 00 00 00 00 00
1619074586689 0x60a423fffe0da3d1 00 93 13 02 00 00 00 00 00
1619074587740 0x60a423fffe0da3d1 00 93 14 02 00 00 00 09 05
1619074614711 0x60a423fffe0da3d1 00 93 01 01 00 01
1619074615723 0x60a423fffe0da3d1 00 93 12 02 00 00 00 00 00
1619074616698 0x60a423fffe0da3d1 00 93 13 02 00 00 00 00 00
1619074617695 0x60a423fffe0da3d1 00 93 14 02 00 00 00 09 03
1619074644702 0x60a423fffe0da3d1 00 93 01 01 00 01
1619074645693 0x60a423fffe0da3d1 00 93 12 02 00 00 00 00 00
1619074646692 0x60a423fffe0da3d1 00 94 13 02 00 00 00 00 00
1619074647693 0x60a423fffe0da3d1 00 94 14 02 00 00 00 09 06
1619074674684 0x60a423fffe0da3d1 00 94 01 01 00 01
1619074675692 0x60a423fffe0da3d1 00 94 12 02 00 00 00 00 00
1619074676694 0x60a423fffe0da3d1 00 94 13 02 00 00 00 00 00
1619074677695 0x60a423fffe0da3d1 00 94 14 02 00 00 00 09 05
```

ademobile, 2021-04-22:
Not sure, however this  may have some clues. It is a wifi version, however I am guessing same power monitoring chip.

[https://www.creationfactory.co/2019/12/reverse-engineering-zmai-90-din-rail.html](url)

Koenkk, 2021-04-27:
Merged, thanks!

closed 2021-04-27 by Koenkk

RubenKelevra, 2021-11-08:
Hey guys,

I was wondering if there's a chance to get more data out of this module. I'm interested in an accurate frequency measurement as well as cos-phi and real active power.

The docs states, that the device currently only providing "voltage, power, current, energy". So no frequency and no cos-phi.

Are those values not exported to zigbee or not read by zigbee2mqtt?

How much accuracy is available?

Here's what the device is supposedly able to show on the display:

[image]

Diddlik, 2021-11-24:
I tried to get this values too, but it seams this is not provided to zigbee. Only showed on device.

ProZielinski, 2021-11-25:
it doesn't work for me.