Koenkk/zigbee2mqtt #7159: Immax Neo 07089L

tralalas, 2021-04-21:
Hello I tried to add the Immax Neo 07089L device but without any luck.
My Zigbee2Mqtt is running under a docker container, I have created a .js file inside the data folder: 

immax_07089L.js
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');
const fz = zigbeeHerdsmanConverters.fromZigbeeConverters;
const e = zigbeeHerdsmanConverters.exposes.presets;
const device = {
   zigbeeModel: ['ZBT-CCTfilament-D0000'],
   model: '07089L',
   vendor: 'Immax',
   description: 'Immax LightBulb',
   fromZigbee: [fz.on_off, fz.brightness, fz.ignore_basic_report, fz.color_colortemp],
   toZigbee: [tz.light_onoff_brightness, tz.ignore_transition, tz.ignore_rate, tz.effect, tz.light_brightness_move, tz.light_brightness_step, tz.light_colortemp, tz.light_colortemp_move, tz.light_colortemp_step, tz.light_hue_saturation_move, tz.light_hue_saturation_step, tz.light_color, tz.read],
   exposes: [e.light_brightness_colortemp_colorxy()],
};
module.exports = device;

in configuration.yaml file I added the external_convertors:
external_convertors:
  - immax_07089L.js

 
## Information about the device + link
https://www.alza.cz/immax-neo-smart-led-filament-e27-5w-tepla-studena-bila-stmivatelna-zigbee-3-0-d5653212.htm?o=1

## data/database.db entry of the device
..{"id":2,"type":"Router","ieeeAddr":"0x680ae2fffee976c1","nwkAddr":19809,"manufId":4151,"manufName":"Immax","powerSource":"Mains (single phase)","modelId":"ZBT-CCTfilament-D0000","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821,4096,64642],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZBT-CCTfilament-D0000","manufacturerName":"Immax","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":101,"hwVersion":1,"dateCode":"20200107-244","swBuildId":"2.3"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":101,"hwVersion":1,"dateCode":"20200107-244","swBuildId":"2.3","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1618946478727}

##from the logs
2021-04-21T16:07:33.717369245Z Zigbee2MQTT:info  2021-04-21 16:07:33: Starting zigbee-herdsman (0.13.88)
2021-04-21T16:07:36.116754100Z Zigbee2MQTT:info  2021-04-21 16:07:36: zigbee-herdsman started
2021-04-21T16:07:36.120289786Z Zigbee2MQTT:info  2021-04-21 16:07:36: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
2021-04-21T16:07:36.131247395Z Zigbee2MQTT:info  2021-04-21 16:07:36: Currently 1 devices are joined:
2021-04-21T16:07:36.134733169Z Zigbee2MQTT:info  2021-04-21 16:07:36: BecTv (0x680ae2fffee976c1): Not supported (Router)
2021-04-21T16:07:36.136135211Z Zigbee2MQTT:warn  2021-04-21 16:07:36: `permit_join` set to  `true` in configuration.yaml.
2021-04-21T16:07:36.137479308Z Zigbee2MQTT:warn  2021-04-21 16:07:36: Allowing new devices to join.
2021-04-21T16:07:36.138509605Z Zigbee2MQTT:warn  2021-04-21 16:07:36: Set `permit_join` to `false` once you joined all devices.
2021-04-21T16:07:36.140479344Z Zigbee2MQTT:info  2021-04-21 16:07:36: Zigbee: allowing new devices to join.
2021-04-21T16:07:36.370663163Z Zigbee2MQTT:info  2021-04-21 16:07:36: Connecting to MQTT server at mqtt://localhost
2021-04-21T16:07:36.457830099Z Zigbee2MQTT:info  2021-04-21 16:07:36: Connected to MQTT server
2021-04-21T16:07:36.462470589Z Zigbee2MQTT:info  2021-04-21 16:07:36: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2021-04-21T16:07:36.533308298Z Zigbee2MQTT:info  2021-04-21 16:07:36: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"abd8a09","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.18.2"}'
2021-04-21T16:07:41.475335107Z Zigbee2MQTT:info  2021-04-21 16:07:41: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"dateCode":"20210120","friendly_name":"Coordinator","ieeeAddr":"0x00124b002281a278","lastSeen":1619021261473,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"},{"dateCode":"20200107-244","description":"-","friendly_name":"BecTv","hardwareVersion":1,"ieeeAddr":"0x680ae2fffee976c1","lastSeen":1618946478727,"manufacturerID":4151,"manufacturerName":"Immax","model":"ZBT-CCTfilament-D0000","modelID":"ZBT-CCTfilament-D0000","networkAddress":19809,"powerSource":"Mains (single phase)","softwareBuildID":"2.3","type":"Router","vendor":"-"}]'
2021-04-21T16:07:41.480127044Z Zigbee2MQTT:info  2021-04-21 16:07:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[{"dateCode":"20210120","friendly_name":"Coordinator","ieeeAddr":"0x00124b002281a278","lastSeen":1619021261479,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"},{"dateCode":"20200107-244","description":"-","friendly_name":"BecTv","hardwareVersion":1,"ieeeAddr":"0x680ae2fffee976c1","lastSeen":1618946478727,"manufacturerID":4151,"manufacturerName":"Immax","model":"ZBT-CCTfilament-D0000","modelID":"ZBT-CCTfilament-D0000","networkAddress":19809,"powerSource":"Mains (single phase)","softwareBuildID":"2.3","type":"Router","vendor":"-"}],"type":"devices"}'
2021-04-21T16:08:06.472022501Z Zigbee2MQTT:info  2021-04-21 16:08:06: Starting network scan (includeRoutes 'false')
2021-04-21T16:08:18.504905174Z Zigbee2MQTT:error 2021-04-21 16:08:18: Failed to execute LQI for 'BecTv'

Koenkk, 2021-04-21:
The converter looks good, if you can provide the min and max color temp (https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#31-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature) I can add it.

tralalas, 2021-04-21:
Hi thank you for the fast reply.
According to the vendor this is between Warm White -Cold White light from what I can see on some websites the numbers should be between 3000K - 4200K.

Koenkk, 2021-04-22:
I need the exact number returned by the bulb, see https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#31-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature

tralalas, 2021-04-23:
Hi I think that I'm doing something wrong or I don't get what I need to do:
I'm running this from my shell:
mosquitto_pub -t zigbee2mqtt/BecTv/set -m '{"read": {"cluster": "lightingColorCtrl", "attributes": ["colorTempPhysicalMin", "colorTempPhysicalMax"]}}'

in the container I can see this:
2021-04-23T14:19:01.763821945Z Zigbee2MQTT:debug 2021-04-23 14:19:01: Received MQTT message on 'zigbee2mqtt/BecTv/set' with data '{"read": {"cluster": "lightingColorCtrl", "attributes": ["colorTempPhysicalMin", "colorTempPhysicalMax"]}}'
2021-04-23T14:19:01.765891537Z Zigbee2MQTT:warn  2021-04-23 14:19:01: Device with modelID 'ZBT-CCTfilament-D0000' is not supported.
2021-04-23T14:19:01.767201763Z Zigbee2MQTT:warn  2021-04-23 14:19:01: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Any idea what am I doing wrong ?
Thank you

Koenkk, 2021-04-23:
Not sure what goes wrong here but I've added the device now. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

After this can you execute `mosquitto_pub -t zigbee2mqtt/BecTv/set -m '{"read": {"cluster": "lightingColorCtrl", "attributes": ["colorTempPhysicalMin", "colorTempPhysicalMax"]}}'` again and provide the result here?

tralalas, 2021-04-23:
Hi sure, thank you for the help

tralalas, 2021-04-24:
Hi so now the device is recognized:
2021-04-24T13:45:26.669222494Z Zigbee2MQTT:info  2021-04-24 13:45:26: Currently 1 devices are joined:
2021-04-24T13:45:26.672791813Z Zigbee2MQTT:info  2021-04-24 13:45:26: BecTv (0x680ae2fffee976c1): 07089L - Immax NEO SMART LED E27 5W (Router)

But it still doesn't work:
2021-04-24T13:52:02.441689444Z Zigbee2MQTT:debug 2021-04-24 13:52:02: Received MQTT message on 'zigbee2mqtt/BecTv/set' with data '{"state": "OFF"}'
2021-04-24T13:52:02.443579379Z Zigbee2MQTT:debug 2021-04-24 13:52:02: Publishing 'set' 'state' to 'BecTv'
2021-04-24T13:52:19.787495329Z Zigbee2MQTT:error 2021-04-24 13:52:19: Publish 'set' 'state' to 'BecTv' failed: 'Error: Command 0x680ae2fffee976c1/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
2021-04-24T13:52:19.789697503Z Zigbee2MQTT:debug 2021-04-24 13:52:19: Error: Command 0x680ae2fffee976c1/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))
2021-04-24T13:52:19.789819259Z     at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:339:27)
2021-04-24T13:52:19.789893732Z     at Generator.next (<anonymous>)
2021-04-24T13:52:19.789919161Z     at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
2021-04-24T13:52:19.791279380Z Zigbee2MQTT:info  2021-04-24 13:52:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'BecTv' failed: 'Error: Command 0x680ae2fffee976c1/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"BecTv"},"type":"zigbee_publish_error"}'

Any ideas on what the issue could be ?
Thank you.

Koenkk, 2021-04-24:
Looks like Zigbee connection fails (maybe interference: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html), try repairing.

tralalas, 2021-04-27:
Hi everything is fine now ... I have some wifi issues but with the lightbulb moved closer to the zigbee gw evertyhing works fine.
Thank you for the help

closed 2021-04-27 by tralalas